pulp-platform/ara #18: Multiple operations on the same `vd`, with different `VSEW` and `vl`

mp-17, 2021-03-15:
> VSET(1, e64, m1);
  VLOAD_64(v1, 0xffffffffffffffff);
  VSET(16, e16, m1);
  VLOAD_16(v2, 0x33ca, ..., 0x314f);
  VLOAD_16(v3, 0x33ca, ..., 0xbb98);
  asm volatile("vmfeq.vv v1, v2, v3");
  VSET(1, e64, m1);
  VCMP_U64(13, v1, 0xffffffffffff0001);

The same register is first loaded with one 64-bit element, and then a 16 bits mask is produced from a `e16` comparison. The final store operation de-shuffles the `VRF` elements as the register hosted `e16` data. The problem is that the mask unit does not store the mask results following a `e16` shuffled pattern.

Potentially, the issue is broader:
Ara can have problems with multiple operations on the same register if `VSEW` and `vl` are changed, as not all the elements are re-shuffled during the second operation. Part of the old elements remains shuffled in the register (check this, and open another issue if true).

suehtamacv, 2021-03-16:
Can you provide a smaller code snippet where this error occurs? The actual assembly code would also help. 

This issue could be fixed by making the Mask Unit reorganize all the elements of a vector, up to `vl`, as it is the case with the other functional units.

mp-17, 2021-09-21:
The following example runs smoothly on SPIKE, but it is not correctly handled by Ara.

```
void TEST_CASE5(void) {
  int64_t zero = 0;
  VSET(2, e64, m1);
  VLOAD_64(v0, 0x76543210fedcba98, 0x76543210fedcba98);
  VSET(2, e16, m1);
  asm volatile("vmul.vx v0, v0, %[A]" :: [A] "r"(zero));
  VCMP_U64(13, v0, 0x7654321000000000, 0x76543210fedcba98);
};

```

closed 2023-11-28 by mp-17