pulp-platform/ara #27: RTL simulation getting failed with Verilator

mahmoodulhassan-lm, 2021-03-25:
I am trying to run makefile target `make verilate` but I am getting the following errors on the default settings.

```
$ make verilate 
Makefile:43: "Specified QuestaSim version (questa-2020.1) not found in PATH 
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 > build/verilator/bender_script
/home/ara/install/verilator/bin/verilator -f build/verilator/bender_script                     \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LITENDIAN                                                                \
  -Wno-MODDUP                                                                   \
  -Wno-PINMISSING                                                               \
  -Wno-SYMRSVDWORD                                                              \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  --Mdir build/verilator --trace                                               \
  -Itb/dpi                                                                      \
  -CFLAGS "-std=c++11 -Wall -DTOPLEVEL_NAME=ara_tb_verilator"                       \
  -CFLAGS "-DNR_LANES=4"                                              \
  -CFLAGS -I/home/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  -LDFLAGS "-lelf"                                                              \
  --exe                                                                         \
  /home/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
%Error: /home/ara/hardware/deps/tech_cells_generic/src/rtl/tc_sram.sv:93:38: Unsupported or unknown PLI call: $urandom
   93 |           "random": init_val[i][j] = $urandom();
      |                                      ^~~~~~~~
%Error: /home/ara/hardware/tb/ara_tb.sv:30:28: syntax error, unexpected TIME NUMBER, expecting TYPE-IDENTIFIER
   30 |   localparam ClockPeriod = 1ns;
      |                            ^~~
%Warning-STMTDLY: /home/ara/hardware/tb/ara_tb.sv:48:10: Unsupported: Ignoring delay on this delayed statement.
   48 |   always #(ClockPeriod/2) clk = !clk;
      |          ^
                  ... Use "/* verilator lint_off STMTDLY */" and lint_on around source to disable this message.
%Warning-STMTDLY: /home/ara/hardware/tb/ara_tb.sv:56:7: Unsupported: Ignoring delay on this delayed statement.
   56 |       #(ClockPeriod);
      |       ^
%Warning-STMTDLY: /home/ara/hardware/tb/ara_tb.sv:98:7: Unsupported: Ignoring delay on this delayed statement.
   98 |       #ClockPeriod;
      |       ^
%Error: Exiting due to 2 error(s), 3 warning(s)
        ... See the manual and https://verilator.org for more assistance.
Makefile:123: recipe for target 'build/verilator/Vara_tb_verilator' failed
make: *** [build/verilator/Vara_tb_verilator] Error 1
```

What could be the possible cause for that?

suehtamacv, 2021-03-25:
Regarding the first error, running `make apply-patches` in the hardware folder patches the `tc_sram` module, making it compatible with Verilator, as stated in the README. 

I cannot reproduce the second issue. I wonder if it is not being caused because of the issue with `tc_sram`... Could you please try to apply the patch and try to run Verilator again?

mahmoodulhassan-lm, 2021-03-25:
> Regarding the first error, running `make apply-patches` in the hardware folder patches the `tc_sram` module, making it compatible with Verilator, as stated in the README.
> 
> I cannot reproduce the second issue. I wonder if it is not being caused because of the issue with `tc_sram`... Could you please try to apply the patch and then run Verilator again?

I have applied the patches while configuring the Verilator as per the given instruction in the readme and got the aforementioned error. Now if I try to reapply patches, following error appears:
```

 $ make apply-patches
Makefile:48: "Specified QuestaSim version (questa-2020.1) not found in PATH"
cd deps/tech_cells_generic && git apply ../../patches/0001-tech-cells-generic-sram.patch
error: patch failed: src/rtl/tc_sram.sv:124
error: src/rtl/tc_sram.sv: patch does not apply
Makefile:88: recipe for target 'apply-patches' failed
make: *** [apply-patches] Error 1
```

suehtamacv, 2021-03-25:
Okay, so the patch is already applied, no need to reapply it. 

It is very strange how Verilator is complaining about those constructs. As you can see in our [CI](https://github.com/pulp-platform/ara/runs/2191995850), Ara is verilated without any issues. Could you please run `install/verilator/bin/verilator --version`, to check whether the correct version was installed? On my side, the result of this command is:
```
> install/verilator/bin/verilator --version
Verilator 4.106 2020-12-02 rev v4.106
```

mahmoodulhassan-lm, 2021-03-25:
> `install/verilator/bin/verilator --version`

I am having the following version of Verilator:

Verilator 4.032 2020-04-04 rev v4.032

suehtamacv, 2021-03-25:
In our toolchain installation scripts, we check out a specific version of Verilator before compiling it:

https://github.com/pulp-platform/ara/blob/a4e2f226957b713e25ac937d11d093a72304c8a3/Makefile#L69-L76

As you can see in the following line, the default version of Verilator is v4.106, with which Ara was tested to Verilate:

https://github.com/pulp-platform/ara/blob/a4e2f226957b713e25ac937d11d093a72304c8a3/Makefile#L26

I am not sure why your Verilator is of an older version, but the warnings should go away once you recompile it and update it to the latest version. Let me know if that works!

closed 2021-03-27 by suehtamacv

mahmoodulhassan-lm, 2021-03-29:
@suehtamacv The issue was resolved by updating the Verilator version. Thanks for the cooperation.

adz0612, 2022-03-08:
Hey, can you guide me on how to update Verilator? I'm facing a similar issue.