pulp-platform/ara #30: Ensure correct execution when vl = 0

mp-17, 2021-03-31:
When `vl = 0`, vector loads do not work as expected and generate undefined behaviour.

suehtamacv, 2021-03-31:
I believe any instruction with `vl = 0` can be considered as a NOP. The main dispatcher can drop them.

mp-17, 2021-03-31:
I thought the same thing, but maybe some instructions must still be able to raise exceptions (e.g., exceptions related to the address). What do you think about that?

suehtamacv, 2021-03-31:
Which address?

mp-17, 2021-03-31:
For example, an address generated for a load. But I agree that we can avoid generating the exception since no elements should be loaded.

mp-17, 2021-03-31:
Addressed with this commit (https://github.com/pulp-platform/ara/commit/9f2eda8317de2e443ed1b0cd885bebc686cb46ce): I'll create a MR after the current `hotfixes` will be implemented.

closed 2021-03-31 by mp-17