pulp-platform/ara #31: Correclty store 128-bit unaligned arrays in memory

mp-17, 2021-03-31:
The store unit commits an off-by-one error when tries to store arrays that are not 128-bit aligned in memory (128-bit is the AXI data width). For now, benchmarks work on 128-bit aligned data to overcome this issue.

mp-17, 2021-04-16:
With the commits 1f9c2f54f6919e7b532ec2bd380a559aafcd1afc and 7e4c17dfa80d9d9349213abe0f1d320f8598d5e7, the load operations should now work even when starting from memory locations not aligned to `4*NrLanes` Byte.
For the store operations, I do not find an easy fix. The problem arises since the lanes give the data to the store unit in a synchronized fashion, and receive a grant altogether. In the case of an unaligned store, we might need some of the old data from some lanes and new data from the remaining lanes to fill the AXI beat.
The easiest solution I see is that we need to de-synchronize the lanes, having one `stu_operand_ready_o` signal per lane. Every time we use a value, we acknowledge the corresponding unit. The problem is that we need the same also for the mask values; the mask unit should either be able to store some mask values or process them individually.

mp-17, 2021-05-25:
With commit 650ca1f1938e1ce92a3837c8b7d3b0cd6579d90b, the problem should be solved. The effective length of the AXI bus is reduced to accommodate the misalignment, and the effective length must be a divisor of `8*NrLanes`.

suehtamacv, 2021-05-26:
There are a few style issues with that commit, it is not compliant with the code style, but before I do not know where to review it. The Clang PR is not passing, and I am waiting for that to review it. However, the fixes to this issue are attached to that PR.. What should I do?

suehtamacv, 2021-07-15:
Closed since the commit was merged with #40.

closed 2021-07-15 by suehtamacv