pulp-platform/ara #32: Error when compiling hello_world

strongwind312, 2021-04-02:
i'm trying to run make bin/hello_world, but i got the following errors:

make bin/hello_world
/home/workspace/pulp/ara_lenovo_unix/install/riscv-gcc/bin/riscv64-unknown-elf-gcc -mcmodel=medany -march=rv64gcv -mabi=lp64 -I/home/workspace/pulp/ara_lenovo_unix/apps/common -static -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra  -c hello_world/main.c -o hello_world/main.c.o
/home/workspace/pulp/ara_lenovo_unix/install/riscv-gcc/bin/riscv64-unknown-elf-gcc -mcmodel=medany -march=rv64gcv -mabi=lp64 -I/home/workspace/pulp/ara_lenovo_unix/apps/common -static -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra  -c common/crt0.S -o common/crt0.S.o
common/encoding.h: Assembler messages:
common/encoding.h:1: Error: unknown pseudo-op: `..'
common/crt0.S:80: Error: illegal operands `li t0,PMP_NAPOT|PMP_R|PMP_W|PMP_X'
common/crt0.S:92: Error: illegal operands `li t0,(1<<CAUSE_LOAD_PAGE_FAULT)|(1<<CAUSE_STORE_PAGE_FAULT)|(1<<CAUSE_FETCH_PAGE_FAULT)|(1<<CAUSE_MISALIGNED_FETCH)|(1<<CAUSE_USER_ECALL)|(1<<CAUSE_BREAKPOINT)'
common/crt0.S:102: Error: illegal operands `li t0,(MSTATUS_FS&(MSTATUS_FS>>1))'
common/crt0.S:106: Error: illegal operands `li t0,(MSTATUS_VS&(MSTATUS_VS>>1))'
/home/workspace/pulp/ara_lenovo_unix/apps/common/runtime.mk:66: recipe for target 'common/crt0.S.o' failed
make: *** [common/crt0.S.o] Error 1
rm hello_world/main.c.o

what could be the reason for this ?

suehtamacv, 2021-04-05:
Hi @strongwind312,

I wonder if you are not missing the `encoding.h` file, which contains the definitions of macros such as `PMP_NAPOT` or `MSTATUS_FS`, and so on. In Ara, this file is available in `apps/common/encoding.h` as a soft-link to the file inside the Spike submodule. 

Do you have this `encoding.h` file? Can you read it?

strongwind312, 2021-04-06:
Thank you for your help @suehtamacv 
I have encoding.h,
[image]
and the real file is located in toolchain/riscv-isa-sim/riscv/encoding.h, it contains the required macro definitions.
[image]

I think the problem may be the relative file path "../../xxx", if i copy toolchain/riscv-isa-sim/riscv/encoding.h to apps/common to replace the original encoding.h, make bin/hello_world can run correctly.

suehtamacv, 2021-04-06:
That is interesting... I cannot reproduce it, but I have no issues in making a copy of `encoding.h` to the `apps/common` folder. I will open a PR to handle that.

suehtamacv, 2021-04-06:
Hi @strongwind312,

I created a new branch, `encoding`, which replaces the softlink by a copy of `encoding.h`. Could you please check whether it solves your issue?

strongwind312, 2021-04-06:
Ok, i will have a try and feed back later. @suehtamacv

strongwind312, 2021-04-08:
Hi, @suehtamacv ,
the encoding branch solved my problem.

PR #34 (2021-04-08): Copy encoding.h from Spike to the common Ara runtime
This PR copies `encoding.h` from the Spike submodule to the common Ara runtime.

Closes #32.

## Changelog

### Changed
- The `encoding.h` in the common Ara runtime is now a copy from the `encoding.h` in the Spike submodule

## Checklist

- [x] Automated tests pass
- [x] Changelog updated
- [x] Code style guideline is observed

Please check our [contributing guidelines](CONTRIBUTING.md) before opening a Pull Request.
closed 2021-04-08 by suehtamacv