pulp-platform/ara #42: Error doing RTL simulation with make verilate

ismrgb, 2021-05-27:
<pre><font color="#729FCF"><b>ara/hardware</b></font>$ make verilate
Makefile:43: &quot;Specified QuestaSim version (questa-2020.1) not found in PATH /usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/opt/riscv/bin:/opt/symbioticeda/bin&quot;
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 &gt; build/verilator/bender_script
/home/con/Desktop/ARA_/ara/install/verilator/bin/verilator -f build/verilator/bender_script                     \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LITENDIAN                                                                \
  -Wno-MODDUP                                                                   \
  -Wno-PINMISSING                                                               \
  -Wno-SYMRSVDWORD                                                              \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  --Mdir build/verilator --trace                                               \
  -Itb/dpi                                                                      \
  -CFLAGS &quot;-std=c++11 -Wall -DTOPLEVEL_NAME=ara_tb_verilator&quot;                       \
  -CFLAGS &quot;-DNR_LANES=4&quot;                                              \
  -CFLAGS -I/home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  -LDFLAGS &quot;-lelf&quot;                                                              \
  --exe                                                                         \
  /home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/con/Desktop/ARA_/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
  --top-module ara_tb_verilator &amp;&amp;                                                  \
cd build/verilator &amp;&amp; OBJCACHE=&apos;&apos; make -j4 -f Vara_tb_verilator.mk

Killed
%Error: Command Failed /home/con/Desktop/ARA_/ara/install/verilator/bin/verilator_bin -f build/verilator/bender_script -GNrLanes\=4 -O3 -Wno-BLKANDNBLK -Wno-CASEINCOMPLETE -Wno-CMPCONST -Wno-LITENDIAN -Wno-MODDUP -Wno-PINMISSING -Wno-SYMRSVDWORD -Wno-UNOPTFLAT -Wno-UNPACKED -Wno-UNSIGNED -Wno-WIDTH -Wno-WIDTHCONCAT --Mdir build/verilator --trace -Itb/dpi -CFLAGS -std\=c\+\+11\ -Wall\ -DTOPLEVEL_NAME\=ara_tb_verilator -CFLAGS -DNR_LANES\=4 -CFLAGS -I/home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -CFLAGS -I/home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -CFLAGS -I/home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp -LDFLAGS -lelf --exe /home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/dpi_memutil.cc /home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/sv_scoped.cc /home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/verilator_memutil.cc /home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.cc /home/con/Desktop/ARA_/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc /home/con/Desktop/ARA_/ara/hardware/tb/verilator/ara_tb.cpp --cc --top-module ara_tb_verilator
Makefile:125: recipe for target &apos;build/verilator/Vara_tb_verilator&apos; failed
make: *** [build/verilator/Vara_tb_verilator] Error 137</pre>

roychen1998, 2021-05-30:
Hi Ismrgb,

I also met some issue when 'make verilator' in hardware directory. The case hang when compiling Vara_tb_verilator__Trace__13__Slow.cpp for more than 1 day, then I kill the thread.
"g++  -I.  -MMD -I/media/roychen/Elements/riscv/ara/install/verilator/share/verilator/include -I/media/roychen/Elements/riscv/ara/install/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -Wall -DTOPLEVEL_NAME=ara_tb_verilator -DNR_LANES=4 -I/media/roychen/Elements/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -I/media/roychen/Elements/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -I/media/roychen/Elements/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -std=gnu++14  -c -o Vara_tb_verilator__Trace__13__Slow.o Vara_tb_verilator__Trace__13__Slow.cpp"

Best Regard,
Roy

suehtamacv, 2021-05-31:
Verilator always took a lot of RAM and processing resources to verilate Ara. And with the integration of more functionalinites on Ara, the runtime and RAM usage keeps getting higher. It works on our machines and on the CI of this repo, but as you can see from the runtimes, GitHub's CI takes more than one hour to verilate the design.

I am not sure how to solve this generically, since it seems to be something related to the `gcc` version you are using, the available RAM, and the target system for the verilated model. One idea to reduce the runtime is to verilate the design hierarchically, compiling the lanes separately from the rest of the design. There are performance implications when doing that.

Fog-cake, 2021-06-01:
I also encountered an error when running the command, but I was prompted that something was missing.

ismrgb, 2021-06-02:
Yeah, I'm trying this on a VM and the problem went away after giving it more resources. I also needed to do apt-get install libdw-dev in case anyone ran into something similar, Thanks suehtamacv & roychen1998.

closed 2021-06-02 by ismrgb