pulp-platform/ara #43: run RTL simulation with make verilate report no such file or directory

Fog-cake, 2021-06-01:
I am trying to run makefile，but there were some mistakes
~/riscv/ara/hardware$ make verilate
Makefile:43: "Specified QuestaSim version (questa-2020.1) not found in PATH /home/wu/riscv/gcc/riscv-unknown-elf-gcc/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 > build/verilator/bender_script
/home/wu/riscv/ara/install/verilator/bin/verilator -f build/verilator/bender_script                     \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LITENDIAN                                                                \
  -Wno-MODDUP                                                                   \
  -Wno-PINMISSING                                                               \
  -Wno-SYMRSVDWORD                                                              \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  --Mdir build/verilator --trace                                               \
  -Itb/dpi                                                                      \
  -CFLAGS "-std=c++11 -Wall -DTOPLEVEL_NAME=ara_tb_verilator"                       \
  -CFLAGS "-DNR_LANES=4"                                              \
  -CFLAGS -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  -LDFLAGS "-lelf"                                                              \
  --exe                                                                         \
  /home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/wu/riscv/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
make[1]: Entering directory '/home/wu/riscv/ara/hardware/build/verilator'
g++  -I.  -MMD -I/home/wu/riscv/ara/install/verilator/share/verilator/include -I/home/wu/riscv/ara/install/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -Wall -DTOPLEVEL_NAME=ara_tb_verilator -DNR_LANES=4 -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -std=gnu++14 -Os -c -o ara_tb.o /home/wu/riscv/ara/hardware/tb/verilator/ara_tb.cpp
g++  -I.  -MMD -I/home/wu/riscv/ara/install/verilator/share/verilator/include -I/home/wu/riscv/ara/install/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -Wall -DTOPLEVEL_NAME=ara_tb_verilator -DNR_LANES=4 -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -std=gnu++14 -Os -c -o dpi_memutil.o /home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/dpi_memutil.cc
g++  -I.  -MMD -I/home/wu/riscv/ara/install/verilator/share/verilator/include -I/home/wu/riscv/ara/install/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -Wall -DTOPLEVEL_NAME=ara_tb_verilator -DNR_LANES=4 -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -std=gnu++14 -Os -c -o sv_scoped.o /home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/sv_scoped.cc
g++  -I.  -MMD -I/home/wu/riscv/ara/install/verilator/share/verilator/include -I/home/wu/riscv/ara/install/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=c++11 -Wall -DTOPLEVEL_NAME=ara_tb_verilator -DNR_LANES=4 -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -I/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -std=gnu++14 -Os -c -o verilator_memutil.o /home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/verilator_memutil.cc
/home/wu/riscv/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/dpi_memutil.cc:11:10: fatal error: libelf.h: No such file or directory
   11 | #include <libelf.h>
      |          ^~~~~~~~~~
compilation terminated.
make[1]: *** [Vara_tb_verilator.mk:75: dpi_memutil.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/wu/riscv/ara/hardware/build/verilator'
make: *** [Makefile:127: build/verilator/Vara_tb_verilator] Error 2

Fog-cake, 2021-06-01:
I should have a corresponding library.

wu@ubuntu:~/riscv/ara/hardware$ sudo apt-get install libfl-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libfl-dev is already the newest version (2.6.4-6.2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Fog-cake, 2021-06-01:
hi，I seem to install the following library can run, thanks.

apt-get install libdw-dev

closed 2021-06-01 by Fog-cake

suehtamacv, 2021-06-01:
Hi @Fog-cake,

Indeed, you are correct. Checking the commands we run at GitHub's CI to verilate Ara, we do install `libelf-dev` before running the verilation command.

https://github.com/pulp-platform/ara/blob/795818c3c6daecead2fd6bc9cea48668bb51f5a0/.github/workflows/ci.yml#L210

Matheus

Fog-cake, 2021-06-01:
Sorry, in fact, I still have a question about this. I still have a computer with a centos7 system. I also encounter the same problem when executing this script. Which library should I install? I read that your ci.yml is based on Ubuntu.