pulp-platform/ara #44: Version requirements of RTL simulation tool

Fog-cake, 2021-06-01:
Through reading makefie, I found that your default simulation tool is questasim 2020.1, but for some reasons, I only have version 10.4 of the simulation tool. Can I simulate a lower version than 2020? 
Thanks.

suehtamacv, 2021-06-01:
Hi @Fog-cake,

I am not sure, since I do not have access to this specific version. We do not _support_ Questa 10.4, but there is a good chance it works. I tried simulating Ara with Questa 10.6, and it worked without any issues. 

You could try, and report here any issues you have.. I will leave this issue open.

Matheus

Fog-cake, 2021-06-02:
Hi  suehtamacv,
In order to be able to simulate RTL faster, I used the 2020.1 version. But still encountered a problem. It seems that I installed 32-bit software on a 64-bit system. But I am not sure how to solve it. The complete information is this:

# vsim "+UVM_NO_RELNOTES" "+PRELOAD=/home/wu/riscv/ara/apps/bin/hello_world" -sv_lib work-dpi/ara_dpi -work work -voptargs="+acc" -suppress vsim-3009 -suppress vopt-7033 work.ara_tb -do "../scripts/run.tcl" 
# Start time: 18:28:51 on Jun 02,2021
# Loading /tmp/wu@localhost.localdomain_dpi_3276/linux_gcc-7.4.0/export_tramp.so
# ** Note: (vsim-3812) Design is being optimized...
# ** Note: (vopt-143) Recognized 1 FSM in module "load_unit(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "simd_div(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "ptw(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "addrgen(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_isolate(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__2)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__3)".
# ** Note: (vopt-143) Recognized 1 FSM in module "cva6_icache(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi2apb_64_32(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_llc_config(fast)".
# ** Note: (vopt-143) Recognized 5 FSMs in module "axi_dw_upsizer(fast)".
# ** Warning: /home/wu/riscv/ara/hardware/deps/axi/src/axi_dw_downsizer.sv(719): (vopt-2182) 'mst_req[113]' might be read before written in always_comb or always @* block.
# ** Note: (vopt-143) Recognized 3 FSMs in module "axi_dw_downsizer(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "wt_dcache_missunit(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "sldu(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__1)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_shim(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__1)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__2)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__3)".
# ** Note: (vopt-143) Recognized 1 FSM in module "serdiv(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "serdiv(fast__1)".
# ** Note: (vopt-143) Recognized 1 FSM in module "wt_dcache_ctrl(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_llc_tag_pattern_gen(fast)".
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=0, Warnings=1.
# Loading sv_std.std
# Loading work.ara_tb_sv_unit(fast)
# Loading work.ara_tb(fast)
# Loading work.axi_pkg(fast)
# Loading work.ara_testharness(fast)
# Loading work.riscv(fast)
# Loading work.rvv_pkg(fast)
# Loading work.fpnew_pkg(fast)
# Loading work.dm(fast)
# Loading work.ariane_pkg(fast)
# Loading work.ara_pkg(fast)
# Loading work.ara_soc(fast)
# Loading work.cf_math_pkg(fast)
# Loading work.axi_xbar(fast)
# Loading work.addr_decode(fast)
# Loading work.axi_demux(fast)
# Loading work.spill_register(fast)
# Loading work.axi_demux_id_counters(fast)
# Loading work.delta_counter(fast)
# Loading work.counter(fast)
# Loading work.spill_register(fast__1)
# Loading work.spill_register(fast__2)
# Loading work.rr_arb_tree(fast)
# Loading work.spill_register(fast__3)
# Loading work.spill_register(fast__4)
# Loading work.rr_arb_tree(fast__1)
# Loading work.axi_err_slv(fast)
# Loading work.axi_atop_filter(fast)
# Loading work.stream_register(fast)
# Loading work.fifo_v2(fast)
# Loading work.fifo_v3(fast)
# Loading work.fifo_v3(fast__1)
# Loading work.fifo_v3(fast__2)
# Loading work.fifo_v3(fast__3)
# Loading work.counter(fast__1)
# Loading work.delta_counter(fast__1)
[... 545 lines cut ...]
# 
#    Time: 0 ps  Iteration: 0  Region: /uvm_pkg File: /home/wu/questasim_202001/questasim/linux/../verilog_src/uvm-1.1d/src/uvm_pkg.sv
# Loading ./work-dpi/ara_dpi.so
# ** Error: (vsim-125) The shared library ./work-dpi/ara_dpi.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation. Please use compatible machines to build and load the library
# ** Error (suppressible): (vsim-3197) Load of "./work-dpi/ara_dpi.so" failed: ./work-dpi/ara_dpi.so: wrong ELF class: ELFCLASS64.
# ** Error (suppressible): (vsim-3748) Failed to load DPI object file "./work-dpi/ara_dpi.so" while trying to resolve 'uvm_dpi_regfree'.
# ** Warning: (vsim-3116) Problem reading symbols from ./work-dpi/ara_dpi.so : Not ELFCLASS32.
# ** Warning: ** Warning: (vsim-3772) Failed to find the undefined symbols for shared library file "./work-dpi/ara_dpi.so".
# 
#    Time: 0 ps  Iteration: 0  Region: /uvm_pkg File: /home/wu/questasim_202001/questasim/linux/../verilog_src/uvm-1.1d/src/uvm_pkg.sv
# Loading ./work-dpi/ara_dpi.so
# ** Error: (vsim-125) The shared library ./work-dpi/ara_dpi.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation. Please use compatible machines to build and load the library
# ** Error (suppressible): (vsim-3197) Load of "./work-dpi/ara_dpi.so" failed: ./work-dpi/ara_dpi.so: wrong ELF class: ELFCLASS64.
# ** Error (suppressible): (vsim-3748) Failed to load DPI object file "./work-dpi/ara_dpi.so" while trying to resolve 'uvm_dump_re_cache'.
[... cut ...]
# ** Error: (vsim-125) The shared library ./work-dpi/ara_dpi.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation. Please use compatible machines to build and load the library
# ** Error (suppressible): (vsim-3197) Load of "./work-dpi/ara_dpi.so" failed: ./work-dpi/ara_dpi.so: wrong ELF class: ELFCLASS64.
# ** Error (suppressible): (vsim-3748) Failed to load DPI object file "./work-dpi/ara_dpi.so" while trying to resolve 'uvm_glob_to_re'.
# Error loading design
# End time: 18:30:19 on Jun 02,2021, Elapsed time: 0:01:28
# Errors: 36, Warnings: 29, Suppressed Errors: 26, Suppressed Warnings: 228

Fog-cake, 2021-06-02:
# vsim "+UVM_NO_RELNOTES" "+PRELOAD=/home/wu/riscv/ara/apps/bin/hello_world" -sv_lib work-dpi/ara_dpi -work work -voptargs="+acc" -suppress vsim-3009 -suppress vopt-7033 work.ara_tb -do "../scripts/run.tcl" 
# Start time: 18:28:51 on Jun 02,2021
# Loading /tmp/wu@localhost.localdomain_dpi_3276/linux_gcc-7.4.0/export_tramp.so
# ** Note: (vsim-3812) Design is being optimized...
# ** Note: (vopt-143) Recognized 1 FSM in module "load_unit(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "simd_div(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "ptw(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "addrgen(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_isolate(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__2)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__3)".
# ** Note: (vopt-143) Recognized 1 FSM in module "cva6_icache(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi2apb_64_32(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_llc_config(fast)".
# ** Note: (vopt-143) Recognized 5 FSMs in module "axi_dw_upsizer(fast)".
# ** Warning: /home/wu/riscv/ara/hardware/deps/axi/src/axi_dw_downsizer.sv(719): (vopt-2182) 'mst_req[113]' might be read before written in always_comb or always @* block.
# ** Note: (vopt-143) Recognized 3 FSMs in module "axi_dw_downsizer(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "wt_dcache_missunit(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "sldu(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__1)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_shim(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__1)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__2)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__3)".
# ** Note: (vopt-143) Recognized 1 FSM in module "serdiv(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "serdiv(fast__1)".
# ** Note: (vopt-143) Recognized 1 FSM in module "wt_dcache_ctrl(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_llc_tag_pattern_gen(fast)".
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=0, Warnings=1.
# Loading sv_std.std
# Loading work.ara_tb_sv_unit(fast)
# Loading work.ara_tb(fast)
# Loading work.axi_pkg(fast)
# Loading work.ara_testharness(fast)
# Loading work.riscv(fast)
# Loading work.rvv_pkg(fast)
# Loading work.fpnew_pkg(fast)
# Loading work.dm(fast)
# Loading work.ariane_pkg(fast)
# Loading work.ara_pkg(fast)
# Loading work.ara_soc(fast)
# Loading work.cf_math_pkg(fast)
# Loading work.axi_xbar(fast)
# Loading work.addr_decode(fast)
# Loading work.axi_demux(fast)
# Loading work.spill_register(fast)
# Loading work.axi_demux_id_counters(fast)
# Loading work.delta_counter(fast)
# Loading work.counter(fast)
# Loading work.spill_register(fast__1)
# Loading work.spill_register(fast__2)
# Loading work.rr_arb_tree(fast)
# Loading work.spill_register(fast__3)
# Loading work.spill_register(fast__4)
# Loading work.rr_arb_tree(fast__1)
# Loading work.axi_err_slv(fast)
# Loading work.axi_atop_filter(fast)
# Loading work.stream_register(fast)
# Loading work.fifo_v2(fast)
# Loading work.fifo_v3(fast)
# Loading work.fifo_v3(fast__1)
# Loading work.fifo_v3(fast__2)
# Loading work.fifo_v3(fast__3)
# Loading work.counter(fast__1)
# Loading work.delta_counter(fast__1)
# Loading work.axi_multicut(fast)
# Loading work.axi_mux(fast)
# Loading work.axi_id_prepend(fast)
[... 542 lines cut ...]
# 
#    Time: 0 ps  Iteration: 0  Region: /uvm_pkg File: /home/wu/questasim_202001/questasim/linux/../verilog_src/uvm-1.1d/src/uvm_pkg.sv
# Loading ./work-dpi/ara_dpi.so
# ** Error: (vsim-125) The shared library ./work-dpi/ara_dpi.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation. Please use compatible machines to build and load the library
# ** Error (suppressible): (vsim-3197) Load of "./work-dpi/ara_dpi.so" failed: ./work-dpi/ara_dpi.so: wrong ELF class: ELFCLASS64.
# ** Error (suppressible): (vsim-3748) Failed to load DPI object file "./work-dpi/ara_dpi.so" while trying to resolve 'uvm_dpi_regfree'.
# ** Warning: (vsim-3116) Problem reading symbols from ./work-dpi/ara_dpi.so : Not ELFCLASS32.
# ** Warning: ** Warning: (vsim-3772) Failed to find the undefined symbols for shared library file "./work-dpi/ara_dpi.so".
# 
#    Time: 0 ps  Iteration: 0  Region: /uvm_pkg File: /home/wu/questasim_202001/questasim/linux/../verilog_src/uvm-1.1d/src/uvm_pkg.sv
# Loading ./work-dpi/ara_dpi.so
# ** Error: (vsim-125) The shared library ./work-dpi/ara_dpi.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation. Please use compatible machines to build and load the library
# ** Error (suppressible): (vsim-3197) Load of "./work-dpi/ara_dpi.so" failed: ./work-dpi/ara_dpi.so: wrong ELF class: ELFCLASS64.
# ** Error (suppressible): (vsim-3748) Failed to load DPI object file "./work-dpi/ara_dpi.so" while trying to resolve 'uvm_dump_re_cache'.
# ** Warning: (vsim-3116) Problem reading symbols from ./work-dpi/ara_dpi.so : Not ELFCLASS32.
[... cut ...]
# ** Error: (vsim-125) The shared library ./work-dpi/ara_dpi.so being loaded was built on a 64-bit machine. A 64-bit shared library cannot be loaded in a 32-bit simulation. Please use compatible machines to build and load the library
# ** Error (suppressible): (vsim-3197) Load of "./work-dpi/ara_dpi.so" failed: ./work-dpi/ara_dpi.so: wrong ELF class: ELFCLASS64.
# ** Error (suppressible): (vsim-3748) Failed to load DPI object file "./work-dpi/ara_dpi.so" while trying to resolve 'uvm_glob_to_re'.
# Error loading design
# End time: 18:30:19 on Jun 02,2021, Elapsed time: 0:01:28
# Errors: 36, Warnings: 29, Suppressed Errors: 26, Suppressed Warnings: 228

Fog-cake, 2021-06-02:
In order to be able to simulate RTL faster, I used the 2020.1 version. But still encountered a problem. It seems that I installed 32-bit software on a 64-bit system. But I am not sure how to solve it. The complete information is this:

# vsim "+UVM_NO_RELNOTES" "+PRELOAD=/home/wu/riscv/ara/apps/bin/hello_world" -sv_lib work-dpi/ara_dpi -work work -voptargs="+acc" -suppress vsim-3009 -suppress vopt-7033 work.ara_tb -do "../scripts/run.tcl" 
# Start time: 18:28:51 on Jun 02,2021
# Loading /tmp/wu@localhost.localdomain_dpi_3276/linux_gcc-7.4.0/export_tramp.so
# ** Note: (vsim-3812) Design is being optimized...
# ** Note: (vopt-143) Recognized 1 FSM in module "load_unit(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "simd_div(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "ptw(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "addrgen(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_isolate(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__2)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__3)".
# ** Note: (vopt-143) Recognized 1 FSM in module "cva6_icache(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi2apb_64_32(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_llc_config(fast)".
# ** Note: (vopt-143) Recognized 5 FSMs in module "axi_dw_upsizer(fast)".
# ** Warning: /home/wu/riscv/ara/hardware/deps/axi/src/axi_dw_downsizer.sv(719): (vopt-2182) 'mst_req[113]' might be read before written in always_comb or always @* block.
# ** Note: (vopt-143) Recognized 3 FSMs in module "axi_dw_downsizer(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "wt_dcache_missunit(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "sldu(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "fpnew_divsqrt_multi(fast__1)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_shim(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__1)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__2)".
# ** Note: (vopt-143) Recognized 2 FSMs in module "axi_atop_filter(fast__3)".
# ** Note: (vopt-143) Recognized 1 FSM in module "serdiv(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "serdiv(fast__1)".
# ** Note: (vopt-143) Recognized 1 FSM in module "wt_dcache_ctrl(fast)".
# ** Note: (vopt-143) Recognized 1 FSM in module "axi_llc_tag_pattern_gen(fast)".
# ** Note: (vsim-12126) Error and warning message counts have been restored: Errors=0, Warnings=1.
# Loading sv_std.std
# Loading work.ara_tb_sv_unit(fast)
# Loading work.ara_tb(fast)
# Loading work.axi_pkg(fast)
# Loading work.ara_testharness(fast)
# Loading work.riscv(fast)
# Loading work.rvv_pkg(fast)
# Loading work.fpnew_pkg(fast)
# Loading work.dm(fast)
# Loading work.ariane_pkg(fast)
# Loading work.ara_pkg(fast)
# Loading work.ara_soc(fast)
# Loading work.cf_math_pkg(fast)
# Loading work.axi_xbar(fast)
# Loading work.addr_decode(fast)
# Loading work.axi_demux(fast)
# Loading work.spill_register(fast)
# Loading work.axi_demux_id_counters(fast)
# Loading work.delta_counter(fast)
# Loading work.counter(fast)
# Loading work.spill_register(fast__1)
# Loading work.spill_register(fast__2)
# Loading work.rr_arb_tree(fast)
# Loading work.spill_register(fast__3)
# Loading work.spill_register(fast__4)
# Loading work.rr_arb_tree(fast__1)
# Loading work.axi_err_slv(fast)
# Loading work.axi_atop_filter(fast)
# Loading work.stream_register(fast)
# Loading work.fifo_v2(fast)
# Loading work.fifo_v3(fast)
# Loading work.fifo_v3(fast__1)
# Loading work.fifo_v3(fast__2)
# Loading work.fifo_v3(fast__3)
# Loading work.counter(fast__1)
# Loading work.delta_counter(fast__1)
# Loading work.axi_multicut(fast)
# Loading work.axi_mux(fast)
# Loading work.axi_id_prepend(fast)
# Loading work.rr_arb_tree(fast__2)
# Loading work.fifo_v3(fast__4)
# Loading work.spill_register(fast__5)
# Loading work.spill_register(fast__6)
# Loading work.spill_register(fast__7)
# Loading work.rr_arb_tree(fast__3)
# Loading work.spill_register(fast__8)
# Loading work.spill_register(fast__9)
# Loading work.axi_atop_filter(fast__1)
# Loading work.stream_register(fast__1)
# Loading work.fifo_v2(fast__1)
# Loading work.fifo_v3(fast__5)
# Loading work.cb_filter_pkg(fast)
# Loading work.axi_llc_pkg(fast)
# Loading work.axi_llc_top(fast)
# Loading work.axi_llc_config(fast)
# Loading work.addr_decode(fast__1)
# Loading work.lzc(fast)
# Loading work.counter(fast__2)
# Loading work.delta_counter(fast__2)
# Loading work.axi_lite_regs(fast)
# Loading work.addr_decode(fast__2)
# Loading work.spill_register(fast__10)
# Loading work.spill_register(fast__11)
# Loading work.axi_isolate(fast)
# Loading work.axi_demux(fast__1)
# Loading work.spill_register(fast__12)
# Loading work.axi_demux_id_counters(fast__1)
# Loading work.delta_counter(fast__3)
# Loading work.counter(fast__3)
# Loading work.spill_register(fast__13)
# Loading work.spill_register(fast__14)
# Loading work.rr_arb_tree(fast__4)
# Loading work.spill_register(fast__15)

Fog-cake, 2021-06-02:
Hi suehtamacv,
[error.log](https://github.com/pulp-platform/ara/files/6581733/error.log)
In order to be able to simulate RTL faster, I used the 2020.1 version. But still encountered a problem. It seems that I installed 32-bit software on a 64-bit system. But I am not sure how to solve it. The complete information is in this file.

Fog-cake, 2021-06-21:
I didn't use this version of the tool in the end, so I turned it off for the time being.

closed 2021-06-21 by Fog-cake