pulp-platform/ara #45: Virtual memory support

zaazad, 2021-06-15:
Currently Ara works in the bare metal mode. So to support virtual memory and Linux in Ara, we need:

- TLB logic in Ara (addrgen module?)
- TLB miss handling-page walk
- TLB shootdown mechanism
- Page fault (and interrupt?) handling (Based on the RISC-V vector spec, precise exceptions are required for virtual memory support)
- Context switching

mp-17, 2021-06-22:
Hello zaazad,

Thanks for opening this issue: VM support is an important milestone. Currently, we are focusing our efforts on alignment with the specs + verification + benchmarking, would you like to contribute to the project helping us on implementing VM supports on Ara?

dpetrisko, 2023-08-08:
Hi, curious if this ever became supported. Thanks!

mp-17, 2023-08-16:
Hello @dpetrisko,
What a timing! We are actively working on it right now (virtual memory + OS support). It's not ready yet, but if you are also working on it, we can definitely sync.

mp-17, 2024-07-02:
Addressing this with https://github.com/pulp-platform/ara/pull/311 :-)

closed 2024-07-02 by mp-17