pulp-platform/ara #49: On the implementation of FPGA

Fog-cake, 2021-06-25:
Excuse me, do you have plans to release a fpga example project? This can actually run more programs better than simulation. PULP has a related FPGA project, but you seem to use a lot of other peripheral ip,

Fog-cake, 2021-06-30:
I am currently trying to implement your design on FPGA. It is found that some function functions, such as idx_width, under different Nrlane parameters, will not match the data bit width declared by you. For example, what you declare is a logic signal, but the idx_width may be limited by the bit width. Even if it is 1bit, the vivado software will report an error.

mp-17, 2021-06-30:
Hello Fog-cake,

Thanks a lot for the inputs! We will have a deeper look at it. An FPGA project for Ara is planned, but not in the immediate future. Nevertheless, if you are interested, we would really appreciate your contribution and thoughts on this!

fantasysee, 2021-10-11:
Thank you for your patient and warm reply, @suehtamacv and @niwis . 

Best Regards,
Chao

Fog-cake, 2021-10-11:
Hi, @niwis

Can I reduce the required FPGA resources by configuring parameters such as vlen? Thank you.

Jett-tu, 2022-01-05:
hello
Which files is the top module if I want to implementation of Ara on an FPGA?

hossein1387, 2022-09-19:
@Jett-tu You can  follow up on this PR https://github.com/pulp-platform/ara/pull/146

closed 2023-11-28 by mp-17