pulp-platform/ara #53: Run xtensor

davidbrochart, 2021-07-19:
[xtensor](https://github.com/xtensor-stack/xtensor) is a C++ library for high-performance ND array computing. Among other things, it implements lazy evaluation, which performs loop fusion and avoids temporaries. So it is well suited to be run on a vector/SIMD CPU such as RISC-V with Vector extension.
[xsimd](https://github.com/xtensor-stack/xsimd) is an abstraction layer around various SIMD architectures (Intel, Arm...), and xtensor supports xsimd to accelerate computations. We would like to support RISC-V Vector extension in xsimd, and ara is one of the only open-source projects that I know of which implements the Vector extension.
Do you think it is ready to be used for that purpose? We will need to evaluate performances, is the RTL model the only cycle-accurate model, or are there higher-level behavioral models?

suehtamacv, 2021-07-20:
Salut David,

Regarding the cycle-accurate models we have available, the RTL is indeed the only one. This being said, we use Verilator, which is reasonably fast for performance measurement. As you can see in Ara's CI, the Verilator model of Ara with four lanes simulates 3k cycles per second. I have a student project for the next semester where we will try to improve this further---we already know that the verilated `lzc` module is particularly slow, for example---but depending on the size of the algorithms you want to benchmark, this might suffice. 

Ara does not implement the whole RVV Extension: we lack some hard-to-implement instructions such as reductions and vector register-gather, or even fixed point. But Ara does implement a large part of the specification, as you can see in the `FUNCTIONALITIES.md` file, and we are working on adding the missing instructions. From this point of view, although you might not be able to currently run some benchmarks on Ara, I believe it already supports enough of them to extract some performance results. Reductions are the biggest missing chunk on Ara right now, but @mp-17 is working on that in the [feat.reduction](https://github.com/pulp-platform/ara/tree/feat.reduction) branch.

Ara also currently does not support virtual memory and has an extremely simple runtime. We basically provide a `printf`. From what I saw from `xsimd`, it seems that it can be used in bare-metal mode. Is that correct? If you check our example kernels, you can see that all of our input matrices are [statically allocated](https://github.com/pulp-platform/ara/blob/17c4e28524ce6d805dc07c84aa26ce549cf0c2ce/apps/fmatmul/main.c#L49-L51). There might be work needed there if `xsimd` needs runtime functions that we do not implement, e.g., `malloc`.

Matheus

davidbrochart, 2021-07-20:
Hi Matheus!
Thanks for your answer. Actually xtensor might need to allocate memory for the rare occasions when it needs temporaries, but we could provide custom memory allocators so that memory is eventually statically allocated.
I was also thinking about starting with [Spike](https://github.com/riscv/riscv-isa-sim) only, just to develop the RISC-V target of xsimd, but I guess we won't be able to get any performance metrics out of it.
Anyways, I'll let you know how it goes, thanks again!

closed 2021-09-30 by suehtamacv