pulp-platform/ara #56: Verilator error - importing from missing package cf_math_pkg

yanghao, 2021-07-26:
I hit a verilator error while working on the latest ara code base, and this is different from the other closed verilator issues.

Verilator version are the same as specified in top level Makefile:
```
../install/verilator/bin/verilator --version
Verilator 4.210 2021-07-07 rev v4.210
```

```
 $ make verilate
Makefile:62: "Specified QuestaSim version (questa-2020.1) not found in PATH /home/yanghao/.local/bin:/home/yanghao/bin:/usr/local/cuda-11.1/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/yanghao/bin"
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 > build/verilator/bender_script_default
/mnt/work/yanghao/pulp/ara/install/verilator/bin/verilator -f build/verilator/bender_script_default           \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LATCH                                                                    \
  -Wno-LITENDIAN                                                                \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  --hierarchical                                                                \
  tb/verilator/waiver.vlt                                                       \
  --Mdir build/verilator                                                       \
  -Itb/dpi                                                                      \
  --compiler clang                                                              \
  -CFLAGS "-DTOPLEVEL_NAME=ara_tb_verilator"                                        \
  -CFLAGS "-DNR_LANES=4"                                              \
  -CFLAGS -I/mnt/work/yanghao/pulp/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/mnt/work/yanghao/pulp/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/mnt/work/yanghao/pulp/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  ""                                                             \
  -LDFLAGS "-lelf"                                                              \
  ""                                                              \
  --exe                                                                         \
  /mnt/work/yanghao/pulp/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /mnt/work/yanghao/pulp/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /mnt/work/yanghao/pulp/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /mnt/work/yanghao/pulp/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
%Error-PKGNODECL: /mnt/work/yanghao/pulp/ara/hardware/deps/axi/src/axi_dw_downsizer.sv:56:10: Package/class 'cf_math_pkg' not found, and needs to be predeclared (IEEE 1800-2017 26.3)
   56 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
                  ... For error description see https://verilator.org/warn/PKGNODECL?v=4.210
%Error: /mnt/work/yanghao/pulp/ara/hardware/deps/axi/src/axi_dw_downsizer.sv:56:10: Importing from missing package 'cf_math_pkg'
   56 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/deps/axi/src/axi_dw_upsizer.sv:56:10: Importing from missing package 'cf_math_pkg'
   56 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/deps/cva6/src/acc_dispatcher.sv:51:10: Importing from missing package 'cf_math_pkg'
   51 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/lane/lane_sequencer.sv:11:68: Importing from missing package 'cf_math_pkg'
   11 | module lane_sequencer import ara_pkg::*; import rvv_pkg::*; import cf_math_pkg::idx_width; #(
      |                                                                    ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/lane/operand_queue.sv:11:67: Importing from missing package 'cf_math_pkg'
   11 | module operand_queue import ara_pkg::*; import rvv_pkg::*; import cf_math_pkg::idx_width; #(
      |                                                                   ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/lane/operand_requester.sv:75:10: Importing from missing package 'cf_math_pkg'
   75 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/masku/masku.sv:46:10: Importing from missing package 'cf_math_pkg'
   46 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/sldu/sldu.sv:41:10: Importing from missing package 'cf_math_pkg'
   41 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/vlsu/addrgen.sv:47:10: Importing from missing package 'cf_math_pkg'
   47 |   import cf_math_pkg::ceil_div;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/vlsu/vldu.sv:52:10: Importing from missing package 'cf_math_pkg'
   52 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/vlsu/vstu.sv:60:10: Importing from missing package 'cf_math_pkg'
   60 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/lane/valu.sv:46:10: Importing from missing package 'cf_math_pkg'
   46 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/lane/vmfpu.sv:48:10: Importing from missing package 'cf_math_pkg'
   48 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: /mnt/work/yanghao/pulp/ara/hardware/src/ara.sv:48:10: Importing from missing package 'cf_math_pkg'
   48 |   import cf_math_pkg::idx_width;
      |          ^~~~~~~~~~~
%Error: Exiting due to 15 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
make: *** [Makefile:146: build/verilator/Vara_tb_verilator] Error 1
```

suehtamacv, 2021-07-26:
Could you please attach your `build/script/compile_script_default` file? 

Moreover, can you find the `hardware/deps/common_cells/src/cf_math_pkg.sv` file?

yanghao, 2021-07-26:
I cannot find the compile_script_default file:

```
ls hardware/build/
verilator  work-dpi
```

And yes, the cf_math_pkg.sv exists, and there are two:

```
$ find hardware/ -iname "cf_math_pkg.sv"
hardware/deps/cva6/src/common_cells/src/cf_math_pkg.sv
hardware/deps/cva6/src/fpu/src/common_cells/src/cf_math_pkg.sv
```

yanghao, 2021-07-26:
I manually checked out files in common_cells and now it works! somehow all the files in that folder were deleted and even added into git cache ... I think it could due to during the clone process it was interrupted due to proxy issue. Thanks a lot for the hint!

closed 2021-07-26 by yanghao