pulp-platform/ara #58: Error 77 on verilator compilation

mohammadhosein1997, 2021-08-08:
When I try to make verilator according to the README file, I face this error:
```
$ make verilator 
# Checkout the right version
cd /home/hpc-user/ara/toolchain/verilator && git reset --hard && git fetch && git checkout v4.210
HEAD is now at 8e2ba6a0 Version bump
HEAD is now at 8e2ba6a0 Version bump
# Compile verilator
cd /home/hpc-user/ara/toolchain/verilator && git clean -xfdf && autoconf && \
CC=clang CXX=clang++ CXXFLAGS="" LDFLAGS="" \
	./configure --prefix=/home/hpc-user/ara/install/verilator && make -j8 && make install
Removing autom4te.cache/
Removing config.log
Removing configure
configuring for Verilator 4.210 2021-07-07
checking whether to perform partial static linking of Verilator binary... yes
checking whether to use tcmalloc... check
checking whether to use -m32... no
checking whether to build for coverage collection... no
checking whether to use hardcoded paths... yes
checking whether to show and stop on compilation warnings... no
checking whether to run long tests... no
checking for gcc... clang
checking whether the C compiler works... no
configure: error: in `/home/hpc-user/ara/toolchain/verilator':
configure: error: C compiler cannot create executables
See `config.log' for more details
make: *** [Makefile:144: /home/hpc-user/ara/install/verilator] Error 77
```
Here is the error I face in the config.log:
```
configure:2647: clang -V >&5
clang-13: error: argument to '-V' is missing (expected 1 value)
clang-13: error: no input files
configure:2658: $? = 1
configure:2647: clang -qversion >&5
clang-13: error: unknown argument '-qversion'; did you mean '--version'?
clang-13: error: no input files
configure:2658: $? = 1
configure:2678: checking whether the C compiler works
configure:2700: clang       conftest.c  >&5
ld.lld: error: unable to find library -lc
ld.lld: error: unable to find library -lm
ld.lld: error: unable to find library -lclang_rt.builtins-riscv64
clang-13: error: ld.lld command failed with exit code 1 (use -v to see invocation)
configure:2704: $? = 1
configure:2742: result: no
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "Verilator"
| #define PACKAGE_TARNAME "verilator"
| #define PACKAGE_VERSION "4.210 2021-07-07"
| #define PACKAGE_STRING "Verilator 4.210 2021-07-07"
| #define PACKAGE_BUGREPORT "https://verilator.org"
| #define PACKAGE_URL "https://verilator.org"
| #define PACKAGE_VERSION_NUMBER_STRING "4.210"
| /* end confdefs.h.  */
| 
| int
| main ()
| {
| 
|   ;
|   return 0;
| }
configure:2747: error: in `/home/hpc-user/ara/toolchain/verilator':
configure:2749: error: C compiler cannot create executables
See `config.log' for more details
```

mp-17, 2021-08-08:
Hello mohammadhosein1997,

It seems that the problem is your `clang` compiler: I guess you need a newer version of it. 
I would try to update it and then retry.
When you have installed the newer `clang` and `clang++`, you can run the make command specifying the new path to the executables with the following command:

`CLANG_CC=${PATH_TO_CLANG} CLANG_CXX=${PATH_TO_CLANG++} make verilator`

In the main Makefile you can also see some of the flags that can be passed to the compilers when compiling verilator.
Let us know if the problem is still there.

Best regards,
Matteo

mohammadhosein1997, 2021-08-09:
I installed clang-12, and the problem is solved. 
Thanks Matteo.

closed 2021-08-09 by mohammadhosein1997