pulp-platform/ara #59: Verilator Simulation Error

mohammadhosein1997, 2021-08-09:
When I run this command:
```
~/ara/hardware$make apply-patches 
```
I face this error: 
```
Makefile:62: "Specified QuestaSim version (questa-2020.1) not found in PATH /home/hpc-user/xilinx/Vivado/2016.2/bin:/home/hpc-user/intelFPGA_pro/21.2/modelsim_ase/bin:/usr/bin/sbt:/home/hpc-user/riscv-gnu-toolchain/build/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
cd deps/tech_cells_generic && git apply ../../patches/0001-tech-cells-generic-sram.patch
error: patch failed: src/rtl/tc_sram.sv:124
error: src/rtl/tc_sram.sv: patch does not apply
make: *** [Makefile:101: apply-patches] Error 1
```

I ignored this error, and I ran the next one: 
```
~/ara/hardware$make verilate
```
Again, I face this error:
```
Makefile:62: "Specified QuestaSim version (questa-2020.1) not found in PATH /home/hpc-user/xilinx/Vivado/2016.2/bin:/home/hpc-user/intelFPGA_pro/21.2/modelsim_ase/bin:/usr/bin/sbt:/home/hpc-user/riscv-gnu-toolchain/build/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
mkdir -p build
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 > build/verilator/bender_script_default
bash: ./bender: No such file or directory
make: *** [Makefile:145: build/verilator/Vara_tb_verilator] Error 127
make: *** Waiting for unfinished jobs....
Successfully installed bender 0.21.0 in '/home/hpc-user/ara/hardware'.
bender 0.21.0 available.
```
The Second Time I run this command, face:
```
Makefile:62: "Specified QuestaSim version (questa-2020.1) not found in PATH /home/hpc-user/xilinx/Vivado/2016.2/bin:/home/hpc-user/intelFPGA_pro/21.2/modelsim_ase/bin:/usr/bin/sbt:/home/hpc-user/riscv-gnu-toolchain/build/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 > build/verilator/bender_script_default
/home/hpc-user/ara/install/verilator/bin/verilator -f build/verilator/bender_script_default           \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LATCH                                                                    \
  -Wno-LITENDIAN                                                                \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  --hierarchical                                                                \
  tb/verilator/waiver.vlt                                                       \
  --Mdir build/verilator                                                       \
  -Itb/dpi                                                                      \
  --compiler clang                                                              \
  -CFLAGS "-DTOPLEVEL_NAME=ara_tb_verilator"                                        \
  -CFLAGS "-DNR_LANES=4"                                              \
  -CFLAGS -I/home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  ""                                                             \
  -LDFLAGS "-lelf"                                                              \
  ""                                                              \
  --exe                                                                         \
  /home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/hpc-user/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed /home/hpc-user/ara/install/verilator/bin/verilator_bin -f build/verilator/bender_script_default -GNrLanes=4 -O3 -Wno-BLKANDNBLK -Wno-CASEINCOMPLETE -Wno-CMPCONST -Wno-LATCH -Wno-LITENDIAN -Wno-UNOPTFLAT -Wno-UNPACKED -Wno-UNSIGNED -Wno-WIDTH -Wno-WIDTHCONCAT --hierarchical tb/verilator/waiver.vlt --Mdir build/verilator -Itb/dpi --compiler clang -CFLAGS -DTOPLEVEL_NAME=ara_tb_verilator -CFLAGS -DNR_LANES=4 -CFLAGS -I/home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -CFLAGS -I/home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -CFLAGS -I/home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -LDFLAGS -lelf  --exe /home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/dpi_memutil.cc /home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/sv_scoped.cc /home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/verilator_memutil.cc /home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.cc /home/hpc-user/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc /home/hpc-user/ara/hardware/tb/verilator/ara_tb.cpp --cc --top-module ara_tb_verilator
make: *** [Makefile:146: build/verilator/Vara_tb_verilator] Error 255
```

The version of the Verilator is 4.210.

suehtamacv, 2021-08-09:
Hi Hosein,

Wow, there are a lot of errors happening there. We will need to tackle them one by one. 

First, regarding the patches. You are supposed to run this only once. Have you run `make apply-patches` before? Please show me the status of the `deps/tech_cells_generic` submodule. What is the output of `git status` in that folder? What about the result of `git diff`? 

Matheus

mohammadhosein1997, 2021-08-09:
First, I ran the `make apply-patches`, but I faced the mentioned error.([Makefile:101: apply-patches] Error )
the output of `git status ` int the `deps/tech_cells_generic` directory is: 
```
HEAD detached at 203038f
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   src/rtl/tc_sram.sv

no changes added to commit (use "git add" and/or "git commit -a")
```
the output of `git diff` is:
```
index 53530e0..4bd3846 100644
--- a/src/rtl/tc_sram.sv
+++ b/src/rtl/tc_sram.sv
@@ -124,9 +124,11 @@ module tc_sram #(
   // write memory array
   always_ff @(posedge clk_i or negedge rst_ni) begin
     if (!rst_ni) begin
+      `ifndef VERILATOR
       for (int unsigned i = 0; i < NumWords; i++) begin
         sram[i] <= init_val[i];
       end
+      `endif
       for (int i = 0; i < NumPorts; i++) begin
         r_addr_q[i] <= {AddrWidth{1'b0}};
         // initialize the read output register for each port
@@ -149,12 +151,14 @@ module tc_sram #(
       for (int unsigned i = 0; i < NumPorts; i++) begin
         if (req_i[i]) begin
           if (we_i[i]) begin
+            `ifndef VERILATOR
             // update value when write is set at clock
             for (int unsigned j = 0; j < DataWidth; j++) begin
               if (be_i[i][j/ByteWidth]) begin
                 sram[addr_i[i]][j] <= wdata_i[i][j];
               end
             end
+            `endif
           end else begin
             // otherwise update read address for subsequent non request cycles
             r_addr_q[i] <= addr_i[i];
@@ -164,6 +168,23 @@ module tc_sram #(
     end // if !rst_ni
   end
 
+  `ifdef VERILATOR
+  for (genvar i = 0; i < NumPorts; i++) begin
+    // update value when write is set at clock
+    for (genvar j = 0; j < DataWidth; j++) begin
+      always_ff @(posedge clk_i or negedge rst_ni) begin
+        if (!rst_ni) begin
+        end else begin
+          if (req_i[i])
+            if (we_i[i])
+              if (be_i[i][j/ByteWidth])
+                sram[addr_i[i]][j] <= wdata_i[i][j];
+        end
+      end
+    end
+  end
+  `endif
+
 // Validate parameters.
 // pragma translate_off
 `ifndef VERILATOR
@@ -204,4 +225,59 @@ module tc_sram #(
 `endif
 `endif
 // pragma translate_on
+
+  // Copyright lowRISC contributors.
+  // Licensed under the Apache License, Version 2.0, see LICENSE for details.
+  // SPDX-License-Identifier: Apache-2.0
+
+  /**
+   * Memory loader for simulation
+   *
+   * Include this file in a memory primitive to load a memory array from
+   * simulation.
+   *
+   * Requirements:
+   * - A memory array named `sram`.
+   * - A parameter `DataWidth` giving the memory width (word size) in bit.
+   * - A parameter `NumWords` giving the memory depth in words.
+   */
+
+  `ifndef SYNTHESIS
+  // Task for loading 'sram' with SystemVerilog system task $readmemh()
+  export "DPI-C" task simutil_memload;
+
+  task simutil_memload;
+    input string file;
+    $readmemh(file, sram);
+  endtask
+
+  // Function for setting a specific element in |sram|
+  // Returns 1 (true) for success, 0 (false) for errors.
+  export "DPI-C" function simutil_set_mem;
+  function int simutil_set_mem(input int index, input bit [511:0] val);
+    // Function will only work for memories <= 512 bits
+    if (DataWidth > 512)
+      return 0;
+    if (index >= NumWords)
+      return 0;
+
+    sram[index] = val[DataWidth-1:0];
+    return 1;
+  endfunction
+
+  // Function for getting a specific element in |sram|
+  export "DPI-C" function simutil_get_mem;
+  function int simutil_get_mem(input int index, output bit [511:0] val);
+    // Function will only work for memories <= 512 bits
+    if (DataWidth > 512)
+      return 0;
+    if (index >= NumWords)
+      return 0;
+
+    val                = 0;
+    val[DataWidth-1:0] = sram[index];
+    return 1;
+  endfunction
+  `endif
+
 endmodule
```

suehtamacv, 2021-08-09:
Hi,

Based on the diff, you already applied the patches, which means that `make apply-patches` worked. You received the error because you tried to run it once again, and the patches could not be applied.

Regarding the second issue, when you run `make verilate`, I am not sure what is happening. This command works in the CI without any issues. I wonder if you do not have a problem in your environment... I know that you were using a different version of `clang` yesterday, to compile verilator. Are you using the same version of clang that you used to compile Verilator to compile the Ara files?

kuopinghsu, 2021-08-10:
I have the same issue with Verilator showing internal fault. Verilator is compiled with clang-12. If I change to clang-11, the issue is resolved.

mohammadhosein1997, 2021-08-10:
Thank you. 
clang-11 did the trick.

kuopinghsu, 2021-09-26:
Notes: Verilator has been resolved this issue in https://github.com/verilator/verilator/issues/3148.

suehtamacv, 2021-09-28:
Great! As soon as Verilator release a new tagged version containing this fix I will update the version we use in this repository.

PR #83 (2021-11-02): Synthesis improvements
This PR brings a lot of small fixes to ensure that we can reach a high frequency target. Most of the fixes are related to the interface between the lane and the system, which have some `in2out` paths that had to be removed manually.

Closes #59.
Closes #68.

## Changelog

### Changed

- Retime address calculation of the `addrgen`
- Push `MASKU` operand muxing from the lanes to the Mask Unit
- Reduce CVA6's default cache size
- Update Verilator to `v4.214`
- Update bender to `v0.23.1`

## Checklist

- [x] Automated tests pass
- [x] Changelog updated
- [x] Code style guide [... cut]
closed 2021-11-02 by suehtamacv