pulp-platform/ara #60: Unable to dump waveform using Verilator

kuopinghsu, 2021-08-13:
I got core dump when dump waveform in Verilator.

Here is my patch.

```
diff --git a/hardware/Makefile b/hardware/Makefile
index 7a4c666..175fbfa 100644
--- a/hardware/Makefile
+++ b/hardware/Makefile
@@ -176,19 +176,20 @@ $(veril_library)/V$(veril_top): $(config_file) Makefile ../Bender.yml $(shell fi
   $(ROOT_DIR)/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
   $(ROOT_DIR)/tb/verilator/ara_tb.cpp                                           \
   --cc                                                                          \
+  $(if $(DEBUG),--trace-fst -Wno-INSECURE,)                                     \
   --top-module $(veril_top) &&                                                  \
 	cd $(veril_library) && OBJCACHE='' make -j4 -f V$(veril_top).mk
 
 # Simulation
 .PHONY: simv
 simv:
-	$(veril_library)/V$(veril_top) -l ram,$(app_path)/$(app),elf
+	$(veril_library)/V$(veril_top) $(if $(DEBUG),-t,) -l ram,$(app_path)/$(app),elf
 
 .PHONY: riscv_tests_simv
 riscv_tests_simv: $(tests)
 
 $(tests): rv%: $(app_path)/rv%
-	$(veril_library)/V$(veril_top) -l ram,$<,elf &> $(buildpath)/$@.trace
+	$(veril_library)/V$(veril_top) $(if $(DEBUG),-t,) -l ram,$<,elf &> $(buildpath)/$@.trace
 
 # DPIs
 .PHONY: dpi
diff --git a/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.h b/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.h
index 1d7cc7d..e6c8d58 100644
--- a/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.h
+++ b/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.h
@@ -31,10 +31,10 @@
 #define VM_TRACE 0
 #endif
 
-// VM_TRACE_FMT_FST must be set by the user when calling Verilator with
+// VM_TRACE_FST must be set by the user when calling Verilator with
 // --trace-fst. VM_TRACE is set by Verilator itself.
 #if VM_TRACE == 1
-#ifdef VM_TRACE_FMT_FST
+#ifdef VM_TRACE_FST
 #include "verilated_fst_c.h"
 #define VM_TRACE_CLASS_NAME VerilatedFstC
 #else
diff --git a/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc b/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc
index 0721d7c..fad078f 100644
--- a/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc
+++ b/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc
@@ -264,7 +264,7 @@ void VerilatorSimCtrl::PrintStatistics() const {
 }
 
 const char *VerilatorSimCtrl::GetTraceFileName() const {
-#ifdef VM_TRACE_FMT_FST
+#ifdef VM_TRACE_FST
   return "sim.fst";
 #else
   return "sim.vcd";
@@ -280,6 +280,8 @@ void VerilatorSimCtrl::Run() {
     top_->trace(tracer_, 99, 0);
   }
 
+  Trace();
+
   // Evaluate all initial blocks, including the DPI setup routines
   top_->eval();
 
@@ -288,7 +290,6 @@ void VerilatorSimCtrl::Run() {
 
   time_begin_ = std::chrono::steady_clock::now();
   UnsetReset();
-  Trace();
 
   unsigned long start_reset_cycle_ = initial_reset_delay_cycles_;
   unsigned long end_reset_cycle_ = start_reset_cycle_ + reset_duration_cycles_;
```

I did the following patch,

1. Solve the core dump issue
2. Change macro VM_TRACE_FMT_FST to VM_TRACE_FST to support fst waveform dump.
3. Add DEBUG in Makefile to generate the waveform (default fst format)

Users can use the following command to generate trace for Verilator

```
$ make verilate DEBUG=1
```

Using following command to run the diag

```
$ make simv app=benchmarks DEBUG=1
```

suehtamacv, 2021-08-23:
Hi @kuopinghsu 

Thanks a lot for checking that out! I thought that traces were working out-of-the-box, to be honest, I am surprised you had to patch `lowrisc_dv_verilator_simutil_verilator`. 

Would it be possible for you to create a PR based on that? Otherwise I can do it myself, but if you did at least you would be credited as the author of the commits =) 

Matheus

kuopinghsu, 2021-08-23:
Thanks Matheus. I will create a PR.

PR #76 (2021-09-30): Dump waveforms with Verilator
This PR adds waveform dumping capabilities to the Verilator model, through the usage of the `trace=1` option. This is based on @kuopinghsu's commit and work on the PR #62, which I changed to align it with our code style. 

Closes #60.
Closes #62.

## Changelog

### Added

- Makefile option (`trace=1`) to generate waveform traces when running simulations with Verilator

## Checklist

- [ ] Automated tests pass
- [x] Changelog updated
- [x] Code style guideline is observed

Please check our [contributing guidelines](CONTRIBUTING.md) before opening a Pull Request.
closed 2021-09-30 by suehtamacv