pulp-platform/ara #69: Make  bin/hello_world failed

zhuxuanlong, 2021-09-11:
Hi,
The ligloss library cannot be found when compiling hello_world.  When I add the tool chain install directory to the path, the same problem still exists.
`
root@zte-R5300-G4-riscv:/share/zhuxuanlong/Vector_Work/ara/apps# make bin/hello_world
chmod +x /share/zhuxuanlong/Vector_Work/ara/apps/common/script/align_sections.sh
rm -f /share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld && cp /share/zhuxuanlong/Vector_Work/ara/apps/common/arch.link.ld /share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld
/share/zhuxuanlong/Vector_Work/ara/apps/common/script/align_sections.sh 4 /share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c hello_world/main.c -o hello_world/main.c.o
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/crt0.S -o common/crt0-llvm.S.o
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/printf.c -o common/printf-llvm.c.o
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/string.c -o common/string-llvm.c.o
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/serial.c -o common/serial-llvm.c.o
mkdir -p bin/
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -Iinclude -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -o bin/hello_world hello_world/main.c.o common/crt0-llvm.S.o common/printf-llvm.c.o common/string-llvm.c.o common/serial-llvm.c.o -static -nostartfiles -lm -T/share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld
ld.lld: error: unable to find library -lgloss
clang-13: error: ld command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:59: bin/hello_world] Error 1
rm hello_world/main.c.o common/string-llvm.c.o common/crt0-llvm.S.o common/printf-llvm.c.o common/serial-llvm.c.o
`

mp-17, 2021-09-21:
Hello zhuxuanlong,

`libgloss` should be in `/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/riscv64-unknown-elf/lib`. Can you please report the output of the command:
`ls /share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/riscv64-unknown-elf/lib`?

If it's not there, something went wrong during the installation of newlib.

Let us know! ;-)
Matteo

zhuxuanlong, 2021-09-22:
> Hello zhuxuanlong,
>
> `libgloss` should be in `/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/riscv64-unknown-elf/lib`. Can you please report the output of the command:
> `ls [...]

Hi mp-17,
Thank you for your reply. This file exists.

_root@zte-R5300-G4-riscv:/share/zhuxuanlong/Vector_Work/ara/apps# ls ../install/riscv-llvm/riscv64-unknown-elf/lib/
crt0.o  libc.a  libg.a  libgloss.a  libm.a  libnosys.a  libsemihost.a  libsim.a  nano.specs  nosys.specs  rv64imafdc  semihost.specs  sim.specs_

mp-17, 2021-09-22:
Okay, let's see where the linker searches for the libraries.

Can you please apply this patch to the Makefile in the `apps` directory?

You just need to download [this file](https://github.com/pulp-platform/ara/files/7210058/issue_69_0.txt) in the repository root directory, and then execute from there:

`git apply issue_69_0.txt`

Then, from the `apps` directory, you can try:

`make bin/hello_world | grep "ld.lld"`

Can you please post the output of this command here? We should see now the verbose output of the linker, with the directories that it visit when looking for the libraries.

zhuxuanlong, 2021-09-22:
> make bin/hello_world | grep "ld.lld"

Thank you. The output of the command is as follows:

`root@zte-R5300-G4-riscv:/share/zhuxuanlong/Vector_Work/ara/apps# make bin/hello_world
chmod +x /share/zhuxuanlong/Vector_Work/ara/apps/common/script/align_sections.sh
rm -f /share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld && cp /share/zhuxuanlong/Vector_Work/ara/apps/common/arch.link.ld /share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld
/share/zhuxuanlong/Vector_Work/ara/apps/common/script/align_sections.sh 4 /share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c hello_world/main.c -o hello_world/main.c.o
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/crt0.S -o common/crt0-llvm.S.o
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/printf.c -o common/printf-llvm.c.o
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/string.c -o common/string-llvm.c.o
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/serial.c -o common/serial-llvm.c.o
mkdir -p bin/
/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/clang -Iinclude -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/share/zhuxuanlong/Vector_Work/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -o bin/hello_world hello_world/main.c.o common/crt0-llvm.S.o common/printf-llvm.c.o common/string-llvm.c.o common/serial-llvm.c.o -static -nostartfiles -lm -T/share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld -v
clang version 13.0.0 (https://github.com/llvm/llvm-project.git b14c8f5f6eb96e70adb553cc8a4257524881e029)
Target: riscv64-unknown-unknown-elf
Thread model: posix
InstalledDir: /share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin
Found candidate GCC installation: /usr/lib/gcc-cross/riscv64-linux-gnu/9
Selected GCC installation: /usr/lib/gcc-cross/riscv64-linux-gnu/9
 "/share/zhuxuanlong/Vector_Work/ara/install/riscv-llvm/bin/ld.lld" -m elf64lriscv hello_world/main.c.o common/crt0-llvm.S.o common/printf-llvm.c.o common/string-llvm.c.o common/serial-llvm.c.o -lm -L/usr/lib/gcc-cross/riscv64-linux-gnu/9 -L/usr/lib/gcc-cross/riscv64-linux-gnu/9/../../../../riscv64-linux-gnu/lib -T /share/zhuxuanlong/Vector_Work/ara/apps/common/link.ld --start-group -lc -lgloss --end-group -lgcc -o bin/hello_world
ld.lld: error: unable to find library -lgloss
clang-13: error: ld command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:59: bin/hello_world] Error 1
rm hello_world/main.c.o common/string-llvm.c.o common/crt0-llvm.S.o common/printf-llvm.c.o common/serial-llvm.c.o
`

mp-17, 2021-09-22:
Your LLVM found a candidate RISC-V GCC installation on the system and it is searching for libraries there. 
I must change the options we give to LLVM when it is installed to fix this in a more elegant way, but, in the meantime, you can apply another patch to solve the problem.

Revert the patch that you applied before, from the root repository directory:
`git checkout -- apps/Makefile`

Download and apply the [new patch](https://github.com/pulp-platform/ara/files/7211697/issue_69_1.txt):
`git apply issue_69_1.txt`

Then, it should compile smoothly. Let us know!

Best,
Matteo

zhuxuanlong, 2021-09-23:
> Your LLVM found a candidate RISC-V GCC installation on the system and it is searching for libraries there.
> I must change the options we give to LLVM when it is installed to fix this in a more [...]

Hi Matteo,
The above solution can solve the helloword compilation error. Thank you again for your support.  In addition, two targets need to be modified in the apps/makefile, such as `bin/$(1)-ara-%` and `bin/$(1)-ara-%.` Otherwise, `make riscv_tests` has the same problem. 
After compilation, all executable programs under bin are as follows:

`root@zte-R5300-G4-riscv:/share/zhuxuanlong/Vector_Work/ara/ara/apps# ls bin/
hello_world                   rv64ud-ara-structural.dump  rv64ui-ara-bgeu          rv64ui-ara-simple.dump  rv64ui-ara-xori         rv64uv-ara-vfcvt.dump         rv64uv-ara-vfsub          rv64uv-ara-vmfge.dump    rv64uv-ara-vmulh             rv64uv-ara-vslidedown.dump
hello_world.dump              rv64uf-ara-fadd             rv64ui-ara-bgeu.dump     rv64ui-ara-sll          rv64ui-ara-xori.dump    rv64uv-ara-vfmacc             rv64uv-ara-vfsub.dump     rv64uv-ara-vmfgt         rv64uv-ara-vmulh.dump        rv64uv-ara-vslideup
rv64si-ara-csr                rv64uf-ara-fadd.dump        rv64ui-ara-blt           rv64ui-ara-sll.dump     rv64um-ara-div          rv64uv-ara-vfmacc.dump        rv64uv-ara-vfwadd         rv64uv-ara-vmfgt.dump    rv64uv-ara-vmulhsu           rv64uv-ara-vslideup.dump
rv64si-ara-csr.dump           rv64uf-ara-fclass           rv64ui-ara-blt.dump      rv64ui-ara-slli         rv64um-ara-div.dump     rv64uv-ara-vfmadd             rv64uv-ara-vfwadd.dump    rv64uv-ara-vmfle         rv64uv-ara-vmulhsu.dump      rv64uv-ara-vsll
rv64si-ara-dirty              rv64uf-ara-fclass.dump      rv64ui-ara-bltu          rv64ui-ara-slli.dump    rv64um-ara-divu         rv64uv-ara-vfmadd.dump        rv64uv-ara-vfwcvt         rv64uv-ara-vmfle.dump    rv64uv-ara-vmulhu            rv64uv-ara-vsll.dump
rv64si-ara-dirty.dump         rv64uf-ara-fcmp             rv64ui-ara-bltu.dump     rv64ui-ara-slliw        rv64um-ara-divu.dump    rv64uv-ara-vfmax              rv64uv-ara-vfwcvt.dump    rv64uv-ara-vmflt         rv64uv-ara-vmulhu.dump       rv64uv-ara-vsra
rv64si-ara-icache-alias       rv64uf-ara-fcmp.dump        rv64ui-ara-bne           rv64ui-ara-slliw.dump   rv64um-ara-divuw        rv64uv-ara-vfmax.dump         rv64uv-ara-vfwmacc        rv64uv-ara-vmflt.dump    rv64uv-ara-vmv               rv64uv-ara-vsra.dump
rv64si-ara-icache-alias.dump  rv64uf-ara-fcvt             rv64ui-ara-bne.dump      rv64ui-ara-sllw         rv64um-ara-divuw.dump   rv64uv-ara-vfmerge            rv64uv-ara-vfwmacc.dump   rv64uv-ara-vmfne         rv64uv-ara-vmv.dump          rv64uv-ara-vsrl
rv64si-ara-ma_fetch           rv64uf-ara-fcvt.dump        rv64ui-ara-fence_i       rv64ui-ara-sllw.dump    rv64um-ara-divw         rv64uv-ara-vfmerge.dump       rv64uv-ara-vfwmsac        rv64uv-ara-vmfne.dump    rv64uv-ara-vmxnor            rv64uv-ara-vsrl.dump
rv64si-ara-ma_fetch.dump      rv64uf-ara-fcvt_w           rv64ui-ara-fence_i.dump  rv64ui-ara-slt          rv64um-ara-divw.dump    rv64uv-ara-vfmin              rv64uv-ara-vfwmsac.dump   rv64uv-ara-vmin          rv64uv-ara-vmxnor.dump       rv64uv-ara-vss
rv64si-ara-sbreak             rv64uf-ara-fcvt_w.dump      rv64ui-ara-jal           rv64ui-ara-slt.dump     rv64um-ara-mul          rv64uv-ara-vfmin.dump         rv64uv-ara-vfwmul         rv64uv-ara-vmin.dump     rv64uv-ara-vmxor             rv64uv-ara-vss.dump
rv64si-ara-sbreak.dump        rv64uf-ara-fdiv             rv64ui-ara-jal.dump      rv64ui-ara-slti         rv64um-ara-mul.dump     rv64uv-ara-vfmsac             rv64uv-ara-vfwmul.dump    rv64uv-ara-vminu         rv64uv-ara-vmxor.dump        rv64uv-ara-vsub
rv64si-ara-scall              rv64uf-ara-fdiv.dump        rv64ui-ara-jalr          rv64ui-ara-slti.dump    rv64um-ara-mulh         rv64uv-ara-vfmsac.dump        rv64uv-ara-vfwnmacc       rv64uv-ara-vminu.dump    rv64uv-ara-vnmsac            rv64uv-ara-vsub.dump
rv64si-ara-scall.dump         rv64uf-ara-fmadd            rv64ui-ara-jalr.dump     rv64ui-ara-sltiu        rv64um-ara-mulh.dump    rv64uv-ara-vfmsub             rv64uv-ara-vfwnmacc.dump  rv64uv-ara-vmnand        rv64uv-ara-vnmsac.dump       rv64uv-ara-vwadd
rv64si-ara-wfi                rv64uf-ara-fmadd.dump       rv64ui-ara-lb            rv64ui-ara-sltiu.dump   rv64um-ara-mulhsu       rv64uv-ara-vfmsub.dump        rv64uv-ara-vfwnmsac       rv64uv-ara-vmnand.dump   rv64uv-ara-vnmsub            rv64uv-ara-vwadd.dump
rv64si-ara-wfi.dump           rv64uf-ara-fmin             rv64ui-ara-lb.dump       rv64ui-ara-sltu         rv64um-ara-mulhsu.dump  rv64uv-ara-vfmul              rv64uv-ara-vfwnmsac.dump  rv64uv-ara-vmnor         rv64uv-ara-vnmsub.dump       rv64uv-ara-vwaddu
rv64uc-ara-rvc                rv64uf-ara-fmin.dump        rv64ui-ara-lbu           rv64ui-ara-sltu.dump    rv64um-ara-mulhu        rv64uv-ara-vfmul.dump         rv64uv-ara-vfwsub         rv64uv-ara-vmnor.dump    rv64uv-ara-vnsra             rv64uv-ara-vwaddu.dump
rv64uc-ara-rvc.dump           rv64uf-ara-ldst             rv64ui-ara-lbu.dump      rv64ui-ara-sra          rv64um-ara-mulhu.dump   rv64uv-ara-vfmv               rv64uv-ara-vfwsub.dump    rv64uv-ara-vmor          rv64uv-ara-vnsra.dump        rv64uv-ara-vwmacc
[... cut ...]      rv64uv-ara-vzext.dump
rv64ud-ara-recoding.dump      rv64ui-ara-bge              rv64ui-ara-sh.dump       rv64ui-ara-xor          rv64uv-ara-vfadd.dump   rv64uv-ara-vfslide1up         rv64uv-ara-vmfeq.dump     rv64uv-ara-vmul          rv64uv-ara-vslide1up.dump
rv64ud-ara-structural         rv64ui-ara-bge.dump         rv64ui-ara-simple        rv64ui-ara-xor.dump     rv64uv-ara-vfcvt        rv64uv-ara-vfslide1up.dump    rv64uv-ara-vmfge          rv64uv-ara-vmul.dump     rv64uv-ara-vslidedown
`

mp-17, 2021-09-23:
Perfect, happy that it helped! :-)

Best,
Matteo

closed 2021-09-23 by mp-17