pulp-platform/ara #84: ctype.h

forthyen, 2021-11-08:


suehtamacv, 2021-11-08:
Hi @forthyen 

I am not sure, I never heard of this file. Do you have a more detailed log?

Matheus

forthyen, 2021-11-09:
chmod +x /mnt/d/ara/apps/common/script/align_sections.sh
/mnt/d/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/mnt/d/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c hello_world/main.c -o hello_world/main.c.o
/mnt/d/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/mnt/d/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/crt0.S -o common/crt0-llvm.S.o
/mnt/d/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/mnt/d/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/printf.c -o common/printf-llvm.c.o
/mnt/d/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/mnt/d/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/string.c -o common/string-llvm.c.o
/mnt/d/ara/install/riscv-llvm/bin/clang -march=rv64gcv0p10 -mabi=lp64d -menable-experimental-extensions -mno-relax -fuse-ld=lld -mcmodel=medany -I/mnt/d/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/serial.c -o common/serial-llvm.c.o
rm -f /mnt/d/ara/apps/common/link.ld && cp /mnt/d/ara/apps/common/arch.link.ld /mnt/d/ara/apps/common/link.ld
/mnt/d/ara/apps/common/script/align_sections.sh 4 /mnt/d/ara/apps/common/link.ld
common/string.c:1:10: fatal error: 'ctype.h' file not found
#include <ctype.h>
         ^~~~~~~~~
hello_world/main.c:20:10: fatal error: 'string.h' file not found
#include <string.h>
         ^~~~~~~~~~
1 error generated.
make: *** [/mnt/d/ara/apps/common/runtime.mk:101: common/string-llvm.c.o] Error 1
make: *** Waiting for unfinished jobs....
1 error generated.
make: *** [/mnt/d/ara/apps/common/runtime.mk:107: hello_world/main.c.o] Error 1
rm common/crt0-llvm.S.o common/printf-llvm.c.o common/serial-llvm.c.o

suehtamacv, 2021-11-09:
I think @mp-17 wrote this file. Do you know what is this header?

suehtamacv, 2021-11-09:
This being said, `string.h` is part of the standard C++ library. You should have it. Maybe you need to install some packages in your system?

mp-17, 2021-11-09:
Yes, exactly, it seems more of an environmental problem. Check if the file `/mnt/d/ara/install/riscv-llvm/riscv64-unknown-elf/include/string.h` exists. BTW, do you have the folder `/mnt/d/ara/install/riscv-llvm/riscv64-unknown-elf/include`? If not, maybe `newlib` was not compiled correctly. 
Try to check if you can make `toolchain-llvm-newlib` again, and if everything runs smoothly.

forthyen, 2021-11-10:
Thanks. I re-make toolchain-llvm and then I get back one new folder "install/riscv-llvm/riscv64-unknown-elf" !

closed 2021-11-10 by forthyen