pulp-platform/ara #85: Failed to build toolchain-llvm using Makefile

kaitoukito, 2021-11-08:
I failed to build toolchain-llvm on MacOS.
There are several similar errors, and the first FAILED log is as follows:
[image]
I'm not sure if these warning logs previous to the FAILED logs will be helpful for debugging.
[image]
[image]

mp-17, 2021-11-09:
Hello @kaitoukito,

We never tried to install the toolchain on MacOS, but we will try. 
I guess the problem here occurs when executing the `toolchain-llvm-rt` Makefile target. Is this right? So, `LLVM` and `newlib` are installed correctly.
Anyway, it would be nice to enable a `verbose` compilation to have more details about what went wrong.

Best regards,
Matteo

kaitoukito, 2021-11-09:
Thanks, I will try it on Linux through VirtualBox :)

closed 2021-11-09 by kaitoukito