pulp-platform/ara #86: vmv.s.x/vmv.x.s is supported?

yanghao, 2021-11-08:
version: main branch from Oct. 1st.

From the https://github.com/pulp-platform/ara/blob/main/FUNCTIONALITIES.md file, vmv instructions are supported, but a test shows this seems only refering to vmv.v.* instructions, not the single element move instructions?

A very common use case is to build up a list of values from scalar computation and shift those values one by one into vector registers. Without this instruction, a few more instructions would be needed to build up a mask and then use vmv.v.* instruction, however vmv.v.* instructions are encoded as unmasked instructions so this essentially cannot be done.

Thanks.

mp-17, 2021-11-09:
Hello @yanghao, 

You are right, the instruction is currently not supported, but we plan to add it soon. 
I am unsure if I got the use case: `vmv.s.x` transfers a scalar value to the element 0 of a vector register. 
If you want to pile up elements, you could maybe use `vslide1down`. You can slide down a vector of length N by one position, injecting the scalar elements from the top, one by one, until the vector is complete.

Best regards,
Matteo

yanghao, 2021-11-09:
Thanks for the info @mp-17 .

vslideup/down both source and destiation has to be vector registers, only the shifts are immediate or scalar register values. Problem is to get a value into vector register.

https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1631-vector-slideup-instructions

closed 2021-11-09 by yanghao