pulp-platform/ara #87: ara core hangs with 2 lanes and 4 lanes configuration

yanghao, 2021-11-11:
We saw the same program (binary) runs fine on 8 and 16 lanes configuration, however hangs when running on 2 lanes or 4 lanes configuration. And it is not introduced by any particular vector instruction, it seems to be a mix of scalar and vector instructions that is causing this hang. Tested on latest commits and showing also the same result.

Just wondering if this is a known issue, and do you need a minimum sequence that can reproduce this issue?

suehtamacv, 2021-11-11:
Hi @yanghao,

It is hard to say what is happening. It could be a bug, or it could be an issue with the code you are running. Could you please provide a minimum example? 

Matheus

yanghao, 2021-11-11:
@suehtamacv OK, I will try to create a simpler test that can reproduce this. So far it is a big program, but it works fine on spike and ara8/16. The very interesting thing is:

* for VL==128, only ara-2-lanes hang.
* for VL==256, ara-2-lanes and ara-4-lanes hang.

I haven't tried VL > 256 as spike right now only supports 4Kib VLEN (EEW == 16), but seems there is a pattern here, which maybe when VL==512, ara-8-lanes starts to hang and if VL==1024 ara-16-lanes will hang. Will try to provide a simple code, but could take a while to extract the essence of the instruction flow that triggered the issue.

mp-17, 2021-11-11:
Hello @yanghao,

This sounds like a bug to me, but it's hard to tell what's going on under the hood. A minimum example would be perfect and would help a lot! 
BTW, if you also want to share the source code and your implementation, we can integrate it into our tests (if it is a program that you want to open-source).

Thanks for reporting the problem,
Matteo

yanghao, 2021-11-12:
https://github.com/pulp-platform/ara/pull/88

Please check the minimal example that reproduce the hang on ara-2-lanes. Hope it helps.

mp-17, 2021-11-12:
Thanks a lot for the example!

It is a problem with the lane sequencer, which does not make the mask unit operand requester fetch enough elements in the case of masked operations. Fewer lanes and more elements highlight the bug, which depends also on your vector element width (indeed, if you use `e64` instead of `e16` it works with 2 lanes as well).
I will have a deeper look and fix it in the next few days and hopefully release a fixed version soon. Unluckily, next week will be busy, but we will patch this as soon as possible. 

Best Regards,
Matteo

closed 2022-02-10 by suehtamacv