pulp-platform/ara #89: make toolchain-gcc failed

kaitoukito, 2021-11-13:
I follow the guideline in README, and when I go to step
`make riscv_tests`

it seems that I need to make toolchain-gcc first
[image]

when I make toolchian-gcc, another problem comes
[image]

the log shows several similar errors like this "cannot stat 'xxx.gmo': No such file or directory"
[image]

How should I solve this problem?

suehtamacv, 2021-11-15:
Did you run a `git submodule --update --recursive` before compiling the toolchain? It might be the case that git did not manage to download all submodules correctly. They are pretty large. What is the output of `git status`?

kaitoukito, 2021-11-16:
> Did you run a `git submodule --update --recursive` before compiling the toolchain? It might be the case that git did not manage to download all submodules correctly. They are pretty large. What is the output of `git status`?

Firstly, I used "git submodule --update --recursive". The project is pretty large, and I waited for a whole day and it still hadn't finished. I had to use "ctrl + c" to interrupt the progress, and I restarted it again using "git submodule --update --recursive". After I repeated several times, it finally finished, and the shell didn't return any error message when I did this for the last time.

Now, when I use "git status", it seems I pull this project correctly. However, I'm not sure whether I have pulled this project recursively, as I once interrupted and restarted the command for serveral times.

hellcer, 2022-02-17:
I am facing the same problem. Please fix it or give suggestions.

suehtamacv, 2022-02-17:
`gcc` is a very large submodule, and you need to ensure that it was cloned correctly before compiling it. Check whether the output of `git status` is clean from the `riscv-gcc` folder. Sometimes `git` creates an empty repository when it is not able to clone it correctly. 

Unfortunately, there is not much I can do regarding this, the submodule is simply too large.

closed 2022-02-27 by kaitoukito

mahdi259, 2023-10-05:
Have you installed prerequisites of riscv-gnu-toolchain?

$ `sudo apt-get install autoconf automake autotools-dev curl python3 python3-pip libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev ninja-build git cmake libglib2.0-dev`

(https://github.com/riscv-collab/riscv-gnu-toolchain)