pulp-platform/ara #98: How do I segment the output waveform

little-blue, 2022-01-17:
I use the verilator on ara to simulate a benchmark. But the output sim.fst file is too large to analysis. It is almost 39GB. How can I change the ara source code to output multiple files and each of them less than for example 500MB.

mp-17, 2022-01-17:
Hi little-blue,

I don't know if you can split the trace into smaller files automatically with Verilator, but you can use the `--trace-depth` option to tune the depth of tracing, e.g. `--trace-depth 1` to dump only the top level signals. This should reduce the file size, even if it will also worsen the internal signal visibility.
Do you think this can be useful to you?

Best,
Matteo

little-blue, 2022-01-18:
Yes, thank you for answer my question, it helps me a lot.

closed 2022-01-18 by little-blue