pulp-platform/ara #99: Tracing modules with more depth

mhezarei, 2022-01-18:
Greetings,

After looking at the Verilator's tracing output with GTKWave, I realized that signals from deeper modules (such as `simd_alu`) are not present. Is it intended, or am I making a mistake? If it is normal, then how could I look for deeper modules' waveforms?

Thanks in advance.

alan2031, 2022-09-15:
> Hi @mhezarei I had the same problem. Have you found a solution? Thanks in advance.

Comment out line 7 to line 12 in hardware/tb/verilator/waiver.vlt, which blocks lanes.

mp-17, 2024-07-02:
Fixing with https://github.com/pulp-platform/ara/pull/317/commits/5cec856d417a74bd591808e8b9b1997c0e715d57 in https://github.com/pulp-platform/ara/pull/317

closed 2024-07-02 by mp-17