pulp-platform/ara #116: Bugs from integer reduction instructions

kuoyaoming93, 2022-05-30:
Hi, I'm having similar issues as described in #112, and applied the fix #114 to da92dee72b615c95e561b3836012cb27036203a1. 

In my case, the program stalls in the first iteration. Suppose the C code below.
Could you please fix this bug? Thanks in advance.

```c
#include <stdint.h>

#include "printf.h"
#include "runtime.h"

#define q_field 16

int64_t wires[q_field][q_field] = {
    {0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15},
    {1,0,5,9,15,2,11,14,10,3,8,6,13,12,7,4},
    {2,5,0,6,10,1,3,12,15,11,4,9,7,14,13,8},
    {3,9,6,0,7,11,2,4,13,1,12,5,10,8,15,14},
    {4,15,10,7,0,8,12,3,5,14,2,13,6,11,9,1},
    {5,2,1,11,8,0,9,13,4,6,15,3,14,7,12,10},
    {6,11,3,2,12,9,0,10,14,5,7,1,4,15,8,13},
    {7,14,12,4,3,13,10,0,11,15,6,8,2,5,1,9},
    {8,10,15,13,5,4,14,11,0,12,1,7,9,3,6,2},
    {9,3,11,1,14,6,5,15,12,0,13,2,8,10,4,7},
    {10,8,4,12,2,15,7,6,1,13,0,14,3,9,11,5},
    {11,6,9,5,13,3,1,8,7,2,14,0,15,4,10,12},
    {12,13,7,10,6,14,4,2,9,8,3,15,0,1,5,11},
    {13,12,14,8,11,7,15,5,3,10,9,4,1,0,2,6},
    {14,7,13,15,9,12,8,1,6,4,11,10,5,2,0,3},
    {15,4,8,14,1,10,13,9,2,7,5,12,11,6,3,0}
};

int main(void) 
{
    int i,j;
    int64_t result[q_field];
    unsigned long int block_size_p;

    // Vector configuration
    asm volatile("vsetvli %0, %1, e64, m1, ta, ma" : "=r"(block_size_p) : "r"(q_field));

    for(i=0; i<q_field;i++){
        asm volatile("vle64.v v0, (%0);" ::"r"(&wires[i][0]));      // Extract rows in v0

        // Min find
        asm volatile("vmv.v.i v2, 15;"); 
        asm volatile("vredmin.vs v4, v0, v2;"); 

        // Store result
        asm volatile("vse64.v v4, (%0);" ::"r"(result));      // Extract rows in v0

        // Print minimum
        printf("Minimum of row [%d]: %d\n", i, result[0]);
    }
}
```

The output from Verilator 4.214 2021-10-17 rev v4.214:
[image]

mp-17, 2022-05-30:
Hello @kuoyaoming93,

I will have a look at it, thanks a lot for reporting! :-)

Best,
Matteo

mp-17, 2022-07-09:
Hello Yao-Ming, 

It should be fixed with https://github.com/pulp-platform/ara/pull/123.

Best,
Matteo

closed 2022-08-02 by mp-17