pulp-platform/ara #161: Trouble generating Vector Trace

StyleDiablo, 2022-10-31:
I am having trouble getting the vector trace generator to work.I tried to compile with the given command but nothing.I tried 
make bin/${rv64uv-ara-vxor}.ideal for example but all is returning is : No rule to make target 'bin/ara-vxor.ideal'.  Stop.
Can you help me please?

quswar-abid, 2022-11-01:
Can you share more detail on it? Command line logs etc. Anything related to it. Especially the commands you used to run it.

quswar-abid, 2022-11-01:
Try adding the rv64uv prefix if you are doing some stuff manually like this: bin/ara-vxor.ideal ---> bin/ **rv64uv-** ara-vxor.ideal

StyleDiablo, 2022-11-03:
I tried the command that was given , meaning: **make bin/${program}.ideal** but with rv64uv-ara-vxor where the program is.
I tried it with and without the {} or without the dollar sign $ but it keeps giving me the same error, Something along tghe lines of 
**make: *** No rule to make target 'bin/ara-vxor.ideal'.  Stop.**
my exact commands are :
**make bin/${rv64uv-ara-vxor}.ideal** or
**make bin/$rv64uv-ara-vxor.ideal**,
**make bin/rv64uv-ara-vxor.ideal**
I dont understand what am I doing wrong.

StyleDiablo, 2022-11-03:
What about the vector trace?

quswar-abid, 2022-11-03:
If you are talking about the waveform you need to make verilate with trace=1 and waveform will be generated whenever you set the trace flag while doing make simv ...

The command would be like `make verilate trace=1` for building rtl and

```
make simv app=* xyz * trace=1
```

for generation of waveform in file named `sim.fst`

Btw did the previous solution work?

StyleDiablo, 2022-11-04:
Yes I am talking about the wavefrom but i was under the impression that if you use the ideal dispatcher it produces a waveform with just the vector signals am I wrong?

quswar-abid, 2022-11-04:
Not sure about the "ideal dispatcher" but the whole hierarchy is traced when you set the flag. Scalar core etc. And everything is recorded ..

StyleDiablo, 2022-11-04:
Ok thank you

quswar-abid, 2022-11-04:
If your query is fully addressed, consider closing this issue.

Thanks, :)

mp-17, 2024-08-26:
Hello, when in `ideal-dispatcher` mode, the scalar core is replaced with a FIFO pre-loaded with the vector instructions to execute. The FIFO issues these vector instructions to Ara. To generate the trace, you should use Verilator's tracing capabilities or simulate with a simulator like QuestaSim.

closed 2024-08-26 by mp-17