pulp-platform/ara #162: Test failures in Vector Integer Arithmetic Instructions

quswar-abid, 2022-11-01:
## Test failures in Vector Integer Arithmetic Instructions

#### Description:

Tests for the unit testing have been added and necessary environment changes have been made to run the tests on Spike and ARA. Following tests are found to be failing on the RTL. The tests can be verified by running on Spike using the usual Make targets in ara/apps (using`make riscv_tests_spike`)

In this issue, vector integer arithmetic instructions related failure have been reported. Other tests are also available.

#### Failing combinations to be debugged:

### VADD

```
vadd.vv_LMUL2SEW8
vadd.vi_LMUL2SEW8
vadd.vx_LMUL2SEW8
vadd.vi_LMUL1SEW8
vadd.vx_LMUL1SEW8
vadd.vv_LMUL1SEW8
vadd.vi_LMUL4SEW16
vadd.vx_LMUL4SEW16
vadd.vv_LMUL4SEW16
vadd.vi_LMUL4SEW8
vadd.vv_LMUL4SEW8
vadd.vx_LMUL4SEW8
vadd.vx_LMUL2SEW16
vadd.vi_LMUL2SEW16
vadd.vv_LMUL2SEW16
vadd.vi_LMUL8SEW32
vadd.vx_LMUL8SEW32
vadd.vx_LMUL8SEW16
vadd.vv_LMUL8SEW32
vadd.vi_LMUL8SEW16
vadd.vv_LMUL8SEW16
vadd.vx_LMUL8SEW8
vadd.vi_LMUL8SEW8
vadd.vv_LMUL8SEW8
vadd.vi_LMUL4SEW32
vadd.vx_LMUL4SEW32
vadd.vv_LMUL4SEW32
vadd.vx_LMUL8SEW64
vadd.vi_LMUL8SEW64
vadd.vv_LMUL8SEW64
```

### VRSUB

```
vrsub.vx_LMUL2SEW8
vrsub.vi_LMUL2SEW8
vrsub.vi_LMUL1SEW8
vrsub.vx_LMUL1SEW8
vrsub.vi_LMUL4SEW16
vrsub.vx_LMUL4SEW16
vrsub.vx_LMUL4SEW8
vrsub.vi_LMUL4SEW8
vrsub.vx_LMUL2SEW16
vrsub.vi_LMUL2SEW16
vrsub.vi_LMUL8SEW32
vrsub.vx_LMUL8SEW32
vrsub.vx_LMUL8SEW16
vrsub.vi_LMUL8SEW16
vrsub.vi_LMUL8SEW8
vrsub.vx_LMUL8SEW8
vrsub.vi_LMUL4SEW32
vrsub.vx_LMUL4SEW32
vrsub.vx_LMUL8SEW64
vrsub.vi_LMUL8SEW64
```

[... 1733 lines cut ...]

#### Steps to recreate this issue:

1. `git clone https://github.com/pulp-platform/ara.git`
2. `git remote add test_repo "https://github.com/10x-Engineers/ara"`
3. `git fetch test_repo`
4. `git checkout main_verif_10x`
5. `cd apps`
6. `make riscv_tests` 
7. `cd ../hardware`
8. make simv app="name of individual ELF" or `make riscv_tests_simv -j4` to run regression

### (Optional) Steps to setup the pre-built `riscv-gcc` toolchain:

Download `riscv-gcc` toolchain from [this release](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2022.06.10/riscv64-elf-ubuntu-20.04-nightly-2022.06.10-nightly.tar.gz), then go to ara top directory and follow the steps below:

- update the submodules recursively (`git submodule update --recursive`)
- `mkdir install && cd install`
- extract the downloaded toolchain here and rename the folder to `riscv-gcc`
- `cd riscv-gcc` (you should see bin/ include/ and other folders)
- go back to ara top directory
- `cd apps && make riscv_tests`
- `cd ../hardware && make apply-patches && make verilate`
  - if you get an error here, go to `ara/hardware/deps/cva6` and update the submodules recursively again (`git submodule update --recursive`)
- `make simv app=rv64uv-ara-vadd` to verify the setup

mp-17, 2022-11-03:
Amazing! Thanks a lot for the huge work. We will start fixing as soon as possible.

mp-17, 2022-11-08:
Unluckily the pre-compiled toolchain was clashing with some of our system shared libraries; I re-compiled the last `gcc` from the `main` branch, and it works now.

I hypothesize that these tests are "hardcoded" for `VLEN == 4096`. So, I tried with Ara 4 lanes and `VLEN == 4096`.

I looked at the `vadd.vi_LMUL1SEW8` test to see what is failing.
I think there is a problem with the test; for example, the test composed of the sub-tests from 386 to 449 fails. 
With `sew == 8 bit` and `VLEN == 4096`, `vlmax` is 512 elements with `LMUL == 1`. 

1) When you load the mask vector with this line:
https://github.com/10x-Engineers/ara/blob/main_verif_10x/apps/riscv-tests/isa/rv64uv/vadd.vi_LMUL1SEW8.S#L553
you are trying to load 512 B from the memory, even if the memory is initialized for only 32 B from the `mask` data section:
https://github.com/10x-Engineers/ara/blob/main_verif_10x/apps/riscv-tests/isa/rv64uv/vadd.vi_LMUL1SEW8.S#L1214
The content of the memory after those 32 B depends depends on the other files / linker script, so the executable compiled for Spike will differ from the one for Ara.

2) Since the mask is specified only for 256 bits (32 B), the rest of the mask is undefined / unpredictable. Maybe Spike works since at compile time `gcc` padded the memory after the last valid mask bits with zeroes.
For this reason, the first 256 results from Ara are correct. Then, the last 256 do not corresponds to the "golden" ones.

Maybe also the other tests have the same issue, but I have not checked yet.

quswar-abid, 2022-11-09:
Hi @mp-17 thanks for the response.

The `gcc` related part really was an optional one. And you assumed correctly about the configuration. I'll update on other filed issues as well that with which config ARA was built (i.e. `nrLanes==4` and `VLEN==4096`)

What you have pointed out is correct about the failing tests. Infact, this problem is same with other tests as well. I have changed the mask and golden reference values as well by properly regenerating the tests. I have observed that now, after regeneration, all tests related to VADD instructions pass successfully. I am sure it will have impacted other tests too, so I will be updating the issue text detail soon to communicate it properly. I will then inform again (in this thread) when updated.

P.S. The new tests generated will have a mask for 4096 elements (which is a case with `VLEN=4096`, `SEW=8`, and `LMUL=8`) This way a mask for max. number of elements will be available and only required will be loaded and utilized in different tests.

Quswar,

quswar-abid, 2022-11-11:
@mp-17 Here is an update:

I have modified the tests to have sufficient mask for all cases. The test has been generated again to have renewed golden values. Following are few of the instructions that are passing in all of its combinations (in available tests) and the rest of them are failing in few or all of their combinations. I'll update when I go through the rest of them later. Meanwhile, I'll be removing the following instructions from the issue since they are working fine.

### Instructions working correctly:
```
vadd
vrsub
vsub
adc
vsbc
vand
vor
xor
vsll
vsra
vmax
maxu
vmin
vminu
vmul
vmulh
vmulhsu
vmulhu
vdivu
vwmul
vwmulu
vwmulsu
vmacc
vmadd
vnmsac
vnmsub
vmerge
vremu
```

Thanks,
Quswar

quswar-abid, 2022-11-13:
Hi @mp-17 I have updated the list of failing and hanging tests in [the issue](https://github.com/pulp-platform/ara/issues/162#issue-1431418034) while the list of passing tests has already been [shared](https://github.com/pulp-platform/ara/issues/162#issuecomment-1311275442).

mp-17, 2022-11-13:
Thank you very much @quswar-abid! I will try to look at the new tests asap! :-)