pulp-platform/ara #164: Test failures in Vector Floating-point Arithmetic Instructions

quswar-abid, 2022-11-03:
## Test failures in Vector Floating-point Arithmetic Instructions

#### Description:

Tests for the unit testing have been added and necessary environment changes have been made to run the tests on Spike and ARA. Following tests are found to be failing on the RTL. The tests can be verified by running on Spike using the usual Make targets in ara/apps (using `make riscv_tests_spike`)

In this issue, vector floating-point arithmetic instructions related failure have been reported. Other tests are also available. Each of these combination is a test file in `ara/apps/riscv-tests/isa/rv64uv/` and is also listed in its `Makefrag`.

#### Failing tests:

### VFADD

```
vfadd.vf_LMUL8SEW32
vfadd.vv_LMUL8SEW32
vfadd.vv_LMUL4SEW32
vfadd.vf_LMUL4SEW32
vfadd.vf_LMUL8SEW64
vfadd.vv_LMUL8SEW64
```

### VFSUB

```
vfsub.vf_LMUL8SEW32
vfsub.vv_LMUL8SEW32
vfsub.vf_LMUL4SEW32
vfsub.vv_LMUL4SEW32
vfsub.vf_LMUL8SEW64
vfsub.vv_LMUL8SEW64
```

### VFWADD

```
vfwadd.wv_LMUL1SEW32
vfwadd.vf_LMUL1SEW32
vfwadd.vv_LMUL1SEW32
vfwadd.vf_LMUL2SEW32
vfwadd.wv_LMUL2SEW32
vfwadd.vv_LMUL2SEW32
vfwadd.vf_LMUL4SEW32
vfwadd.vv_LMUL4SEW32
vfwadd.wv_LMUL4SEW32
vfwadd.wf_LMUL1SEW32
vfwadd.wf_LMUL2SEW32
vfwadd.wf_LMUL4SEW32
```

### VFWSUB

```
vfwsub.vf_LMUL1SEW32
vfwsub.wv_LMUL1SEW32
vfwsub.vv_LMUL2SEW32
vfwsub.vf_LMUL2SEW32
vfwsub.vv_LMUL1SEW32
vfwsub.wv_LMUL2SEW32
vfwsub.vf_LMUL4SEW32
vfwsub.wv_LMUL4SEW32
vfwsub.vv_LMUL4SEW32
vfwsub.wf_LMUL1SEW32
vfwsub.wf_LMUL4SEW32
vfwsub.wf_LMUL2SEW32
```

### VFMUL

```
vfmul.vf_LMUL8SEW32
[... 601 lines cut ...]

#### Steps to recreate this issue:

1. `git clone https://github.com/pulp-platform/ara.git`
2. `git remote add test_repo "https://github.com/10x-Engineers/ara"`
3. `git fetch test_repo`
4. `git checkout main_verif_10x`
5. `cd apps`
6. `make riscv_tests` 
7. `cd ../hardware`
8. make simv app="name of individual ELF" or `make riscv_tests_simv -j4` to run regression

### (Optional) Steps to setup the pre-built `riscv-gcc` toolchain:

Download `riscv-gcc` toolchain from [this release](https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2022.06.10/riscv64-elf-ubuntu-20.04-nightly-2022.06.10-nightly.tar.gz), then go to ara top directory and follow the steps below:

- update the submodules recursively (`git submodule update --recursive`)
- `mkdir install && cd install`
- extract the downloaded toolchain here and rename the folder to `riscv-gcc`
- `cd riscv-gcc` (you should see bin/ include/ and other folders)
- go back to ara top directory
- `cd apps && make riscv_tests`
- `cd ../hardware && make apply-patches && make verilate`
  - if you get an error here, go to `ara/hardware/deps/cva6` and update the submodules recursively again (`git submodule update --recursive`)
- `make simv app=rv64uv-ara-vadd` to verify the setup

quswar-abid, 2022-11-21:
Issue updated with failing/hanging status of available tests.

Shafi10x, 2023-04-17:
The issue has been updated with the failing or hanging status of available tests.Additionally, some new tests have been added that are failing due to updates in RTL.

**vfrdiv**

Failing:

```
vfrdiv.vf_LMUL1SEW32
vfrdiv.vf_LMUL1SEW64
vfrdiv.vf_LMUL2SEW32
vfrdiv.vf_LMUL2SEW64
vfrdiv.vf_LMUL4SEW32
vfrdiv.vf_LMUL4SEW64
vfrdiv.vf_LMUL8SEW32
vfrdiv.vf_LMUL8SEW64
```

**vmfeq**

Failing:

```
vmfeq.vf_LMUL2SEW32
vmfeq.vf_LMUL4SEW32
vmfeq.vf_LMUL8SEW32
vmfeq.vf_LMUL8SEW64
vmfeq.vv_LMUL2SEW32
vmfeq.vv_LMUL4SEW32
vmfeq.vv_LMUL8SEW32
vmfeq.vv_LMUL8SEW64
```

Hanging:

```
vmfeq.vf_LMUL1SEW64.trace:0
vmfeq.vf_LMUL2SEW64.trace:0
vmfeq.vf_LMUL4SEW64.trace:0
vmfeq.vv_LMUL1SEW64.trace:0
vmfeq.vv_LMUL2SEW64.trace:0
vmfeq.vv_LMUL4SEW64.trace:0
```

**vmfge**

Failing:

```
vmfge.vf_LMUL2SEW32
vmfge.vf_LMUL4SEW32
vmfge.vf_LMUL8SEW32
vmfge.vf_LMUL8SEW64
```

Hanging:

```
vmfge.vf_LMUL1SEW64.trace:0
vmfge.vf_LMUL2SEW64.trace:0
vmfge.vf_LMUL4SEW64.trace:0
```

**vmfgt**

Failing:

```
vmfgt.vf_LMUL2SEW32
vmfgt.vf_LMUL4SEW32
vmfgt.vf_LMUL8SEW32
vmfgt.vf_LMUL8SEW64
```

Hanging:

```
vmfgt.vf_LMUL1SEW64.trace:0
vmfgt.vf_LMUL2SEW64.trace:0
vmfgt.vf_LMUL4SEW64.trace:0
```

**vmfle**

Failing:

```
vmfle.vf_LMUL2SEW32
vmfle.vf_LMUL4SEW32
vmfle.vf_LMUL8SEW32
vmfle.vf_LMUL8SEW64
vmfle.vv_LMUL2SEW32
vmfle.vv_LMUL4SEW32
vmfle.vv_LMUL8SEW32
vmfle.vv_LMUL8SEW64
```

Hanging:

```
vmfle.vf_LMUL1SEW64.trace:0
vmfle.vf_LMUL2SEW64.trace:0
vmfle.vf_LMUL4SEW64.trace:0
vmfle.vv_LMUL1SEW64.trace:0
vmfle.vv_LMUL2SEW64.trace:0
vmfle.vv_LMUL4SEW64.trace:0
```

**vmflt**

Failing:

```
vmflt.vf_LMUL2SEW32
vmflt.vf_LMUL4SEW32
vmflt.vf_LMUL8SEW32
vmflt.vf_LMUL8SEW64
vmflt.vv_LMUL2SEW32
vmflt.vv_LMUL4SEW32
vmflt.vv_LMUL8SEW32
vmflt.vv_LMUL8SEW64
```

Hanging:

```
vmflt.vf_LMUL1SEW64.trace:0
vmflt.vf_LMUL2SEW64.trace:0
vmflt.vf_LMUL4SEW64.trace:0
vmflt.vv_LMUL1SEW64.trace:0
vmflt.vv_LMUL2SEW64.trace:0
vmflt.vv_LMUL4SEW64.trace:0
```

**vmfne**

Failing:

```
vmfne.vf_LMUL2SEW32
vmfne.vf_LMUL4SEW32
vmfne.vf_LMUL8SEW32
vmfne.vf_LMUL8SEW64
vmfne.vv_LMUL2SEW32
vmfne.vv_LMUL4SEW32
vmfne.vv_LMUL8SEW32
vmfne.vv_LMUL8SEW64
```

Hanging:

```
vmfne.vf_LMUL1SEW64.trace:0
vmfne.vf_LMUL2SEW64.trace:0
vmfne.vf_LMUL4SEW64.trace:0
vmfne.vv_LMUL1SEW64.trace:0
vmfne.vv_LMUL2SEW64.trace:0
vmfne.vv_LMUL4SEW64.trace:0
```

**vfmv**

Failing:

```
vfmv.s.f_LMUL1SEW32
vfmv.s.f_LMUL2SEW32
vfmv.s.f_LMUL4SEW32
vfmv.s.f_LMUL8SEW32
vfmv.v.f_LMUL1SEW32
vfmv.v.f_LMUL2SEW32
vfmv.v.f_LMUL4SEW32
vfmv.v.f_LMUL8SEW32
```

**vfslide**

Failing:

```
vfslide1down.vf_LMUL1SEW32
vfslide1down.vf_LMUL1SEW64
vfslide1down.vf_LMUL2SEW32
vfslide1down.vf_LMUL2SEW64
vfslide1down.vf_LMUL4SEW32
vfslide1down.vf_LMUL4SEW64
vfslide1down.vf_LMUL8SEW32
vfslide1down.vf_LMUL8SEW64
vfslide1up.vf_LMUL1SEW32
vfslide1up.vf_LMUL1SEW64
vfslide1up.vf_LMUL2SEW32
vfslide1up.vf_LMUL2SEW64
vfslide1up.vf_LMUL4SEW32
vfslide1up.vf_LMUL4SEW64
vfslide1up.vf_LMUL8SEW32
vfslide1up.vf_LMUL8SEW64
```

**vfsqrt**

Failing:

```
vfsqrt.v_LMUL1SEW32
vfsqrt.v_LMUL2SEW32
vfsqrt.v_LMUL4SEW32
vfsqrt.v_LMUL8SEW32
```