pulp-platform/ara #169: [VCS] How to run ara on VCS (or irun) ?

Heresyrac, 2022-11-04:
I am trying to move ara on VCS, but met too many errors.

jin8495, 2023-01-18:
Hello, do you have any progression?

jamesningd, 2023-02-11:
I also want to run it with VCS. Did you success to run it with VCS?

takeshineshiro, 2023-02-20:
use this makefile content like below to replace the makefile in haredware directory:   
       # Copyright 2021 ETH Zurich and University of Bologna.
# Solderpad Hardware License, Version 0.51, see LICENSE for details.
# SPDX-License-Identifier: SHL-0.51
#
# Author: Matheus Cavalcante <matheusd@iis.ee.ethz.ch>

SHELL = /usr/bin/env bash
ROOT_DIR := $(patsubst %/,%, $(dir $(abspath $(lastword $(MAKEFILE_LIST)))))
ARA_DIR := $(shell git rev-parse --show-toplevel 2>/dev/null || echo $$ARA_DIR)
#INSTALL_DIR := $(abspath $(ROOT_DIR)/../install)
INSTALL_DIR :=/project/STPU2/wanghaisheng/ara/install

VERILATOR_INCLUDE := $(INSTALL_DIR)/verilator/share/verilator/include/vltstd

# Choose Ara's configuration
ifndef config
	ifdef ARA_CONFIGURATION
		config := $(ARA_CONFIGURATION)
	else
		config := default
	endif

endif

# Include configuration
config_file := $(ROOT_DIR)/../config/$(config).mk
include $(abspath $(ROOT_DIR)/../config/$(config).mk)

# Clang flags for Verilator command
ifneq (${CLANG_PATH},)
	CLANG_CXXFLAGS := -CFLAGS "-nostdinc++ -isystem $(CLANG_PATH)/include/c++/v1"
	CLANG_LDFLAGS  := -LDFLAGS "-L $(CLANG_PATH)/lib -Wl,-rpath,$(CLANG_PATH)/lib -lc++ -nostdlib++"
else
	CLANG_CXXFLAGS := ""
	CLANG_LDFLAGS  := ""
endif

export ARA_ROOT=$(abspath `pwd`/../../)

#vcs  env 
export SNPS_AMBA_VIP_HOME=/nfs54/EDA/vip/snps_axi4
export SNPS_AMBA_VIP_HOME=/nfs54/project/SoCVIP/snps_vip_svtb
export UVM_HOME=/EDA/synopsys/vcs-mx/N-2017.12-SP2/etc/uvm-1.2
export VERDI_HOME=/EDA/synopsys/verdi/Q-2020.03-SP2-4
MODE     = 
COV_OPT += line+fsm+tgl+cond

NR_LANES=4 
VLEN=4096

ifeq ( ${ARA_CONFIGURATION},4_lanes) 
    NR_LANES=4 
    VLEN=4096
else ifeq(${ARA_CONFIGURATION},2_lanes)
     NR_LANES=4 
     VLEN=2048
else ifeq(${ARA_CONFIGURATION},8_lanes) 
      NR_LANES=8
      VLEN=8192
else ifeq(${ARA_CONFIGURATION},16_lanes)
       NR_LANES=16
       VLEN=16384
else 
      NR_LANES=4 
      VLEN=4096  
  
endif

#-ntb_opts uvm -full64 
[... 290 lines cut ...]
SNPS_SG ?= spyglass-2022.06
lint: spyglass/tmp/files spyglass/sdc/func.sdc spyglass/scripts/run_lint.tcl
	cd spyglass; $(SNPS_SG) sg_shell -tcl scripts/run_lint.tcl

spyglass/tmp/files: $(bender)
	mkdir -p spyglass/tmp
	./bender script verilator -t rtl -t spyglass -t cva6_test $(bender_defs) --define SPYGLASS > spyglass/tmp/files

# DPIs
.PHONY: dpi
dpi: $(buildpath)/$(dpi_library)/ara_dpi.so

$(buildpath)/$(dpi_library)/%.o: tb/dpi/%.cc
	mkdir -p $(buildpath)/$(dpi_library)
	$(CXX) -shared -fPIC -std=c++11 -Bsymbolic -c $< -I$(VERILATOR_INCLUDE) -I$(INSTALL_DIR)/riscv-isa-sim/include -o $@

$(buildpath)/$(dpi_library)/ara_dpi.so: $(dpi)
	mkdir -p $(buildpath)/$(dpi_library)
	$(CXX) -shared -m64 -o $(buildpath)/$(dpi_library)/ara_dpi.so $?

# Clean targets
.PHONY: clean
clean:
	rm -rf $(buildpath)
	#rm -f bender

takeshineshiro, 2023-02-20:
also   the  content ofrtl_cmod.f   is  like  below:
    +incdir+${ARA_ROOT}

${ARA_ROOT}/hardware/deps/tech_cells_generic/src/rtl/tc_sram.sv
${ARA_ROOT}/hardware/deps/tech_cells_generic/src/deprecated/cluster_clk_cells.sv
${ARA_ROOT}/hardware/deps/tech_cells_generic/src/deprecated/pulp_clk_cells.sv
${ARA_ROOT}/hardware/deps/tech_cells_generic/src/rtl/tc_clk.sv
${ARA_ROOT}/hardware/deps/tech_cells_generic/src/deprecated/pulp_clock_gating_async.sv

+incdir+${ARA_ROOT}/hardware/deps/common_cells
+incdir+${ARA_ROOT}/hardware/deps/common_cells/include
${ARA_ROOT}/hardware/deps/common_cells/src/binary_to_gray.sv
${ARA_ROOT}/hardware/deps/common_cells/src/cb_filter_pkg.sv
${ARA_ROOT}/hardware/deps/common_cells/src/cdc_2phase.sv
${ARA_ROOT}/hardware/deps/common_cells/src/cf_math_pkg.sv
${ARA_ROOT}/hardware/deps/common_cells/src/clk_div.sv
${ARA_ROOT}/hardware/deps/common_cells/src/delta_counter.sv
${ARA_ROOT}/hardware/deps/common_cells/src/ecc_pkg.sv
${ARA_ROOT}/hardware/deps/common_cells/src/edge_propagator_tx.sv
${ARA_ROOT}/hardware/deps/common_cells/src/exp_backoff.sv
${ARA_ROOT}/hardware/deps/common_cells/src/fifo_v3.sv
${ARA_ROOT}/hardware/deps/common_cells/src/gray_to_binary.sv
${ARA_ROOT}/hardware/deps/common_cells/src/isochronous_spill_register.sv
${ARA_ROOT}/hardware/deps/common_cells/src/lfsr.sv
${ARA_ROOT}/hardware/deps/common_cells/src/lfsr_16bit.sv
${ARA_ROOT}/hardware/deps/common_cells/src/lfsr_8bit.sv
${ARA_ROOT}/hardware/deps/common_cells/src/mv_filter.sv
${ARA_ROOT}/hardware/deps/common_cells/src/onehot_to_bin.sv
${ARA_ROOT}/hardware/deps/common_cells/src/plru_tree.sv
${ARA_ROOT}/hardware/deps/common_cells/src/popcount.sv
${ARA_ROOT}/hardware/deps/common_cells/src/rr_arb_tree.sv
${ARA_ROOT}/hardware/deps/common_cells/src/rstgen_bypass.sv
${ARA_ROOT}/hardware/deps/common_cells/src/serial_deglitch.sv
${ARA_ROOT}/hardware/deps/common_cells/src/shift_reg.sv
${ARA_ROOT}/hardware/deps/common_cells/src/spill_register_flushable.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_demux.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_filter.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_fork.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_intf.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_join.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_mux.sv
${ARA_ROOT}/hardware/deps/common_cells/src/sub_per_hash.sv
${ARA_ROOT}/hardware/deps/common_cells/src/sync.sv
${ARA_ROOT}/hardware/deps/common_cells/src/sync_wedge.sv
${ARA_ROOT}/hardware/deps/common_cells/src/unread.sv
${ARA_ROOT}/hardware/deps/common_cells/src/addr_decode.sv
${ARA_ROOT}/hardware/deps/common_cells/src/cb_filter.sv
${ARA_ROOT}/hardware/deps/common_cells/src/cdc_fifo_2phase.sv
${ARA_ROOT}/hardware/deps/common_cells/src/counter.sv
${ARA_ROOT}/hardware/deps/common_cells/src/ecc_decode.sv
${ARA_ROOT}/hardware/deps/common_cells/src/ecc_encode.sv
${ARA_ROOT}/hardware/deps/common_cells/src/edge_detect.sv
${ARA_ROOT}/hardware/deps/common_cells/src/lzc.sv
${ARA_ROOT}/hardware/deps/common_cells/src/max_counter.sv
${ARA_ROOT}/hardware/deps/common_cells/src/rstgen.sv
${ARA_ROOT}/hardware/deps/common_cells/src/spill_register.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_delay.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_fifo.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_fork_dynamic.sv
${ARA_ROOT}/hardware/deps/common_cells/src/cdc_fifo_gray.sv
${ARA_ROOT}/hardware/deps/common_cells/src/fall_through_register.sv
${ARA_ROOT}/hardware/deps/common_cells/src/id_queue.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_to_mem.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_arbiter_flushable.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_register.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_xbar.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_arbiter.sv
${ARA_ROOT}/hardware/deps/common_cells/src/stream_omega_net.sv
${ARA_ROOT}/hardware/deps/common_cells/src/deprecated/clock_divider_counter.sv
${ARA_ROOT}/hardware/deps/common_cells/src/deprecated/find_first_one.sv
[... 245 lines cut ...]
${ARA_ROOT}/hardware/src/masku/masku.sv
${ARA_ROOT}/hardware/src/sldu/sldu.sv
${ARA_ROOT}/hardware/src/vlsu/addrgen.sv
${ARA_ROOT}/hardware/src/vlsu/vldu.sv
${ARA_ROOT}/hardware/src/vlsu/vstu.sv
${ARA_ROOT}/hardware/src/lane/operand_queues_stage.sv
${ARA_ROOT}/hardware/src/lane/valu.sv
${ARA_ROOT}/hardware/src/lane/vmfpu.sv
${ARA_ROOT}/hardware/src/lane/fixed_p_rounding.sv
${ARA_ROOT}/hardware/src/vlsu/vlsu.sv
${ARA_ROOT}/hardware/src/lane/vector_fus_stage.sv
${ARA_ROOT}/hardware/src/lane/lane.sv
${ARA_ROOT}/hardware/src/ara.sv
${ARA_ROOT}/hardware/src/ara_system.sv
${ARA_ROOT}/hardware/src/ara_soc.sv
${ARA_ROOT}/hardware/tb/ara_testharness.sv
${ARA_ROOT}/hardware/tb/ara_tb.sv
${ARA_ROOT}/hardware/src/accel_dispatcher_ideal.sv
${ARA_ROOT}/hardware/tb/ara_tb_verilator.sv

+incdir+${ARA_ROOT}/hardware/deps/cva6/src/util
+incdir+${ARA_ROOT}/hardware/deps/cva6/include
${ARA_ROOT}/hardware/deps/cva6/include/instr_tracer_pkg.sv
${ARA_ROOT}/hardware/deps/cva6/src/util/instr_tracer_if.sv
${ARA_ROOT}/hardware/deps/cva6/src/util/instr_tracer.sv

lczzz29, 2023-07-16:
@takeshineshiro hello may you update your modified makefile for VCS and share a file link (or as an attachment) plz? your above answers are unreadable because of the text format used in github issues (and maybe recognized as markdown syntax incorrectly). thx.