pulp-platform/ara #196: Can't run a single simulation of any kind

StyleDiablo, 2022-12-20:
Hello I am trying to run some rtl simulations with verilator but i can't. When I run `make riscv_tests_simv` it crashes and it's giving me `make: *** [Makefile:214: rv64uv-ara-vaadd] Error 139`. When I run `app=rv64uv-ara-vadd make simv` or `app=hello_world make simv` it gives me `make: *** [Makefile:208: simv] Segmentation fault (core dumped)` .
Can anyone please help I don't know what to do.
Thank you.

mp-17, 2022-12-20:
Hello @StyleDiablo,
Are you using the Verilator installed from this repository, or another one? 
In the meantime, try with this commit https://github.com/pulp-platform/ara/pull/195/commits/175a04200d31c201f4bdccc0166d5a519a6cd05e and re-install Verilator. 
Does it get better?

mp-17, 2023-02-13:
Following in https://github.com/pulp-platform/ara/issues/197

closed 2023-02-13 by mp-17