pulp-platform/ara #197: Failing to generate wavefrom

StyleDiablo, 2022-12-21:
The problem i am experiencing is that everytime I compile the hardware with the command **trace=1** to generate waveform traces the program compiles but after that when I run the simulation with a random binary loaded the program crashes with the message `make: *** [Makefile:208: simv] Segmentation fault (core dumped)` . If I dont use the **trace=1** the program runs fine. I also have a problem generating the vector trace. I am not sure what exact command am i supposed to use. The insrtuctions say to compile a program like this `make bin/${program}.ideal` but I cant figure out what the "{program}" is supposed to be. Is it a binary like **rv64uv-ara-vxor** or something else? 
Can anyone please help with these problems?
Thank you.

mp-17, 2023-02-13:
Hello @StyleDiablo, also other users reported a similar problem. What we saw is that the version of Verilator can be determinant. Which version are you using now, the default one of the repo? Have you already tried to compile the latest `verilator` version? I will try as well and let you know!

Best,
Matteo

A-Eren, 2023-08-17:
Have same problem with Verilator 4.214 2021-10-17 rev v4.214

A-Eren, 2023-08-21:
I downloaded my own verilator 5. Changed hardware's make file with it's direction then compiled. It made it right with make verilate command. I traced the direction to be sure. But still same problem, also this problem prevent to observe mutidimensional arrays. It problaly dumps while tracing vrf and its signals.

A-Eren, 2023-08-21:
[image]
Here is my computer's features, I am running ubuntu at hdd

mp-17, 2023-09-12:
We merged the commit with Verilator 5 in `main`. 
Any luck with it?

llhe110, 2023-12-18:
Dear  @StyleDiablo,
I also encountered this problem, how do you deal with it now?

```
llhe@llhe-virtual-machine:~/rscratch/workspace/ara/ara/hardware$ app=hello_world make simv trace=1
Makefile:84: "Specified QuestaSim version (questa-2021.2) not found in PATH /home/llhe/rscratch/cadtools/QuestaSim2021.2.1Linux/install/questasim/RUVM_2021.2:/home/llhe/rscratch/cadtools/QuestaSim2021.2.1Linux/install/questasim/linux_x86_64:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/llhe/rscratch/cadtools/interlijIDEA/idea-IC-232.10227.8/bin"
build/verilator/Vara_tb_verilator -t -l ram,/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world,elf
Program header number 0 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' low is 80000000
Program header number 0 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' high is 80000fbd
Program header number 1 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' high is 8000139f
Program header number 2 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' high is 800013c7
Program header number 3 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' is not of type PT_LOAD; ignoring.
Set `ram TOP.ara_tb_verilator.dut.i_ara_soc.i_dram 10 0x80000000 0x100000 write with offset: 0x0 write with size: 0x13c8
Simulation of Ara
=================

Tracing can be toggled by sending SIGUSR1 to this process:
$ kill -USR1 15d8
make: *** [Makefile:232: simv]
```

DuzaBF, 2024-12-21:
Have the same issue with getting traces with Verilator on commit  8fc5ebd17d04a8e4608a987834884985d27dea2b . Was able to resolve it by verilating without optimization, changing the `-O3` to `-O0` in the `hardware/Makefile`
```diff
diff --git a/hardware/Makefile b/hardware/Makefile
index 1f1c8c76..5619f803 100644
--- a/hardware/Makefile
+++ b/hardware/Makefile
@@ -189,7 +189,7 @@ $(veril_library)/V$(veril_top): $(config_file) Makefile ../Bender.yml $(shell fi
        $(veril_path)/verilator -f $(veril_library)/bender_script_$(config)           \
   -GNrLanes=$(nr_lanes)                                                         \
   -GVLEN=$(vlen)                                                                \
-  -O3                                                                           \
+  -O0                                                                          \
   --hierarchical \
   -Wno-fatal                                                                    \
   -Wno-PINCONNECTEMPTY                                                          \

```