pulp-platform/ara #213: [toolchain] newlib path is broken

jin8495, 2023-02-11:
Hi, I tried to clone this repo and build it.
But git couldn't update newlib.

So I fixed `ara/.gitmodules` 
from
```
...
[submodule "toolchain/newlib"]
	path = toolchain/newlib
	url = https://sourceware.org/git/newlib-cygwin.git
	ignore = dirty
...
```
to
```
...
[submodule "toolchain/newlib"]
	path = toolchain/newlib
	url = https://github.com/bminor/newlib.git
	ignore = dirty
...
```

and it seems like working now.

Anyway, thanks for your hard work. cheers. 👏👏

jin8495, 2023-02-11:
While I am building toolchain, I realized that the submodules of riscv-gnu-toolchain are also broken.
I think this will be resolved with a different riscv-gnu-toolchain commit.

mp-17, 2023-02-13:
Hello @jin8495,

Thank you for reporting :-) 
Which is the error message that you get? Just to understand if it's a problem with your setup or with the dependencies.

Best,
Matteo

jin8495, 2023-02-13:
@mp-17 

This is the case for the newlib. 
```shell
jin8495@DESKTOP:~/tmp/ara/toolchain$ git submodule update --init toolchain/newlib/
Submodule 'toolchain/newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Cloning into '/home/jin8495/tmp/ara/toolchain/newlib'...
fatal: unable to access 'https://sourceware.org/git/newlib-cygwin.git/': server certificate verification failed. CAfile: none CRLfile: none
fatal: clone of 'https://sourceware.org/git/newlib-cygwin.git' into submodule path '/home/jin8495/tmp/ara/toolchain/newlib' failed
Failed to clone 'toolchain/newlib'. Retry scheduled
Cloning into '/home/jin8495/tmp/ara/toolchain/newlib'...
fatal: unable to access 'https://sourceware.org/git/newlib-cygwin.git/': server certificate verification failed. CAfile: none CRLfile: none
fatal: clone of 'https://sourceware.org/git/newlib-cygwin.git' into submodule path '/home/jin8495/tmp/ara/toolchain/newlib' failed
Failed to clone 'toolchain/newlib' a second time, aborting
```

This is the case for the riscv-gnu-toolchain. Cloning works for the riscv-gnu-toolchain, but git failed to clone the submodules of the riscv-gnu-toolchain.
```shell
jin8495@DESKTOP:~/tmp/ara/toolchain$ git submodule update --init riscv-gnu-toolchain/
Submodule 'toolchain/riscv-gnu-toolchain' (https://github.com/riscv/riscv-gnu-toolchain.git) registered for path 'riscv-gnu-toolchain'
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain'...
Submodule path 'riscv-gnu-toolchain': checked out '5842fde8ee5bb3371643b60ed34906eff7a5fa31'
jin8495@DESKTOP:~/tmp/ara/toolchain$ cd riscv-gnu-toolchain
jin8495@DESKTOP:~/tmp/ara/toolchain/riscv-gnu-toolchain$ git submodule update --init *
Submodule 'qemu' (https://git.qemu.org/git/qemu.git) registered for path 'qemu'
Submodule 'riscv-binutils' (https://github.com/riscv/riscv-binutils-gdb.git) registered for path 'riscv-binutils'
Submodule 'riscv-dejagnu' (https://github.com/riscv/riscv-dejagnu.git) registered for path 'riscv-dejagnu'
Submodule 'riscv-gcc' (https://github.com/riscv/riscv-gcc.git) registered for path 'riscv-gcc'
Submodule 'riscv-gdb' (https://github.com/riscv/riscv-binutils-gdb.git) registered for path 'riscv-gdb'
Submodule 'riscv-glibc' (https://github.com/riscv/riscv-glibc.git) registered for path 'riscv-glibc'
Submodule 'riscv-newlib' (https://github.com/riscv/riscv-newlib.git) registered for path 'riscv-newlib'
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/qemu'...
fatal: unable to access 'https://git.qemu.org/git/qemu.git/': server certificate verification failed. CAfile: none CRLfile: none
fatal: clone of 'https://git.qemu.org/git/qemu.git' into submodule path '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/qemu' failed
Failed to clone 'qemu'. Retry scheduled
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/riscv-binutils'...
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/riscv-dejagnu'...
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/riscv-gcc'...
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/riscv-gdb'...
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/riscv-glibc'...
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/riscv-newlib'...
Cloning into '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/qemu'...
fatal: unable to access 'https://git.qemu.org/git/qemu.git/': server certificate verification failed. CAfile: none CRLfile: none
fatal: clone of 'https://git.qemu.org/git/qemu.git' into submodule path '/home/jin8495/tmp/ara/toolchain/riscv-gnu-toolchain/qemu' failed
Failed to clone 'qemu' a second time, aborting
```
```shell
jin8495@DESKTOP:~/tmp/ara/toolchain/riscv-gnu-toolchain$ ls *
LICENSE  Makefile.in  README.md  configure  configure.ac

linux-headers:
include

qemu:

regression:
Makefile

riscv-binutils:

riscv-dejagnu:

riscv-gcc:

riscv-gdb:

riscv-glibc:

riscv-newlib:

scripts:
cp_s  spike.exp  testsuite-filter  wrapper

test:
benchmarks  whitelist
```

And for the riscv-isa-sim, I thought git could not clone either. But, it is working fine.

mp-17, 2023-02-13:
I think that it's a problem with verifying the certificates.
Can this help?

https://stackoverflow.com/questions/21181231/server-certificate-verification-failed-cafile-etc-ssl-certs-ca-certificates-c/67698986#67698986

jin8495, 2023-02-13:
Oh yes, it works fine.
I thought it is a problem with the submodule's URL, but it is not.
Thank you for the help.

closed 2023-02-13 by jin8495