pulp-platform/ara #221: In vfwadd, subnormal numbers and zero are not handled

Shafi10x, 2023-03-15:
In **vfwadd** and other widening arithmetic instructions, the number is first converted to a double width format. It seems that during the conversion to double-width, implementations does not handle subnormal numbers and zero properly. This could potentially lead to unexpected results or errors in certain computations that involve these special cases. Therefore, it's important to ensure that the conversion process includes proper handling of subnormal numbers and zero.
**Example:**

If a number is zero in single-precision floating-point format, it should also be zero when converted to double-precision floating-point format but implementation changes the number zero.

mp-17, 2023-04-17:
Thanks @Shafi10x,

I will have a look at this! Very much appreciated!

Shafi10x, 2023-04-17:
@mp-17 Welcome, I have created PR for this issue https://github.com/pulp-platform/ara/pull/222.