pulp-platform/ara #223: ld.lld: error: unable to find library -lgloss

pigen01, 2023-04-07:
when I tried `make bin/hello_world` 
return
```
/home/<user_name>/Documents/ara/install/riscv-llvm/bin/clang -Iinclude -march=rv64gcv_zfh_zvfh0p1 -menable-experimental-extensions -mabi=lp64d -mno-relax -fuse-ld=lld -fno-vectorize -mllvm -scalable-vectorization=off -mllvm -riscv-v-vector-bits-min=0 -Xclang -target-feature -Xclang +no-optimized-zero-stride-load -mcmodel=medany -I/home/<user_name>/Documents/ara/apps/common -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -ffunction-sections -fdata-sections -o bin/hello_world hello_world/data.S.o hello_world/main.c.o common/crt0-llvm.S.o common/printf-llvm.c.o common/string-llvm.c.o common/serial-llvm.c.o common/util-llvm.c.o -static -nostartfiles -lm -Wl,--gc-sections -T/home/yepeijun/Documents/ara/apps/common/link.ld -v                 clang version 16.0.0 (https://github.com/llvm/llvm-project.git 5047104a76445bb424e480594f0eb2ed768315b3)                                                                                                           Target: riscv64-unknown-unknown-elf                                                                                                                                                                                Thread model: posix
InstalledDir: /home/<user_name>/Documents/ara/install/riscv-llvm/bin
Found candidate GCC installation: /usr/lib/gcc-cross/riscv64-linux-gnu/10
Found candidate GCC installation: /usr/lib/gcc/riscv64-unknown-elf/8.3.0                                                                                                                                           Selected GCC installation: /usr/lib/gcc-cross/riscv64-linux-gnu/10
 "/home/<user_name>/Documents/ara/install/riscv-llvm/bin/ld.lld" -m elf64lriscv -X hello_world/data.S.o hello_world/main.c.o common/crt0-llvm.S.o common/printf-llvm.c.o common/string-llvm.c.o common/serial-llvm.c.o common/util-llvm.c.o -lm --gc-sections -L/usr/lib/gcc-cross/riscv64-linux-gnu/10 -L/usr/lib/gcc-cross/riscv64-linux-gnu/10/../../../../riscv64-linux-gnu/lib -T /home/yepeijun/Documents/ara/apps/common/link.ld --start-group -lc -lgloss --end-group -lgcc -o bin/hello_world
ld.lld: error: unable to find library -lgloss
clang-16: error: ld command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:105: bin/hello_world] Error 1
```
I also had tried to debug as the closed issue, but it seems not work for me. 

I guess this is because the clang find the wrong library (should change `$$RISCV_LDFLAGS_GCC` according to the Makefile) but have no idea how to do that.

mp-17, 2023-04-17:
Hello @pigen01,

Is `/home/<user_name>/Documents/ara/install/riscv-llvm/bin/clang` an existing path? Or maybe `<user_name>` is a mistake?

pigen01, 2023-05-04:
Hello @mp-17 
 
Yes, it is an existing path, but it is not the default path for clang to find riscv-toolchain library,  so I add "-L" option to avoid the wrong linking to another library.

closed 2023-05-04 by pigen01