pulp-platform/ara #228: Issue in vector single-width averaging add and subtract instructions: wrong results

SoufSilence, 2023-05-16:
In vector single-width averaging add and subtract instructions, it seems that the code of `vaadd`, `vasubu`, and `vasub` give wrong results.

For `vaadd`: 
https://github.com/pulp-platform/ara/blob/a9d8ed5007fe6e3b0916e75fb3326c57dc2ddc97/hardware/src/lane/simd_alu.sv#L186-L196
I think code `{sum[7], sum[7:1]} + r` will not give correct results. 
For example, if `opa.w8` and `opb.w8` equals to `0x80`, the results is `0`. However, I believe the real result should be `0x80`.

For `vasub` and `vasubu`:
https://github.com/pulp-platform/ara/blob/a9d8ed5007fe6e3b0916e75fb3326c57dc2ddc97/hardware/src/lane/simd_alu.sv#L322-L332
I believe that right shift `sub[7:0]` by 1 bit won't give correct results.

Of course, maybe my understanding is not correct, if so, I hope you can point out my problem, thanks a lot!

mp-17, 2023-09-06:
Hello @SoufSilence,

Sorry for the late reply; I was busy with other projects! :-)
I will get back to this asap, thanks for reporting!

Best,
Matteo

davidharrishmc, 2026-10-12:
Here is a reproducer.

[ara-228-vaadd-vasub.tar.gz](https://github.com/user-attachments/files/33327715/ara-228-vaadd-vasub.tar.gz)