pulp-platform/ara #233: Vivado Synthesis Error

abel-kun, 2023-07-21:
*** Running vivado
    with args -log vlsu.vds -m64 -product Vivado -mode batch -messageDb vivado.pb -notrace -source vlsu.tcl

****** Vivado v2018.1 (64-bit)
  **** SW Build 2188600 on Wed Apr  4 18:40:38 MDT 2018
  **** IP Build 2185939 on Wed Apr  4 20:55:05 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source vlsu.tcl -notrace
Command: synth_design -top vlsu -part xczu7ev-ffvf1517-1LV-i
Starting synth_design
Attempting to get a license for feature 'Synthesis' and/or device 'xczu7ev'
INFO: [Common 17-349] Got license for feature 'Synthesis' and/or device 'xczu7ev'
INFO: Launching helper process for spawning children vivado processes
INFO: Helper process launched with PID 2008 
WARNING: [Synth 8-2490] overwriting previous definition of module fpnew_pkg [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/fpu/src/fpnew_pkg.sv:485]
WARNING: [Synth 8-1921] elaboration system task error violates IEEE 1800 syntax [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/vlsu/vlsu.sv:242]
WARNING: [Synth 8-1921] elaboration system task error violates IEEE 1800 syntax [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/vlsu/vlsu.sv:245]
WARNING: [Synth 8-1921] elaboration system task error violates IEEE 1800 syntax [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/vlsu/vlsu.sv:248]
---------------------------------------------------------------------------------
Starting RTL Elaboration : Time (s): cpu = 00:00:04 ; elapsed = 00:00:04 . Memory (MB): peak = 372.145 ; gain = 113.059
---------------------------------------------------------------------------------
INFO: [Synth 8-6157] synthesizing module 'vlsu' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/vlsu/vlsu.sv:11]
	Parameter NrLanes bound to: 32'b00000000000000000000000000000100 
	Parameter AxiDataWidth bound to: 32'b00000000000000000000000001000000 
	Parameter AxiAddrWidth bound to: 32'b00000000000000000000000001000000 
	Parameter DataWidth bound to: 32'b00000000000000000000000001000000 
INFO: [Synth 8-6157] synthesizing module 'axi_cut' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/axi_cut.sv:20]
	Parameter Bypass bound to: 1'b0 
INFO: [Synth 8-6157] synthesizing module 'spill_register' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/spill_register.sv:17]
	Parameter Bypass bound to: 1'b0 
INFO: [Synth 8-6157] synthesizing module 'spill_register_flushable' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/spill_register_flushable.sv:17]
	Parameter Bypass bound to: 1'b0 
INFO: [Synth 8-6155] done synthesizing module 'spill_register_flushable' (1#1) [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/spill_register_flushable.sv:17]
INFO: [Synth 8-6155] done synthesizing module 'spill_register' (2#1) [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/spill_register.sv:17]
ERROR: [Synth 8-6038] cannot resolve hierarchical name for the item 'aw_valid' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/axi_cut.sv:50]
ERROR: [Synth 8-6038] cannot resolve hierarchical name for the item 'aw_ready' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/axi_cut.sv:51]
ERROR: [Synth 8-6038] cannot resolve hierarchical name for the item 'aw' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/axi_cut.sv:52]
ERROR: [Synth 8-6038] cannot resolve hierarchical name for the item 'aw_valid' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/axi_cut.sv:53]
ERROR: [Synth 8-6038] cannot resolve hierarchical name for the item 'aw_ready' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/axi_cut.sv:54]
ERROR: [Synth 8-6038] cannot resolve hierarchical name for the item 'aw' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/axi_cut.sv:55]
ERROR: [Synth 8-6156] failed synthesizing module 'axi_cut' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/src/axi_cut.sv:20]
ERROR: [Synth 8-6156] failed synthesizing module 'vlsu' [D:/ara_modular/lane/ara_lane.srcs/sources_1/imports/vlsu/vlsu.sv:11]
---------------------------------------------------------------------------------
Finished RTL Elaboration : Time (s): cpu = 00:00:04 ; elapsed = 00:00:04 . Memory (MB): peak = 409.617 ; gain = 150.531
---------------------------------------------------------------------------------
RTL Elaboration failed
INFO: [Common 17-83] Releasing license: Synthesis
8 Infos, 4 Warnings, 0 Critical Warnings and 9 Errors encountered.
synth_design failed
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details
INFO: [Common 17-206] Exiting Vivado at Fri Jul 21 10:56:12 2023...

I get [Synth 8-6038] cannot resolve hierarchical name for the item 'aw', for every axi related module, while synthesizing on Vivado.
Please help!

AD738560581, 2024-09-04:
Hello, our team is also deploying FPGA for ARA. Have you ever encountered a problem where the always block is too large in decode module and causes a comprehensive abnormal interruption？

fulcrum34, 2024-09-23:
Hello @AD738560581, I was able to synthesize ara_soc.sv completely using VIVADO 2019.1 and generated the bit. I did not see anything relevant to always too big etc. (Which synth tool are you using???)

However I did see an issue relevant to **this issue** (cannot resolve hierarichal name ..........). This issue appears when the top file does not have a struct defined in it. Instead it is inheriting it from its upper module. So make sure you have `ara_soc.sv` as top and it should work.