pulp-platform/ara #249: Running C++ workloads on Ara instead of C

marchuang6272, 2023-09-11:
How could I run C++ workloads? More specifically, I am trying to run https://github.com/RALC88/riscv-vectorized-benchmark-suite on Ara. This project comes with precompiled binaries, but nothing comes out on Verilator when I preload the binaries onto Ara. Therefore, I am considering using the gnu and llvm toolchain in order to create the proper binaries to run on Ara.

rseac, 2025-01-22:
@mp-17 Which benchmarks from RiVEC did you port?  I'm trying to create a take RiVEC and strip out the system calls, and file inputs in some way.  It would be helpful to see what you did to get one of those benchmarks to work so that I can follow.

mp-17, 2025-01-23:
Hey @rseac, I have ported (and modified a bit) `jacobi2d`, the `cos`/`exp`/`log` functions, `lavamd`, `pathfinder`. I have also started with `particlefilter` locally, but something was off when comparing the scalar and vector results. `axpy` is straightforward to port (or to re-write). We have our version of `matmul` and porting rivec's `matmul` should also be easy enough. Same for `spmv`.
I have not ported `blackscholes`, `canneal`, `somier`, `streamcluster, and `swaptions`, and want to try them under Linux on FPGA.

rseac, 2025-01-28:
@mp-17 I've been looking at the ported versions.  I was wondering if there is any documentation you have on how to write applications or port applications to ARA?  i.e. loading data correctly via the data.S files, etc.?