pulp-platform/ara #250: a bunch of hangs and crashes

camel-cdr, 2023-09-20:
I noticed that a bunch of the programs I want to execute on ara start hanging at some point, but since that isn't that reproducible, I've tried to find some short examples that are definitely broken.

My approach to finding the bugs below was to shuffle a [list of all instructions](https://github.com/pulp-platform/ara/files/12677616/instructions.txt), see if it breaks, and try to minimize the code while keeping the breakage. This sound time-consuming, but it wasn't because it almost always started hanging with just a few instructions.

I'm using the llvm toolchain, added the implemented foo in a new file asm.S and modified the hello_world main to the following:

```c
int main(void) {
        extern void foo(void);
        printf("BEGIN\n");
        foo();
        printf("END\n");
        return 0;
}
```

### Hangs

```asm
.text
.global foo
foo:
    vsetvli t0, x0, e32, m1, ta, ma
    vfirst.m t0,v8,v0.t
    ret
```

```asm
foo:
        vsetvli t0, x0, e32, m1, ta, ma
        vmseq.vi v8,v16,13
        vcpop.m t0,v8,v0.t
        ret
```

```asm
foo:
        vsetvli t0, x0, e32, m1, ta, ma
        vredmax.vs v8,v16,v24,v0.t
        vslidedown.vx v8,v16,t0,v0.t
        vmv.v.v v8,v16
        ret
```

```asm
foo:
        vsetvli t0, x0, e32, m1, ta, ma
        vredor.vs v8,v16,v24
        vslidedown.vi v8,v16,3
        vmxnor.mm v0,v8,v16
        ret
```

```asm
foo:
        vsetvli t0, x0, e32, m1, ta, ma
        vid.v v0
        vid.v v8
        vid.v v16
        vid.v v24
        vadd.vv v0, v8, v16 # these can be almost any two instructions
        vsub.vv v0, v8, v16 # and it will also hang, e.g: vsub&vmul,vmseq&vxor, ...
        ret
```

```asm
foo:
        vsetvli t0, x0, e32, m1, ta, ma
        vrsub.vx v8,v16,t0
        vmsbf.m v8,v16
        vsrl.vi v8,v16,13,v0.t
        vslideup.vi v8,v16,3,v0.t
        vsbc.vvm v8,v16,v24,v0
        ret
```

### Crashes

```asm
foo:
        vsetvli t0, x0, e16, m8, ta, ma
        vid.v v0
        ret
```

## Full Reproduction

```Docker
FROM ubuntu:22.04

RUN apt-get update \
	&& apt-get install -y build-essential wget git unzip python3 verilator make cmake zstd python3 ninja-build texi2html \
	texinfo device-tree-compiler autoconf autotools-dev curl lsb-release clang flex libfl-dev libelf-dev bison libmpc-dev htop  \
	help2man vim \
	&& apt-get clean \
	&& rm -rf /var/lib/apt/lists/*

RUN git clone https://github.com/pulp-platform/ara/

WORKDIR ara

COPY main.c apps/vfmv_test/main.c

RUN git submodule update --init --recursive

RUN make toolchain-llvm
RUN make riscv-isa-sim
RUN make verilator

RUN /usr/bin/install -c /ara/install/verilator/bin/verilator_bin /ara/install/verilator/share/verilator/

RUN make -C hardware apply-patches && make -C hardware verilate

```

```bash
$ # modify apps/hello_world/main.c and  apps/hello_world/asm.c
$ make bin/hello_world && app=hello_world make -C ../hardware/ simv
```

Note: I haven't looked at floating point, widening and narrowing instructions yet.

I also ran the built-in tests and `rv64uv-ara-vl_nocheck` hangs, and `rv64uv-ara-vssub` fails.

mp-17, 2023-11-15:
Hi @camel-cdr, thanks a lot for reporting. We really need to integrate random sequence generation tests in our verification flow since, thus far, we only used hand-assembled tests and vectorized benchmarks that do not expose more exotic patterns. 
The repo has been silent for a while because of internal exploration and evaluation, I will get back to this asap :-)

mp-17, 2023-12-08:
Hello @quswarabid, thank you very much; that would be fantastic! If you want and have time, we can schedule a meeting and discuss it.

mp-17, 2023-12-08:
Thank you, @quswarabid, waiting for it! Have a great weekend too! :-)

mp-17, 2024-06-27:
Hey @camel-cdr,

I am now dedicating some time every week to verification and bug fixes.
This branch should solve all the hangs/crashes you posted: 
https://github.com/pulp-platform/ara/tree/mperotti/debug

I want to start solving all the bugs highlighted in the repo `Issues`. If you like, we can also have a call and discuss the problems you encountered. I see you have a repo with benchmarks https://github.com/camel-cdr/rvv-bench; I have not tried them yet, but I guess you have met problems running them.
Let me know!

camel-cdr, 2024-06-27:
Hi @mp-17,
thats great to hear. Yes, I run into problems trying my benchmark, and created the small script so I can share reproducible bugs. (The credit in the header is enough :+1:)

I haven't tested on ara for some time now, just followed the commits, but I'll look into it again after I'm back from the rv summit.

closed 2025-01-20 by camel-cdr