pulp-platform/ara #253: invalid cast from type ‘VlWide<4>’ to type ‘IData’ (make verilate)

lianakoleva, 2023-10-04:
Running `~/ara/hardware$ make verilate` results in the following errors
```
Vara_tb_verilator___024root__DepSet_ha4067062__3.cpp: In function ‘void Vara_tb_verilator___024root___act_comb__TOP__85(Vara_tb_verilator___024root*)’:
Vara_tb_verilator___024root__DepSet_ha4067062__3.cpp:280:33: 
    error: invalid cast from type ‘VlWide<4>’ to type ‘IData’ {aka ‘unsigned int’}
...
Vara_tb_verilator___024root__DepSet_ha4067062__3.cpp:280:13: 
    error: ‘VL_ADD_I’ was not declared in this scope; did you mean ‘VL_ADD_W’?

```
Setup 
- `Ubuntu 22.04`
- `ara` and subdirectories up to date with main branch (`ec0e379`)
- `riscv-gnu-toolchain`, `newlib`, `riscv-isa-sim`, `riscv-llvm` installed from `make toolchain-llvm` command in root directory yesterday
- `verilator 5.017` installed from `make verilator` command in root directory today
- `bender 0.27.3` precompiled binary from pulp-platform [bender repo](https://github.com/pulp-platform/bender/releases)
- `libelf 0.186-1build1` is the only other thing I installed from `apt` package manager

See the whole log [here](https://github.com/pulp-platform/ara/files/12806491/log.txt).

Potentially related: running `~/ara/hardware$ make apply-patches` yields
```
Makefile:82: "Specified QuestaSim version (questa-2021.2) not found in PATH /tools/Xilinx/Vitis_HLS/2023.1/bin:/tools/Xilinx/Model_Composer/2023.1/bin:/tools/Xilinx/Vivado/2023.1/bin:/tools/Xilinx/DocNav:/home/liana/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/tmp/code:/home/liana/Documents/sbt/sbt/bin:/home/liana/.local/bin:/home/liana/RISCV/bin:/home/liana/verilator/bin"
cd deps/tech_cells_generic && git apply ../../patches/0001-tech-cells-generic-sram.patch
error: patch failed: src/rtl/tc_sram.sv:124
error: src/rtl/tc_sram.sv: patch does not apply
make: *** [Makefile:125: apply-patches] Error 1
```

lianakoleva, 2023-10-05:
Upon further reflection, I think the `make apply-patches` is unrelated and due to it being the non-first time I ran this command.

ckf104, 2023-10-10:
How about trying older verilator version?  I don't encounter this kind of error when using version 4.214

lianakoleva, 2023-10-16:
@ckf104 Are you up to date with `main`? I installed v4.214 and am now getting `Error: Unknown warning specified: -Wno-ENUMVALUE`.

ckf104, 2023-10-16:
Yes, I did use the older main branch. How about trying to delete this option in makefile, since older verilator won't check enum value conversion. This option line has been added for bumping to v5.012. See [this commit](https://github.com/pulp-platform/ara/commit/5327d45a27fbca65ba76d6bf83bc20803787390a).

lianakoleva, 2023-10-20:
@ckf104 Great catch, thank you! This, along with setting a symbolic link between `$VERILATOR_ROOT/share/verilator/include` and `$VERILATOR_ROOT/include` (was getting `Vara_tb_verilator.mk:71: $VERILATOR_ROOT/include/verilated.mk: No such file or directory`), got things running for me.

closed 2023-10-20 by lianakoleva