pulp-platform/ara #257: calloc throws *** FAILED *** (tohost = 1337) code when executes

SiriEmb, 2023-11-03:
Hi
I am trying to run an application on SPIKE which uses calloc, time(). The code is built properly with make bin/myownapp.spike. 

But when running it throws *** FAILED *** (tohost = 1337). 
After debugging using prints, I realized, whenever I use time() or calloc() spike throws this error code. Not sure how many other stdlib calls have this issue. 
I updated riscv-llvm to latest and I use **-march=rv64gcv_zfh_zvfh1p0.**
I am targetting for baremetal build with flags **-nostdlib -lc -lgcc**

Any insights are highly appreciated.

Best
SiriEmb

SiriEmb, 2023-11-04:
Can someone please help on this?

mp-17, 2023-11-28:
Hello @SiriEmb, I don't think you can run functions you would normally use with an OS like `time()`; they are not supported by bare-metal SPIKE. 

Best,
Matteo