pulp-platform/ara #260: `make verilate` fails

mablinov, 2023-11-08:
Hi all, first time user:

I've checked out the repo and followed all the instructions on the front page README.md. Everything works up until I get to the `make verilate` step, this is what I get:

```sh
maxim@prime:~/src/ara/hardware$ make verilate 2>&1 | tee -a out
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 > build/verilator/bender_script_default
/home/maxim/src/ara/install/verilator/bin/verilator -f build/verilator/bender_script_default           \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LATCH                                                                    \
  -Wno-LITENDIAN                                                                \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  -Wno-ENUMVALUE                                                                \
  -Wno-COMBDLY \
  --hierarchical                                                                \
  tb/verilator/waiver.vlt                                                       \
  --Mdir build/verilator                                                       \
  -Itb/dpi                                                                      \
  --compiler clang                                                              \
  -CFLAGS "-DTOPLEVEL_NAME=ara_tb_verilator"                                        \
  -CFLAGS "-DNR_LANES=4"                                              \
  -CFLAGS -I/home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  ""                                                             \
  -LDFLAGS "-lelf"                                                              \
  ""                                                              \
  --exe                                                                         \
  /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/maxim/src/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
                                       \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
make[1]: Entering directory '/home/maxim/src/ara/hardware/build/verilator'
make[2]: Entering directory '/home/maxim/src/ara/hardware'
/home/maxim/src/ara/install/verilator/bin/verilator -f build/verilator/Vlane_e_hierMkArgs.f
make[2]: Leaving directory '/home/maxim/src/ara/hardware'
make[1]: Leaving directory '/home/maxim/src/ara/hardware/build/verilator'
Makefile:80: "Specified QuestaSim version (questa-2021.2) not found in PATH /home/maxim/opt/node-v18.12.1-linux-x64/bin:/home/maxim/.local/bin:/home/maxim/opt/node-v18.12.1-linux-x64/bin:/home/maxim/.local/bin:/home/maxim/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
rm -rf build/verilator; mkdir -p build/verilator
./bender script verilator -t rtl -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define RVV_ARIANE=1 > build/verilator/bender_script_default
[33;1mwarning:[m Name issue with "cva6", `export_include_dirs` not handled
	Could relate to name mismatch, see `bender update`
/home/maxim/src/ara/install/verilator/bin/verilator -f build/verilator/bender_script_default           \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LATCH                                                                    \
  -Wno-LITENDIAN                                                                \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  -Wno-ENUMVALUE                                                                \
  -Wno-COMBDLY \
[... 10 lines cut ...]
  ""                                                             \
  -LDFLAGS "-lelf"                                                              \
  ""                                                              \
  --exe                                                                         \
  /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/maxim/src/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
                                       \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
make[1]: Entering directory '/home/maxim/src/ara/hardware/build/verilator'
make[2]: Entering directory '/home/maxim/src/ara/hardware'
/home/maxim/src/ara/install/verilator/bin/verilator -f build/verilator/Vlane_e_hierMkArgs.f
sh: 1: exec: /home/maxim/src/ara/install/verilator/share/verilator/verilator_bin: not found
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/maxim/src/ara/install/verilator/share/verilator/verilator_bin -f build/verilator/Vlane_e_hierMkArgs.f
make[2]: *** [build/verilator/Vara_tb_verilator_hier.mk:361: hier_launch_verilator] Error 127
make[2]: Leaving directory '/home/maxim/src/ara/hardware'
make[1]: *** [Vara_tb_verilator_hier.mk:370: Vlane_e/lane_e.sv] Error 2
make[1]: Leaving directory '/home/maxim/src/ara/hardware/build/verilator'
%Error: make -C build/verilator -f Vara_tb_verilator_hier.mk  -j 1  hier_verilation exited with 2
%Error: Command Failed ulimit -s unlimited 2>/dev/null; exec /home/maxim/src/ara/install/verilator/bin/verilator_bin -f build/verilator/bender_script_default -GNrLanes=4 -O3 -Wno-BLKANDNBLK -Wno-CASEINCOMPLETE -Wno-CMPCONST -Wno-LATCH -Wno-LITENDIAN -Wno-UNOPTFLAT -Wno-UNPACKED -Wno-UNSIGNED -Wno-WIDTH -Wno-WIDTHCONCAT -Wno-ENUMVALUE -Wno-COMBDLY --hierarchical tb/verilator/waiver.vlt --Mdir build/verilator -Itb/dpi --compiler clang -CFLAGS -DTOPLEVEL_NAME=ara_tb_verilator -CFLAGS -DNR_LANES=4 -CFLAGS -I/home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp -CFLAGS -I/home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp -CFLAGS -I/home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp  -LDFLAGS -lelf  --exe /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/dpi_memutil.cc /home/ma [... cut ...] dpi/cpp/sv_scoped.cc /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/verilator_memutil.cc /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilated_toplevel.cc /home/maxim/src/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/verilator_sim_ctrl.cc /home/maxim/src/ara/hardware/tb/verilator/ara_tb.cpp --cc --top-module ara_tb_verilator
make: *** [Makefile:168: build/verilator/Vara_tb_verilator] Error 2
```

mablinov, 2023-11-08:
@quswarabid That did the trick! I can build the normal verilator model (and run it) without having to run manual commands.

Unfortunately, the `trace=1` variant is still segfaulting (probably an unrelated issue):

```shell
$ make simv app=hello_world trace=1
Makefile:80: "Specified QuestaSim version (questa-2021.2) not found in PATH /home/maxim/opt/node-v18.12.1-linux-x64/bin:/home/maxim/.local/bin:/home/maxim/opt/node-v18.12.1-linux-x64/bin:/home/maxim/.local/bin:/home/maxim/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin"
build/verilator/Vara_tb_verilator -t -l ram,/home/maxim/src/ara/apps/bin/hello_world,elf
Program header number 0 in `/home/maxim/src/ara/apps/bin/hello_world' low is 80000000
Program header number 0 in `/home/maxim/src/ara/apps/bin/hello_world' high is 80000fbd
Program header number 1 in `/home/maxim/src/ara/apps/bin/hello_world' high is 8000139f
Program header number 2 in `/home/maxim/src/ara/apps/bin/hello_world' high is 800013c7
Program header number 3 in `/home/maxim/src/ara/apps/bin/hello_world' is not of type PT_LOAD; ignoring.
Set `ram TOP.ara_tb_verilator.dut.i_ara_soc.i_dram 10 0x80000000 0x100000 write with offset: 0x0 write with size: 0x13c8
Simulation of Ara
=================

Tracing can be toggled by sending SIGUSR1 to this process:
$ kill -USR1 3dd0db
make: *** [Makefile:210: simv] Segmentation fault (core dumped)
```

Version:
```shell
$ ../install/verilator/bin/verilator_bin --version
Verilator 5.012 2023-06-13 rev v5.012
```

mablinov, 2023-11-08:
@quswarabid 

I replaced the `-O3` with `-O0` in [this line](https://github.com/pulp-platform/ara/blob/ec0e37916b901961beab3974acc44ab5ea422db6/hardware/Makefile#L171), and I can generate the traces now. Maybe the segfault is really just an assert or something?

closed 2023-11-13 by mablinov

llhe110, 2023-12-18:
Dear quswarabid,
   I also encountered this problem, how do you deal with it now?

```
llhe@llhe-virtual-machine:~/rscratch/workspace/ara/ara/hardware$ app=hello_world make simv trace=1
Makefile:84: "Specified QuestaSim version (questa-2021.2) not found in PATH /home/llhe/rscratch/cadtools/QuestaSim2021.2.1Linux/install/questasim/RUVM_2021.2:/home/llhe/rscratch/cadtools/QuestaSim2021.2.1Linux/install/questasim/linux_x86_64:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/llhe/rscratch/cadtools/interlijIDEA/idea-IC-232.10227.8/bin"
build/verilator/Vara_tb_verilator -t -l ram,/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world,elf
Program header number 0 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' low is 80000000
Program header number 0 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' high is 80000fbd
Program header number 1 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' high is 8000139f
Program header number 2 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' high is 800013c7
Program header number 3 in `/home/llhe/rscratch/workspace/ara/ara/apps/bin/hello_world' is not of type PT_LOAD; ignoring.
Set `ram TOP.ara_tb_verilator.dut.i_ara_soc.i_dram 10 0x80000000 0x100000 write with offset: 0x0 write with size: 0x13c8
Simulation of Ara
=================

Tracing can be toggled by sending SIGUSR1 to this process:
$ kill -USR1 15d8
make: *** [Makefile:232: simv]
```

Tanishqgithub, 2024-07-04:
@quswarabid @llhe110 

1. 
For fst i tried the solution by changing -O3 to -O0 but the simulation sort of stopped midway and gave this o/p : 
```
Set `ram TOP.ara_tb_verilator.dut.i_ara_soc.i_dram 10 0x80000000 0x100000 write with offset: 0x0 write with size: 0x81ff8

Simulation of Ara

=================

Tracing can be toggled by sending SIGUSR1 to this process:

$ kill -USR1 2515

Tracing enabled.

Writing simulation traces to sim.fst

Simulation running, end by pressing CTRL-c.

=============

  FMATMUL  =

============

------------------------------------------------------------

Calculating a (4 x 4) x (4 x 4) matrix multiplication...

-----------------------------------------------------------

alculating fmatmul...

The execution took 744 cycles.

he performance is 0.172043 FLOP/cycle (2.150538% utilization).

------------------------------------------------------------

Calculating a (8 x 8) x (8 x 8) matrix multiplication...

-----------------------------------------------------------

Calculating fmatmul...

The execution took 931 cycles.

The performance is 1.099893 FLOP/cycle (13.748657% utilization).

------------------------------------------------------------

Calculating a (16 x 16) x (16 x 16) matrix multiplication...

-----------------------------------------------------------

Calculating fmatmul...

The execution took 2593 cycles.

The performance is 3.159275 FLOP/cycle (39.490936% utilization).

------------------------------------------------------------

Calculating a (32 x 32) x (32 x 32) matrix multiplication...

-----------------------------------------------------------

Calculating fmatmul...

The execution took 11446 cycles.

[121474] %Warning: ara_tb_verilator.sv:45: TOP.ara_tb_verilator: Core Test *** FAILED *** (tohost = 2)

- /home/tanishq/Documents/ARA_new/ara/hardware/tb/ara_tb_verilator.sv:52: Verilog $finish

Received $finish() from Verilog, shutting down simulation.

Simulation statistics

=====================

Executed cycles:  ed41

Wallclock time:   147.126 s

Simulation speed: 412.823 cycles/s (0.412823 kHz)

Trace file size:  231eb5a B

You can view the simulation traces by calling

$ gtkwave sim.fst

make: *** [Makefile:232: simv] Error 2

make: Leaving directory '/home/tanishq/Documents/ARA_new/ara/hardware'
```
and while viewing sim.fst in terminal : 
```
tanishq@Tanishq:~/Documents/ARA_new/ara$ gtkwave sim.fst

Gtk-Message: 08:07:46.919: Failed to load module "canberra-gtk-module"

GTKWave Analyzer v3.3.104 (w)1999-2020 BSI

GTKWAVE | Could not initialize 'sim.fst', exiting.
```
and by manually opening the sim.fst in the file location i could see only signals of CVA6

2. **Along with this other query is how can i see the signals for the ARA [vector processor] only .** 

3. And in this same issue at last it is suggested to bump down verilator version to v4.214 , i did changes to make file and again ran ```make verilator &> verilator_install_4.log  ``` and the error occured is in : 
[verilator_install4.txt](https://github.com/user-attachments/files/16077099/verilator_install4.txt)