pulp-platform/ara #277: VCD Dumping,where the vcd file is saved?

llhe110, 2023-12-18:
(1)Is this function only available using the questasim tool?

(2)I run the following command.
```
cd ara
make -C apps bin/hello_world vcd_dump=1
make -C hardware simc app=hello_word vcd_dump=1
```

 May I ask where the vcd file is saved? 
I did not send the generated vcd file to.

mp-17, 2024-02-26:
Hello @llhe110, 

You can set up the `vcd_path` variable to specify a path for the `.vcd` file if you want it different from the default one (https://github.com/pulp-platform/ara/blob/main/hardware/Makefile#L76C3-L76C11).

```bash
cd ara
make -C apps bin/hello_world vcd_dump=1
make -C hardware simc app=hello_word vcd_dump=1 vcd_path=${your_fav_path}
```

I'll try with Verilator to see if it works as expected.

Let me know!

Best,
Matteo

Tanishqgithub, 2024-07-03:
@llhe110 @quswarabid 

Did it work for you ? 
If not what are the changes you did or any insights would be useful

Tanishqgithub, 2024-07-05:
> Hi @Tanishqgithub <https://github.com/Tanishqgithub>, I don't really
> remember, sorry. But you can request support from @mp-17
> <https://github.com/mp-17> with more specific details regarding this
> issue.
> Thanks,

Respected sir

Slightly unrelated to this issue .

I tried to contact you through this issue #260
<https://github.com/pulp-platform/ara/issues/260> but as it is closed I am
trying to reach here.
This is regarding that issue only.

1. For fst I tried the solution by changing -O3 to -O0 but the simulation
sort of stopped midway as you said and gave this o/p :
attached in "fst.txt".

and while viewing sim.fst in terminal :
```
***@***.***:~/Documents/ARA_new/ara$ gtkwave sim.fst

Gtk-Message: 08:07:46.919: Failed to load module "canberra-gtk-module"

GTKWave Analyzer v3.3.104 (w)1999-2020 BSI

GTKWAVE | Could not initialize 'sim.fst', exiting.
```
and by manually opening the sim.fst in the file location i could see only
signals of CVA6

2. Along with this other query is how can i see the signals for the ARA
[vector processor] only.

3. And in this same issue at last it is suggested to bump down verilator
version to v4.214 , i did changes to make file and again ran ,
``` make verilator &> verilator_install_4.log ```   and the error occured
is in :
attached in "verilator_install4.txt"

Can you please tell me how to encounter this issue and if the solution is
bumping down the verilator version then please tell me what is the solution
for the error i am occuring .

Set `ram TOP.ara_tb_verilator.dut.i_ara_soc.i_dram 10 0x80000000 0x100000 write with offset: 0x0 write with size: 0x81ff8

Simulation of Ara

=================

Tracing can be toggled by sending SIGUSR1 to this process:

$ kill -USR1 2515

Tracing enabled.

Writing simulation traces to sim.fst

Simulation running, end by pressing CTRL-c.

=============

  FMATMUL  =

============

------------------------------------------------------------

Calculating a (4 x 4) x (4 x 4) matrix multiplication...

-----------------------------------------------------------

alculating fmatmul...

The execution took 744 cycles.
[... 333 lines cut ...]
ccache clang++  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/tanishq/Documents/ARA_new/ara/install/verilator/share/verilator\" -c ../V3GraphDfa.cpp -o V3GraphDfa.o
ccache clang++  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/tanishq/Documents/ARA_new/ara/install/verilator/share/verilator\" -c ../V3GraphPathChecker.cpp -o V3GraphPathChecker.o
ccache clang++  -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/tanishq/Documents/ARA_new/ara/install/verilator/share/verilator\" -c ../V3Delayed.cpp -o V3Delayed.o
ccache clang++  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/tanishq/Documents/ARA_new/ara/install/verilator/share/verilator\" -c ../V3GraphTest.cpp -o V3GraphTest.o
ccache clang++  -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/tanishq/Documents/ARA_new/ara/install/verilator/share/verilator\" -c ../V3Depth.cpp -o V3Depth.o
ccache clang++  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/tanishq/Documents/ARA_new/ara/install/verilator/share/verilator\" -c ../V3Hash.cpp -o V3Hash.o
ccache clang++  -Og -ggdb -gz -DVL_DEBUG -D_GLIBCXX_DEBUG -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/tanishq/Documents/ARA_new/ara/install/verilator/share/verilator\" -c ../V3DepthBlock.cpp -o V3DepthBlock.o
ccache clang++  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc [... cut ...]  -DDEFENV_VERILATOR_ROOT=\"/home/tanishq/Documents/ARA_new/ara/install/verilator/share/verilator\" -c ../V3EmitCImp.cpp -o V3EmitCImp.o
ccache clang++  -O2 -MMD -I. -I.. -I.. -I../../include -I../../include -MP  -Qunused-arguments -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC=\"\" -DDEFENV_SYSTEMC_ARCH=\"\" -DDEFENV_SYSTEMC_INCLUDE=\"\" -DDEFENV_SYSTEMC_LIBDIR=\"\" -DDEFENV_VERILATOR_ROOT=\"/home/ta