pulp-platform/ara #280: Errors in Verilator Simulation for Test Applications

aaqdas, 2023-12-31:
I've compiled and tested different programs (hello_world, iconv2d, dotproduct).
Aside from hello_world, both the other programs fail to output any results. 

An excerpt of the terminal where Verilator is running shows some errors. See Below

`
```shell
Program header number 0 in `/home/aqdas/ara/ara/apps/bin/iconv2d' low is 80000000
Program header number 0 in `/home/aqdas/ara/ara/apps/bin/iconv2d' high is 80002307
Program header number 1 in `/home/aqdas/ara/ara/apps/bin/iconv2d' high is 8004e7df
Program header number 2 in `/home/aqdas/ara/ara/apps/bin/iconv2d' high is 8004ee8f
Program header number 3 in `/home/aqdas/ara/ara/apps/bin/iconv2d' high is 8004eea7
Program header number 4 in `/home/aqdas/ara/ara/apps/bin/iconv2d' is not of type PT_LOAD; ignoring.
Program header number 5 in `/home/aqdas/ara/ara/apps/bin/iconv2d' is not of type PT_LOAD; ignoring.
Set `ram TOP.ara_tb_verilator.dut.i_ara_soc.i_dram 8 0x80000000 0x100000 write with offset: 0x0 write with size: 0x4eea8
Simulation of Ara
=================

Simulation running, end by pressing CTRL-c.

[1724] %Warning: ara_tb_verilator.sv:45: TOP.ara_tb_verilator: Core Test *** FAILED *** (tohost = 1)
- /home/aqdas/ara/ara/hardware/tb/ara_tb_verilator.sv:52: Verilog $finish
Received $finish() from Verilog, shutting down simulation.

Simulation statistics
=====================
Executed cycles:  35e
Wallclock time:   0.246 s
Simulation speed: 3504.07 cycles/s (3.50407 kHz)
make: *** [Makefile:231: simv] Error 1
``` `

aaqdas, 2024-01-02:
Issue was resolved. I made a small change in crt0.s before which may have caused a problem

closed 2024-01-02 by aaqdas

LorcaQAQ, 2024-11-24:
Hi, @aaqdas ,
I noticed that you mentioned encountering errors in verilator simulation for ARA test . I'm currently facing the same problem, and I was wondering if you could share how you resolved it.