pulp-platform/ara #282: Rationale of vrf layout design

ckf104, 2024-01-14:
After reading the paper and docs, It's not very clear to me why ara takes such a bisection-like vrf layout, i.e., the next element in the same lane will be placed into another bisected parts.

In my sense, a more intuitive layout may place elements in the same lane adjacently.  For example, when sew = 16 and NrLane=2, element 0, 2, 4, 6 will be placed in the lane 0 successively, and 1, 3, 5, 7 in the lane 1 similarly.

Will layout shuffling or slide unit benefit from this bisection-like vrf layout ?

jin8495, 2024-01-14:
Hi, ckf104.

I am not an original author of the Ara, but while I have been playing with the codes, I realized that why the VRF layout of Ara is accepted is for narrowing operation.
When you think of the narrowing operation, it should be computed in a single lane. But if you adopt the layout that you suggested, you should pack them to satisfy the layout.

For example, sew=16 and NrLanes=2, and narrowing it down to sew=8.
In the layout that you suggest, like I said, you should add an additional logic to pack them.  
[image]

But, if you use the existing VRF layout.
You don't have to pack them with more effort. This also works for narrowing sew 32 down to 16. 
[image]

I am not the original author, so I just assumed it.
But I am quite sure that this is the reason.

Thank you.

ckf104, 2024-01-14:
Hi, @jin8495, thank you for your informative reply!

I have looked through sv code related  dealing with narrowing instruction in `lane/valu.sv`. It uses a single bit `narrowing_select_q` to indicate which part of 64 bits the narrowed alu result should be inserted into.

The `narrowing_select_q` is sent to `simd_alu` and named `narrowing_select_i` within it. For example
```systemverilog
        VNSRL: unique case (vew_i)
            EW8 : for (int b = 0; b < 4; b++) res.w8 [2*b + narrowing_select_i] = opb.w16[b] >>
                opa.w16[b][3:0];
            EW16: for (int b = 0; b < 2; b++) res.w16[2*b + narrowing_select_i] = opb.w32[b] >>
                opa.w32[b][4:0];
            EW32: for (int b = 0; b < 1; b++) res.w32[2*b + narrowing_select_i] = opb.w64[b] >>
                opa.w64[b][5:0];
          endcase
```
That is what I understand about how ara implement narrowing instruction.

Considering my proposed simple VRF layout, the same thing can be implemented by one bit selector, which indicates alu result should be stored in the low or high 32 bits of 64 bits data.

So it seems to me that no additional efforts required to pack narrowed data.

jin8495, 2024-01-14:
Hi, @ckf104.

Sorry for the misleading information.
If you look into the `vmfpu.sv`. It uses fpnew module which is come from [cvfpu](https://github.com/openhwgroup/cvfpu).
And it returns narrowed data the way I described above.
That's why I said it would be additional logic to pack narrowed data.

Thank you.

ckf104, 2024-01-14:
Hi, @jin8495.

I have reviewed related code in `vmfpu.sv` quickly but still don't get it. It seems that ara has some shuffle effort on output of `fpnew` from the following code in the `vmfpu.sv`
```systemverilog
        unique case (vinsn_processing_q.vtype.vsew)
          EW16: begin
            narrowing_shuffled_result[63:48] = unit_out_result[31:16];
            narrowing_shuffled_result[47:32] = unit_out_result[31:16];
            narrowing_shuffled_result[31:16] = unit_out_result[15:0];
            narrowing_shuffled_result[15:0]  = unit_out_result[15:0];
            narrowing_shuffle_be             = !narrowing_select_out_q ? 4'b0101 : 4'b1010;
          end
          EW32: begin
            narrowing_shuffled_result[63:32] = unit_out_result[31:0];
            narrowing_shuffled_result[31:0]  = unit_out_result[31:0];
            narrowing_shuffle_be             = !narrowing_select_out_q ? 4'b0011 : 4'b1100;
          end
          default: begin
            narrowing_shuffled_result[63:32] = unit_out_result[31:0];
            narrowing_shuffled_result[31:0]  = unit_out_result[31:0];
            narrowing_shuffle_be             = !narrowing_select_out_q ? 4'b0101 : 4'b1010;
          end
        endcase

       // some code is ignored

          if (narrowing(vinsn_processing_q.cvt_resize)) begin
            for (int b = 0; b < 4; b++) begin
              if (narrowing_shuffle_be[b])
                result_queue_d[result_queue_write_pnt_q].wdata[b*16 +: 16] = narrowing_shuffled_result[b*16 +: 16];
            end
          end
```
It looks like that besides one bit selector `narrowing_select_out_q`, ara take additional shuffling due to valid data being low 32 bits of `fpnew` output, which instead indicates that my simple VRF layout is more acceptable?

jin8495, 2024-01-14:
Hi, @ckf104 

I also tried to replace the VRF layout, but I failed somehow. It was quite a long time and I remember the issue was from the narrowing operation. But as you pointed out, narrowing does not seem to be a problem...

Then, I thought wrong back then or there must be some other reason I cannot recall.
Anyway, your point is right, narrowing does not seem to be a problem.

Sorry for the wrong information.

mp-17, 2024-02-26:
Hello @ckf104 and @jin8495,

Thanks for raising the issue! We initially chose this in-lane byte layout for the reason pointed out by @jin8495, but I would say that we are keeping it for historical reasons now, as the new FPU output format requires a shuffle stage anyway. 
We have not changed it yet since the complexity of this shuffle is always limited to 8 bytes in each lane and did not hurt scalability, but we can surely optimize it and probably simplify one level of muxes.

closed 2024-02-26 by ckf104