pulp-platform/ara #284: AXI handshaking logic within the vstu and addrgen violates AXI Specification

T-Szymk, 2024-01-31:
Within the AXI Specification (ARM IHI 0022 Issue K) it states : "A source is not permitted to wait until READY is asserted before asserting VALID."

However on [addrgen.sv line 558](https://github.com/pulp-platform/ara/blob/ec0e37916b901961beab3974acc44ab5ea422db6/hardware/src/vlsu/addrgen.sv#L558) there is a conditional statement which uses the "axi_ax_ready" signal to determine whether the axi_ar/aw_valid_o signals are set on the interface. 

"axi_ax_ready" is combinatorially set using the AXI addr ready signals on [line 546](https://github.com/pulp-platform/ara/blob/ec0e37916b901961beab3974acc44ab5ea422db6/hardware/src/vlsu/addrgen.sv#L546).

There is similar logic within [vlstu.sv line  207](https://github.com/pulp-platform/ara/blob/ec0e37916b901961beab3974acc44ab5ea422db6/hardware/src/vlsu/vstu.sv#L207).

mp-17, 2024-02-26:
Hello @T-Szymk,

You are right; thanks for pointing this out. 
Fixing this in https://github.com/pulp-platform/ara/pull/287/files?diff=unified&w=1

T-Szymk, 2024-02-26:
Hi,

Thanks for responding and looking into that! I didn't experience any problems per se, I was just learning how everything was implemented (as I am using the design in project) and I just happened to notice that the read/valid behaviour seemed strange in the waveform. I should have looked more closely because as you say, there is the cut splitting the comb path from the external interface. Sorry about that! 

I am happy for you to close this issue.

closed 2024-02-26 by mp-17