pulp-platform/ara #285: verilator simulation error

andrewchi77, 2024-02-19:
when i run: 

```shell
~/ara/hardware$ make verilate
``` 

i will have this error:

```shell
Makefile:83: "Specified QuestaSim version (questa-2021.2) not found in PATH /tools/Xilinx/Vitis_HLS/2021.1/bin:/tools/Xilinx/Model_Composer/2021.1/bin:/tools/Xilinx/Vivado/2021.1/bin:/tools/Xilinx/DocNav:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/opt/modelsim_ase/bin"
rm -rf build/verilator; mkdir -p build/verilator
/home/mc2/ara/hardware/../hardware/bender script verilator -t rtl -t cv64a6_imafdcv_sv39 -t tech_cells_generic_include_tc_sram -t tech_cells_generic_include_tc_clk -t ara_test -t cva6_test -t verilator --define NR_LANES=4 --define VLEN=4096 --define ARIANE_ACCELERATOR_PORT=1 > build/verilator/bender_script_default
/home/mc2/ara/install/verilator/bin/verilator -f build/verilator/bender_script_default           \
  -GNrLanes=4                                                         \
  -O3                                                                           \
  -Wno-fatal                                                                    \
  -Wno-PINCONNECTEMPTY                                                          \
  -Wno-BLKANDNBLK                                                               \
  -Wno-CASEINCOMPLETE                                                           \
  -Wno-CMPCONST                                                                 \
  -Wno-LATCH                                                                    \
  -Wno-LITENDIAN                                                                \
  -Wno-UNOPTFLAT                                                                \
  -Wno-UNPACKED                                                                 \
  -Wno-UNSIGNED                                                                 \
  -Wno-WIDTH                                                                    \
  -Wno-WIDTHCONCAT                                                              \
  -Wno-ENUMVALUE                                                                \
  -Wno-COMBDLY                                                                  \
  --hierarchical                                                                \
  tb/verilator/waiver.vlt                                                       \
  --Mdir build/verilator                                                       \
  -Itb/dpi                                                                      \
  --compiler clang                                                              \
  -CFLAGS "-DTOPLEVEL_NAME=ara_tb_verilator"                                        \
  -CFLAGS "-DNR_LANES=4"                                              \
  -CFLAGS -I/home/mc2/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp       \
  -CFLAGS -I/home/mc2/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp \
  -CFLAGS -I/home/mc2/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp \
  ""                                                             \
  -LDFLAGS "-lelf"                                                              \
  ""                                                              \
  --exe                                                                         \
  /home/mc2/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_dpi/cpp/*.cc            \
  /home/mc2/ara/hardware/tb/verilator/lowrisc_dv_verilator_memutil_verilator/cpp/*.cc      \
  /home/mc2/ara/hardware/tb/verilator/lowrisc_dv_verilator_simutil_verilator/cpp/*.cc      \
  /home/mc2/ara/hardware/tb/verilator/ara_tb.cpp                                           \
  --cc                                                                          \
                                       \
  --top-module ara_tb_verilator &&                                                  \
cd build/verilator && OBJCACHE='' make -j4 -f Vara_tb_verilator.mk
%Error: Unknown warning specified: -Wno-LATCH
make: *** [Makefile:187: build/verilator/Vara_tb_verilator] Error 1
``` 

verilator version is 5.021

0rd1narY1, 2024-04-14:
I meet the same error. Do you know how to fix it?

aitesam961, 2024-08-08:
I had this issue but a reclone and build from scratch magically fixed it. Not sure what was there. I did had to change the makefile a tiny bit
https://github.com/pulp-platform/ara/blob/cbab076314fb7554c4295c5dae24967b5f279acf/Makefile#L169

I replaced `CLANG_CC` with `clang` and `CLANG_CXX` with `clang++` as I was not able to build verilator without it.