pulp-platform/ara #289: About Mutilple Vector Core Setup

Rarity0123, 2024-03-11:
Hi All, 
I'm studying the paper of your team "Ara2: Exploring Single- and Multi-Core Vector Processing with an Efficient RVV1.0 Compliant Open-Source Processor" and try to simulate.  However, I only found the setup of single core.  Do you have open source the muti core setup for Apps and Hardware Code?  Thanks!

AD738560581, 2024-03-12:
Hello，I also want to studying the paper of "Ara2: Exploring Single- and Multi-Core Vector Processing with an Efficient RVV1.0 Compliant Open-Source Processor". But I do not found it in the Internet, could you give the way how can i find it?

csgxiong, 2024-03-12:
I am also very interested about this project.
Anyone know about this ?
Thanks!!

@mp-17 @suehtamacv

mp-17, 2024-03-12:
Hello @Rarity0123  and @csgxiong,

Even if we developed it more internally, this is one of the development branches we used and should be a good starting point: https://github.com/pulp-platform/ara/commits/feat/multi-core/

Let me know!
Best,
Matteo

csgxiong, 2024-03-12:
Thank you for your reply. I will check the multi-core branch and wish to learn something from it.
So, here is another question:
when will your team merge the ara2 branch or the multi-core branch to main? 
Or what is your plan to release a stable edition about ara2?

Best Regards

mp-17, 2024-03-12:
Hey @csgxiong,

We are still slowly finishing the cleanup of some of the last features we developed internally, like `vrgather/vcompress` and the segment memory operations (which are the missing features to have the "full" Ara2. The rest, which accounts for 99% of the paper material, should be already merged). 
In the meantime, we are also going on with the architecture development in internal branches, with various explorations that will lead to the next PRs, while merging the multi-core support into `main` is currently not a priority. 

Best,
Matteo

csgxiong, 2024-03-12:
@mp-17 
Thank you, Matteo. Now I get it.
I think this project is a good example for learning RISC-V Vector.
And thank you and your team.
I will keep following this project！

mp-17, 2024-03-12:
That is great news. Let me know if you need help with anything! :-)

closed 2024-03-12 by mp-17

AdhamRagab25, 2024-03-13:
Hello @mp-17;

Thanks for your quick responses above! I am also interested in testing some of the multi-core setup in your project, and was curious about your preferred methods for simulation. For context, I was previously experimenting on the main branch prior to discovering the paper, and subsequently the branch and this issue. I have been using Verilator almost exclusively due to some Questa/ModelSim setup issues on my end. I'm curious if you used Verilator at all to simulate some multi-core setups, or if you tended to favour Questa/ModelSim. For the later, considering it can quite exciting to observe how it works, I might consider taking some time to fix my Questa/ModelSim issues. However, if you did use Verilator, I am curious in how you got it to simulate, as I am running into issues trying to use it.

Thanks in advance!