pulp-platform/ara #291: Multi Core Setup Simulation

AdhamRagab25, 2024-03-14:
Hello @mp-17;

Thanks for your quick responses to #239! I am also interested in testing some of the multi-core setup in your project, and was curious about your preferred methods for simulation. For context, I was previously experimenting on the main branch prior to discovering the paper, and subsequently the branch and this issue. I have been using Verilator almost exclusively due to some Questa/ModelSim setup issues on my end. I'm curious if you used Verilator at all to simulate some multi-core setups, or if you tended to favour Questa/ModelSim. For the later, considering it can quite exciting to observe how it works, I might consider taking some time to fix my Questa/ModelSim issues. However, if you did use Verilator, I am curious in how you got it to simulate, as I am running into issues trying to use it.

Thanks in advance!

mp-17, 2024-03-14:
Hey @AdhamRagab25,

For the multi-core setup, we only used QuestaSim. Unluckily, we did not fix the RTL to be compiled with Verilator, so I am not surprised it does not work smoothly with it :-/

AdhamRagab25, 2024-03-14:
Thanks so much for your reply! I'll sort out my issues with QuestaSim then and get simulating - this is exciting, thank you!

closed 2024-03-14 by AdhamRagab25

shkim-emily, 2024-09-08:
Hello! I clone the feat/multi-core branch for running multi-core simulation.. 
How can I install QuestaSim? 
Could you share how to run the simulation?