pulp-platform/ara #295: Simulate fmatmul using 256x256

xieruoyi, 2024-03-23:
[image]
In the paper "Ara2: Exploring Single- and Multi-Core Vector Processing with an Efficient RVV1.0 Compliant Open-Source Processor", it says fmatmul has a size of 256x256. I tried to change def_args_fmatmul from "128 128 128" to "256 256 256". But ara set a size limit. I wonder how I can simulate the 256x256 case? Thank you very much!
[image]

mp-17, 2024-03-28:
Hello @xieruoyi,

Unluckily, I cannot try it right now, but I think you only need to increase the size of the memory in `ara_soc.sv` to 2**21:
https://github.com/pulp-platform/ara/blob/main/hardware/src/ara_soc.sv#L26
https://github.com/pulp-platform/ara/blob/main/hardware/tb/verilator/ara_tb.cpp#L27

If it does not work, just let me know, and I will try as soon as I am back in the office.

Best,
Matteo

xieruoyi, 2024-03-28:
> Hello @xieruoyi,
>
> Unluckily, I cannot try it right now, but I think you only need to increase the size of the memory in `ara_soc.sv` to 2**21: [...]

Thank you very much! It worked.

closed 2024-03-28 by mp-17