pulp-platform/ara #300: Unclear failure to make toolchain-gcc

magurmach, 2024-05-11:
My attempts to `make toolchain-gcc` failing with the following unclear messeage:

```
/usr/bin/install -c -m 644 b-header-vars /home/sshakib/riscv/ara/install/riscv-gcc/lib/gcc/riscv64-unknown-elf/9.2.0/plugin/include/b-header-vars
make[3]: Leaving directory '/home/sshakib/riscv/ara/toolchain/riscv-gnu-toolchain/build/build-gcc-newlib-stage1/gcc'
make[2]: Leaving directory '/home/sshakib/riscv/ara/toolchain/riscv-gnu-toolchain/build/build-gcc-newlib-stage1'
mkdir -p stamps/ && touch stamps/build-gcc-newlib-stage1
make[1]: Leaving directory '/home/sshakib/riscv/ara/toolchain/riscv-gnu-toolchain/build'
make: *** [Makefile:66: toolchain-gcc] Error 2
```

I am attempting to install toolchain-gcc because `riscv_tests` was failing for missing riscv-gcc specific installations.

When I try riscv_tests now, I get the following:

```
sshakib@pop-os:~/riscv/ara$ make riscv_tests
make -C apps -j4 riscv_tests && \
make -C hardware riscv_tests_simc
make[1]: Entering directory '/home/sshakib/riscv/ara/apps'
chmod +x /home/sshakib/riscv/ara/apps/common/script/align_sections.sh
/home/sshakib/riscv/ara/install/riscv-gcc/bin/riscv64-unknown-elf-gcc -mcmodel=medany -march=rv64gcv -mabi=lp64d -I/home/sshakib/riscv/ara/apps/common -static -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/crt0.S -o common/crt0-gcc.S.o
/home/sshakib/riscv/ara/install/riscv-gcc/bin/riscv64-unknown-elf-gcc -mcmodel=medany -march=rv64gcv -mabi=lp64d -I/home/sshakib/riscv/ara/apps/common -static -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/printf.c -o common/printf-gcc.c.o
/home/sshakib/riscv/ara/install/riscv-gcc/bin/riscv64-unknown-elf-gcc -mcmodel=medany -march=rv64gcv -mabi=lp64d -I/home/sshakib/riscv/ara/apps/common -static -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/string.c -o common/string-gcc.c.o
rm -f /home/sshakib/riscv/ara/apps/common/link.ld && cp /home/sshakib/riscv/ara/apps/common/arch.link.ld /home/sshakib/riscv/ara/apps/common/link.ld
/home/sshakib/riscv/ara/apps/common/script/align_sections.sh 4 /home/sshakib/riscv/ara/apps/common/link.ld
common/string.c:1:10: fatal error: ctype.h: No such file or directory
    1 | #include <ctype.h>
      |          ^~~~~~~~~
compilation terminated.
In file included from common/printf.c:34:
/home/sshakib/riscv/ara/install/riscv-gcc/lib/gcc/riscv64-unknown-elf/9.2.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
/home/sshakib/riscv/ara/install/riscv-gcc/bin/riscv64-unknown-elf-gcc -mcmodel=medany -march=rv64gcv -mabi=lp64d -I/home/sshakib/riscv/ara/apps/common -static -std=gnu99 -O3 -ffast-math -fno-common -fno-builtin-printf  -DNR_LANES=4 -DVLEN=4096 -Wunused-variable -Wall -Wextra -Wno-unused-command-line-argument  -c common/serial.c -o common/serial-gcc.c.o
In file included from common/serial.c:1:
/home/sshakib/riscv/ara/install/riscv-gcc/lib/gcc/riscv64-unknown-elf/9.2.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
compilation terminated.
make[1]: *** [/home/sshakib/riscv/ara/apps/common/runtime.mk:120: common/string-gcc.c.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [/home/sshakib/riscv/ara/apps/common/runtime.mk:120: common/serial-gcc.c.o] Error 1
make[1]: *** [/home/sshakib/riscv/ara/apps/common/runtime.mk:120: common/printf-gcc.c.o] Error 1
rm common/crt0-gcc.S.o
make[1]: Leaving directory '/home/sshakib/riscv/ara/apps'
make: *** [Makefile:174: riscv_tests] Error 2
```

If I understand correctly, it's failure due to missing header libraries. Any advice?

mp-17, 2024-05-15:
Hello Shakib,
I think so. The GCC toolchain installation errored out, but it is not clear what happened. Do you have a verbose log of the installation or a hint about the error that occurred at installation time?

magurmach, 2024-05-17:
Hi,

Here is the following error I am getting with `make VERBOSE=1 toolchain-gcc`:

```
/usr/bin/install -c -m 644 ../../../riscv-gcc/gcc/xcoffout.h /home/sshakib/riscv/ara/install/riscv-gcc/lib/gcc/riscv64-unknown-elf/9.2.0/plugin/include/xcoffout.h
/usr/bin/install -c -m 644 b-header-vars /home/sshakib/riscv/ara/install/riscv-gcc/lib/gcc/riscv64-unknown-elf/9.2.0/plugin/include/b-header-vars
make[3]: Leaving directory '/home/sshakib/riscv/ara/toolchain/riscv-gnu-toolchain/build/build-gcc-newlib-stage1/gcc'
make[2]: Leaving directory '/home/sshakib/riscv/ara/toolchain/riscv-gnu-toolchain/build/build-gcc-newlib-stage1'
mkdir -p stamps/ && touch stamps/build-gcc-newlib-stage1
make[1]: Leaving directory '/home/sshakib/riscv/ara/toolchain/riscv-gnu-toolchain/build'
make: *** [Makefile:66: toolchain-gcc] Error 2
```

I am unclear why this is happening or what specifically is the error source.

mp-17, 2024-05-17:
Maybe the error is raised by one of the four jobs on which `make` runs. 

You can modify here:
https://github.com/pulp-platform/ara/blob/main/Makefile#L68C6-L68C21

```diff
- 	$(MAKE) MAKEINFO=true -j4
+ 	$(MAKE) MAKEINFO=true -j1
```

This should serialize the build so that the error will be hopefully printed closer to when `make` errors out.

magurmach, 2024-05-17:
It was due to missing library.

Fixed by:

```
sudo apt-get install linux-headers-$(uname -r)
```

closed 2024-05-17 by magurmach