pulp-platform/ara #305: RISC-V tests failing

Vinay0203, 2024-06-07:
Hi, I am trying to clone the ara repository , built the toolchain in my Ubuntu system. 
After cloning the repository, while building the toolchain, i am getting the following failures:
`-- Could NOT find Terminfo (missing: Terminfo_LIBRARIES Terminfo_LINKABLE) 
-- Could NOT find ZLIB (missing: ZLIB_LIBRARY) (found version "1.2.11")
-- Could NOT find LibXml2 (missing: LIBXML2_LIBRARY) (found version "2.9.13")
CMake Warning at cmake/Modules/CompilerRTUtils.cmake:326 (message):
  LLVMTestingSupport not found in LLVM_AVAILABLE_LIBS
Call Stack (most recent call first):
  CMakeLists.txt:83 (load_llvm_config)

-- LLVM_MAIN_SRC_DIR: "/home/pescampus/Desktop/ARA/ara/toolchain/riscv-llvm/llvm"
-- Found Python3: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test CXX_SUPPORTS_UNWINDLIB_NONE_FLAG
-- Performing Test CXX_SUPPORTS_UNWINDLIB_NONE_FLAG - Failed
-- Looking for fopen in c
-- Looking for fopen in c - not found
-- Looking for __gcc_personality_v0 in gcc_s
-- Looking for __gcc_personality_v0 in gcc_s - not found
-- Performing Test C_SUPPORTS_NODEFAULTLIBS_FLAG
-- Performing Test C_SUPPORTS_NODEFAULTLIBS_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FFREESTANDING_FLAG
-- Performing Test COMPILER_RT_HAS_FFREESTANDING_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_OMIT_FRAME_POINTER_FLAG
-- Performing Test COMPILER_RT_HAS_OMIT_FRAME_POINTER_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_STD_C11_FLAG
-- Performing Test COMPILER_RT_HAS_STD_C11_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FCF_PROTECTION_FLAG
-- Performing Test COMPILER_RT_HAS_FCF_PROTECTION_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FPIC_FLAG
-- Performing Test COMPILER_RT_HAS_FPIC_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FPIE_FLAG
-- Performing Test COMPILER_RT_HAS_FPIE_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_BUILTIN_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_BUILTIN_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_EXCEPTIONS_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_EXCEPTIONS_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FOMIT_FRAME_POINTER_FLAG
-- Performing Test COMPILER_RT_HAS_FOMIT_FRAME_POINTER_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FUNWIND_TABLES_FLAG
-- Performing Test COMPILER_RT_HAS_FUNWIND_TABLES_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_STACK_PROTECTOR_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_STACK_PROTECTOR_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_SANITIZE_SAFE_STACK_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_SANITIZE_SAFE_STACK_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FVISIBILITY_HIDDEN_FLAG
-- Performing Test COMPILER_RT_HAS_FVISIBILITY_HIDDEN_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FRTTI_FLAG
-- Performing Test COMPILER_RT_HAS_FRTTI_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_RTTI_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_RTTI_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_FUNCTION_SECTIONS_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_FUNCTION_SECTIONS_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FTLS_MODEL_INITIAL_EXEC
-- Performing Test COMPILER_RT_HAS_FTLS_MODEL_INITIAL_EXEC - Failed
-- Performing Test COMPILER_RT_HAS_FNO_LTO_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_LTO_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_PROFILE_GENERATE_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_PROFILE_GENERATE_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_PROFILE_INSTR_GENERATE_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_PROFILE_INSTR_GENERATE_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_PROFILE_INSTR_USE_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_PROFILE_INSTR_USE_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_FNO_COVERAGE_MAPPING_FLAG
-- Performing Test COMPILER_RT_HAS_FNO_COVERAGE_MAPPING_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_MCRC32_FLAG
-- Performing Test COMPILER_RT_HAS_MCRC32_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_MSSE3_FLAG
-- Performing Test COMPILER_RT_HAS_MSSE3_FLAG - Failed
-- Performing Test COMPILER_RT_HAS_MSSE4_2_FLAG
-- Performing Test COMPILER_RT_HAS_MSSE4_2_FLAG - Failed
[... 139 lines cut ...]
-- Performing Test COMPILER_RT_HAS_ASM_LSE
-- Performing Test COMPILER_RT_HAS_ASM_LSE - Failed
-- Builtin supported architectures: riscv64
-- For riscv64 builtins preferring riscv/muldi3.S to muldi3.c
-- For riscv64 builtins preferring riscv/fp_mode.c to fp_mode.c
-- Supported architectures for crt: riscv64
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pescampus/Desktop/ARA/ara/toolchain/riscv-llvm/compiler-rt/build
cd /home/pescampus/Desktop/ARA/ara/toolchain/riscv-llvm/compiler-rt && \
cmake --build build --target install && \
ln -s /home/pescampus/Desktop/ARA/ara/install/riscv-llvm/lib/linux /home/pescampus/Desktop/ARA/ara/install/riscv-llvm/lib/clang/16/lib
[152/153] Install the project...
-- Install configuration: "l
-- Installing: /home/pescampus/Desktop/ARA/ara/install/riscv-llvm/include/orc/c_api.h
-- Installing: /home/pescampus/Desktop/ARA/ara/install/riscv-llvm/lib/linux/libclang_rt.builtins-riscv64.a
-- Installing: /home/pescampus/Desktop/ARA/ara/install/riscv-llvm/lib/linux/clang_rt.crtbegin-riscv64.o
-- Installing: /home/pescampus/Desktop/ARA/ara/install/riscv-llvm/lib/linux/clang_rt.crtend-riscv64.o
`

Although after this step, hello_world program given in apps has worked fine.
In the next step, i am facing an error in RISC-V tests and no further steps given in README are functioning.
[image]

Please help me resolve this issue, so that I can move further to follow the remaining steps in README successfully.

Tanishqgithub, 2024-06-09:
I am also facing similar problem can you please resolve the issue

mp-17, 2024-06-25:
You miss some dependencies. You can try something similar to:

```
sudo apt-get install libncurses5-dev libncursesw5-dev zlib1g-dev libxml2-dev
```
I think you will also need these for the next steps:
```
sudo apt-get install libmpc-dev ninja-build flex libfl-dev help2man libelf-dev
```

(I have not tried these commands on my machine, maybe they need some refinement)

Vinay0203, 2024-06-28:
I will try with these commands.
Thank you.

closed 2024-07-02 by Vinay0203